myhdl/myhdl #12: Problem simulating process with intbv signals of width 1

josyb, 2015-02-27:
In my registerfile module I have these processes:
```python
import os
from myhdl import *
import hdlutils, SimulateAvalon

def regrw(OFFSET, LENGTH, START, WIDTH, Clk, Reset, A, WD, Wr, Q, Pulse = None):
    ''' the WriteRead register '''

    @always_seq( Clk.posedge, reset = Reset)
    def ccregrw():
        if Wr and (A == OFFSET):
            Q.next = WD[WIDTH+START:START]

    return ccregrw

def ccassign( D, Q):
    ''' to assign the output ports from internal signals '''
    @always_comb
    def cca():
        Q.next = D

    return cca

def sim_intbv_width_1(Clk, Reset, A, WD, Wr, Rd, RQ, TestBit, TestVector):
    """   """
    lq1 = Signal(intbv(0)[1:])
    lq2 = Signal(intbv(0)[2:])
    reg1 = regrw(0,1,0,1, Clk, Reset, A, WD, Wr, lq1)
    reg2 = regrw(1,1,0,2, Clk, Reset, A, WD, Wr, lq2)
    ao1 = ccassign( lq1 , TestBit )
    ao2 = ccassign( lq2 , TestVector )

    rbd = ConcatSignal( lq2, lq1)

    @always_seq(Clk.posedge, reset = Reset)
    def mmrdr():
        if Rd:
            RQ.next = 0
            if A == 0 :
                RQ.next = rbd[1:]
            elif A == 1:
                RQ.next = rbd[:1]

    return instances()

def test_sim_intbv_width_1():

    hw_inst = sim_intbv_width_1(Clk, Reset, A, WD, Wr, Rd, RQ, TestBit, TestVector)

    ClkCount = Signal( intbv( 0 )[32:])
    tCK = 20

    @instance
    def clkgen():
        yield hdlutils.genClk(Clk, tCK, ClkCount)

    @instance
    def resetgen():
        yield hdlutils.genReset(Clk, tCK, Reset)

    @instance
    def stimulus():
        yield hdlutils.delayclks(Clk, tCK, 10)
        for i in range(2):
            yield SimulateAvalon.MMread(Clk, tCK, A, Rd, RQ, 1, i, None, None)

        # write a few things
        yield SimulateAvalon.MMwrite(Clk, tCK, A, WD, Wr, 0, 1)
        yield hdlutils.delayclks(Clk, tCK, 2)
        yield SimulateAvalon.MMwrite(Clk, tCK, A, WD, Wr, 0, 0)
        yield SimulateAvalon.MMwrite(Clk, tCK, A, WD, Wr, 1, 0x1)
        yield SimulateAvalon.MMwrite(Clk, tCK, A, WD, Wr, 1, 0x2)

        # read it all back
        for i in range(2):
            yield SimulateAvalon.MMread(Clk, tCK, A, Rd, RQ, 1, i, None, None)

        raise StopSimulation

    return instances()

def convert():
    # force std_logic_vectors instead of unsigned in Interface
    toVHDL.numeric_ports = False
    # Convert
    toVHDL(sim_intbv_width_1, Clk, Reset, A, WD, Wr, Rd, RQ, TestBit, TestVector)

if __name__ == '__main__':
    Clk = Signal(bool(0))
    Reset = ResetSignal(0, active=1, async=True)
    A = Signal(intbv(0)[2:])
    WD, RQ = [Signal(intbv(0)[32:]) for _ in range(2)]
    Wr , Rd = [ Signal(bool(0)) for _ in range(2) ]
    TestBit  = Signal(bool(0))
    TestVector = Signal(intbv(0)[2:])

    hdlutils.simulate(3000, test_sim_intbv_width_1)
    convert()
```
The output of the first process goes to an intermediate intbv signal so we can read the value back. The second process routes the intermediate signal to the output.
This works OK for bitwidths > 1 so *TestVector* goes from 0 to 1 to 2. 
But *Testbit* should go from 0 to 1 to 0, but stays 1 forever as can be seen in the vcd waveform:
[image]
The problem seems to be in the simulation of the ccassign function, in  the case of being handed Signals containing an intbv of width 1.

josyb, 2015-02-28:
One observation occurred to me:
it is actually the case that for object *ao1* the function *ccassign(D,Q)* has an *intbv()[1;]* as source and a *bool()* as destination.
Nonetheless the simulation is incorrect,

jandecaluwe, 2015-03-01:
For debugging I cannot do much with code that is not self-contained and does not fail. But as simulation issues get first priority, I tried to nail down the issue in a minimal self-checking test bench myself. However, I could not get it to fail.
```
def ccassign( D, Q):
    ''' to assign the output ports from internal signals '''
    @always_comb
    def cca():
        Q.next = D

    return cca

def test_sim_intbv_width_1():
    TestBit  = Signal(bool(0))
    lq1 = Signal(intbv(0)[1:])
    ao1 = ccassign( lq1 , TestBit )

    @instance 
    def check():
        for i in range(1000):
            v = randrange(0, 2)
            lq1.next = v 
            yield delay(1)
            assert TestBit == v

    return ao1, check
```

cfelton, 2015-03-02:
At first glance, I thought this might have been related to #18 on the retired repo.  But then I started playing around with Jan's example to see if I could invoke the behavior observed in #18.  I ended up with the test below. 

This test fails but it seems overly complicated, not sure which change made it fail, but it seems to need all three changes (?).

```python
from random import randrange
from random import choice

import myhdl
print(myhdl.__version__)
from myhdl import *

def ccassign( D, Q):
    ''' to assign the output ports from internal signals '''
    @always_comb
    def cca():
        Q.next = D

    return cca

def test_sim_intbv_width_1():
    TestBit  = Signal(bool(0))
    lq1 = Signal(intbv(0)[1:])
    lq2 = Signal(intbv(0)[3:2])
    ao1 = ccassign( lq1 , TestBit )
    ao2 = ccassign( lq2 , TestBit )
    rb = ConcatSignal(lq1, lq2)

    @instance 
    def check():
        try:
            for i in range(1000):
                v = randrange(0, 2)
                v = map(choice([int, bool]), [v])[0]
                lq1.next = v 
                lq2.next = not v
                yield delay(1)
                assert TestBit == v
                assert rb == concat(v, not v)
        except Exception, err:
            print(i, TestBit, v)
            print(i, rb, v, not v)
            raise err

    return ao1, ao2, check

for jj in range(100):
    print("simulation %4d" % (jj,))
    Simulation(test_sim_intbv_width_1()).run()
```

Not sure which change caused the error, there are three changes: random assign int or bool; second signal and ConcatSignal; and loop on simulation.  If I simply increased the inner loop value I didn't consistently get the failure, when I added the second loop it would fail, example output:

```
simulation    0
<class 'myhdl.StopSimulation'>: No more events
simulation    1
<class 'myhdl.StopSimulation'>: No more events
simulation    2
(0, Signal(intbv(True)), 0)
(0, Signal(intbv(1L)), 0, True)
```

josyb, 2015-03-02:
The self-contained example (with apologies)
```python
'''
Created on 27 Feb 2015

@author: Josy
'''

from myhdl import *

# extracted utility functions
def simulate(timesteps, mainclass):
    """Runs simulation for MyHDL Class"""
    # Run Simulation
    tb = traceSignals(mainclass)
    sim = Simulation(tb)
    sim.run(timesteps)

def genClk( Clk, tCK , ClkCount = None ):
    """ generate a clock and possible associate a clockcounter with it """
    while True:
        Clk.next = 1
        yield delay( int( tCK / 2 ))
        Clk.next = 0
        if not (ClkCount is None):
            ClkCount.next = ClkCount + 1
        yield delay( int( tCK / 2 ))

def genReset(Clk, tCK, Reset):
    """ although not strictly needed generates an asynchronous Reset with valid de-assert"""
    Reset.next = 1
    yield delay( int( tCK * 3.5))
    Reset.next = 0

def delayclks( Clk, tCK, count ):
    """ delays 'count' 'Clk's """
    for _ in range(count):
        yield Clk.posedge
    yield delay( int( tCK / 4))

def MMwrite( Clk, tCK, A, WD, Wr, offset , data, WaitRequest = None ):
    """ write a register """
    WD.next = data
    A.next = offset
    Wr.next = 1
    if not (WaitRequest is None):
        yield Clk.posedge
        while WaitRequest:
            yield Clk.posedge
    else:
        yield Clk.posedge

    yield delay( int( tCK / 4))
    Wr.next = 0

def MMread( Clk, tCK, A, Rd , RQ, READ_DELAY, offset, WaitRequest = None, result = None):
    """ read a register
         if waitrequest is used, READ_DELAY is ignored
    """
    A.next = offset
    if WaitRequest is None:
        Rd.next = 1
        yield Clk.posedge
        yield delay( int( tCK / 4))
        Rd.next = 0     # must be a single pulse
        for _ in range(READ_DELAY):
            yield Clk.posedge
        #only transfer after READ_DELAY
        if not result is None:
            result.next = RQ

        yield delay( int( tCK / 4))

    else:
        # here we hold Rd
        Rd.next = 1
        while WaitRequest:
            yield Clk.posedge

        if not result is None:
            result.next = RQ

        yield delay( int( tCK / 4))
        Rd.next = 0

# the RTL under scrutiny
def regrw(OFFSET, LENGTH, START, WIDTH, Clk, Reset, A, WD, Wr, Q ):
    ''' the WriteRead register '''

    @always_seq( Clk.posedge, reset = Reset)
    def ccregrw():
        if Wr and (A == OFFSET):
            Q.next = WD[WIDTH+START:START]

    return ccregrw

def ccassign( D, Q):
    ''' to assign the output ports from internal signals '''
    @always_comb
    def cca():
        Q.next = D

    return cca

def sim_intbv_width_1(Clk, Reset, A, WD, Wr, Rd, RQ, TestBit, TestVector):
    """   """
    lq1 = Signal(intbv(0)[1:])
    lq2 = Signal(intbv(0)[2:])
    reg1 = regrw(0,1,0,1, Clk, Reset, A, WD, Wr, lq1)
    reg2 = regrw(1,1,0,2, Clk, Reset, A, WD, Wr, lq2)
    ao1 = ccassign( lq1 , TestBit )
    ao2 = ccassign( lq2 , TestVector )

    rbd = ConcatSignal( lq2, lq1)

    @always_seq(Clk.posedge, reset = Reset)
    def mmrdr():
        if Rd:
            RQ.next = 0
            if A == 0 :
                RQ.next = rbd[1:]
            elif A == 1:
                RQ.next = rbd[:1]

    return instances()

def test_sim_intbv_width_1():

    hw_inst = sim_intbv_width_1(Clk, Reset, A, WD, Wr, Rd, RQ, TestBit, TestVector)

    ClkCount = Signal( intbv( 0 )[32:])
    tCK = 20

    @instance
    def clkgen():
        yield genClk(Clk, tCK, ClkCount)

    @instance
    def resetgen():
        yield genReset(Clk, tCK, Reset)

    @instance
    def stimulus():
        yield delayclks(Clk, tCK, 10)
        for i in range(2):
            yield MMread(Clk, tCK, A, Rd, RQ, 1, i, None, None)

        # write a few things
        yield MMwrite(Clk, tCK, A, WD, Wr, 0, 1)
        yield delayclks(Clk, tCK, 2)
        yield MMwrite(Clk, tCK, A, WD, Wr, 0, 0)
        yield MMwrite(Clk, tCK, A, WD, Wr, 1, 0x1)
        yield MMwrite(Clk, tCK, A, WD, Wr, 1, 0x2)

        # read it all back
        for i in range(2):
            yield MMread(Clk, tCK, A, Rd, RQ, 1, i, None, None)

        raise StopSimulation

    return instances()

def convert():
    # force std_logic_vectors instead of unsigned in Interface
    toVHDL.numeric_ports = False
    # Convert
    toVHDL(sim_intbv_width_1, Clk, Reset, A, WD, Wr, Rd, RQ, TestBit, TestVector)

if __name__ == '__main__':
    Clk         = Signal(bool(0))
    Reset       = ResetSignal(0, active=1, async=True)
    A           = Signal(intbv(0)[2:])
    WD, RQ      = [Signal(intbv(0)[32:]) for _ in range(2)]
    Wr , Rd     = [Signal(bool(0)) for _ in range(2) ]
    TestBit     = Signal(bool(0))
    TestVector  = Signal(intbv(0)[2:])

    simulate(3000, test_sim_intbv_width_1)
    convert()

```

jandecaluwe, 2015-03-02:
@cfelton In your code, Signal TestBit is assigned from concurrent processes, so you have nondeterministic behavior.

cfelton, 2015-03-02:
@jandecaluwe yes, definitely not intended.  Sorry for the distraction.

cfelton, 2015-03-02:
Failure to generate a failing test suggests the issue might be in the VCD generation.  I added a simple print monitor to @josyb 's complete example and it prints as expected, `TestBit` follows `lq1` (assuming I have no typos / bugs)

snip from the print monitor

```
     333: 1 -> 1
     334: 1 -> 1
     335: 1 -> 1
     336: 1 -> 1
     337: 1 -> 1
     338: 1 -> 1
     339: 1 -> 1
     340: 1 -> 1
     341: 0 -> 0
     342: 0 -> 0
     343: 0 -> 0
     344: 0 -> 0
     345: 0 -> 0
     346: 0 -> 0
     347: 0 -> 0
     348: 0 -> 0
```

This is the code added to `sim_intbv_width_1`

```
@instance
def monsignals():
    while True:
        print("%8d: %d -> %d" % (now(), lq1, TestBit,))
        yield delay(1)
```

I did not inspect the VCD file to determine if it was the VCD file or the VCD viewer.  I did run @josyb 's example and viewed it with gtkwave and saw the same waveforms.

jandecaluwe, 2015-03-02:
Same conclusion here. Can't get things to fail with a test bench, but VCD file indeed looks in error. No idea yet - I need to get this smaller to debug.

josyb, 2015-03-02:
The good news is:
a. two persons looking into the VCD generation
b. two and a half persons  involved with the problem
b. LBNL, no basic issue with simulation, so I can use the component in other modules

I elaborate on Chris' approach to monitor the output:
 and added another monitor:
```python
    lv = Signal(bool(0))
#     lv = Signal(intbv(0)[1:])
    @instance
    def monitor():
        lv.next = TestBit
        while True:
            yield Clk.posedge
            if TestBit != lv:
                print( "TestBit toggled to {} at clock {}" .format( TestBit, ClkCount ))
                lv.next = TestBit
            else:
                print( "Testbit: {} at clock {}" .format( TestBit, ClkCount ))
```
and get this trace:
```text
Testbit: False at clock 10
Testbit: False at clock 11
Testbit: False at clock 12
Testbit: False at clock 13
Testbit: False at clock 14
TestBit toggled to 1 at clock 15
Testbit: 1 at clock 16
Testbit: 1 at clock 17
Testbit: 0 at clock 18
Testbit: 0 at clock 19
Testbit: 0 at clock 20
```

If I add yet another stage in the assignments: 
```python
def sim_intbv_width_1(Clk, Reset, A, WD, Wr, Rd, RQ, TestBit, TestVector):
    """   """
    lq1 = Signal(intbv(0)[1:])
    lq2 = Signal(intbv(0)[2:])
    llq1 = Signal(intbv(0)[1:])

    reg1 = regrw(0,1,0,1, Clk, Reset, A, WD, Wr, lq1)
    reg2 = regrw(1,1,0,2, Clk, Reset, A, WD, Wr, lq2)
    ao1 = ccassign( lq1 , llq1 )
    ao2 = ccassign( lq2 , TestVector )

    rbd = ConcatSignal( lq2, lq1)

    @always_seq(Clk.posedge, reset = Reset)
    def mmrdr():
        if Rd:
            RQ.next = 0
            if A == 0 :
                RQ.next = rbd[1:]
            elif A == 1:
                RQ.next = rbd[:1]

    @always_comb
    def assignout():
        TestBit.next = llq1

    return instances()
```
the trace text remains the same, but in the  ,vcd the 'problem' moves to the assignout() function:
[image]

josyb, 2015-03-02:
Changing the code in the manner we would have to if we were writing VHDL ;)
```python
    @always_comb
    def assignout():
        TestBit.next = llq1[0]
```
 everything's well:
```text
Testbit: False at clock 13
Testbit: False at clock 14
TestBit toggled to True at clock 15
Testbit: True at clock 16
Testbit: True at clock 17
TestBit toggled to False at clock 18
Testbit: False at clock 19
Testbit: False at clock 20
```

jandecaluwe, 2015-03-02:
I  think I found the issue, and as @josyb suspected, it has to do with the interaction between bool's and intbv's of width 1. It is related to the fact that one is an mutable type and the other not; this makes it rather nasty where the simulation seems ok but really isn't in terms of the event-driven paradigm. Also not obvious how to write tests to expose the issue.

Hold on while I try to fix this for both branches.

jandecaluwe, 2015-03-02:
I have tried to solve this in the 2 branches.

josyb, 2015-03-02:
Simulation looks OK!
I wonder whether MyHDL could do without *bool* by replacing them with *intbv(0)[1:]* ?

jandecaluwe, 2015-03-02:
As a matter of style, I always use a bool instead of an intbv(0)[1:] - I think single bit or bool signal warrant their own type, including to make the boolean meanin explicit.

josyb, 2015-03-02:
I do as well. 
But in this case the signals are generated from a description and I didn't  intercept the intbv(0)[1:]. Which, with hindsight, I could (should?) have done. And in this project I have converted before doing the simulation. The generated VHDL looked OK, so I found it plausible to say there was a problem with the simulation. Would I have simulated first, I would have found the intbv(0{1:] *'issue'*, changed my code and this lengthy conversation wouldn't have taken place. 
I would publish my actual module on github but it doesn't convert without my proposed ConcatSignal() enhancement. I have tried various ways around this, but haven't succeeded yet. Aside from the anonymous SIgnals in the ConcatSignal() arguments plenty other anonymous Signals are generated.

jandecaluwe, 2015-03-03:
For your issue with ConcatSignal(), why can't you use named Signals? The concept of "constant wires" should be supported.

josyb, 2015-03-03:
I published the code on github: https://github.com/josyb/ControlStatus with a few caveats.
MyHDL actually does a good job naming the 'unnamed' Signals itself.  
Can you enlighten me about 'The Concept of Constant Wires'?

jandecaluwe, 2015-03-03:
If you use named Signals with the appropriate initial value (for the "constants"), does it work?

josyb, 2015-03-03:
The 'unnamed' Signals, later passed on to ConcatSignal(), are generated by the code: see lines 340 and 356. I have looked around for a way to generate names, but it seems there is no good solution.
The only way I see to avoid ConcatSignal() and  'unnamed' signals (also in lines 292, 301 and 307) is to go back to the drawing board and re-write the project like originally done in VHDL, But where is the progress then?
I just made the test-bench self-checking and the simulation has no issue with the anonymous Signals.

jandecaluwe, 2015-03-03:
I will implement it like in my proposal, hold on.

closed 2015-03-03 by jandecaluwe