myhdl/myhdl #13: Extraneous copies of top-level signals in lower-level blocks in vcd-file

josyb, 2015-02-27:
The .vcd ouput of the simulation of the module described in issue #12 shows the effect:
e.g. the signals in bold for module ao2 are superfluous 
$scope module test_sim_intbv_width_1 $end
$var reg 2 ! A $end
$var reg 1 " Reset $end
$var reg 32 # WD $end
$var reg 32 $ RQ $end
$var reg 1 % Clk $end
$var reg 2 & TestVector $end
$var reg 1 ' Rd $end
$var reg 1 ( TestBit $end
$var reg 1 ) Wr $end
$var reg 32 * ClkCount $end
$scope module hw_inst $end
$var reg 2 ! A $end
$var reg 1 " Reset $end
$var reg 32 # WD $end
$var reg 32 $ RQ $end
$var reg 1 % Clk $end
$var reg 2 & TestVector $end
$var reg 1 + lq1 $end
$var reg 2 , lq2 $end
$var reg 1 ' Rd $end
$var reg 1 ( TestBit $end
$var reg 1 ) Wr $end
$var reg 3 - rbd $end
$scope module ao2 $end
**$var reg 2 ! A $end**
**$var reg 1 " Reset $end**
**$var reg 32 # WD $end**
**$var reg 32 $ RQ $end**
**$var reg 1 % Clk $end**
**$var reg 2 & TestVector $end**
**$var reg 1 ' Rd $end**
**$var reg 1 ( TestBit $end**
$var reg 2 & Q $end
**$var reg 1 ) Wr $end**
$var reg 2 , D $end
$upscope $end
$scope module ao1 $end
$var reg 2 ! A $end
$var reg 1 " Reset $end
$var reg 32 # WD $end
$var reg 32 $ RQ $end
$var reg 1 % Clk $end
$var reg 2 & TestVector $end
$var reg 1 ' Rd $end
$var reg 1 ( TestBit $end
$var reg 1 ( Q $end
$var reg 1 ) Wr $end
$var reg 1 + D $end
$upscope $end
$scope module reg2 $end
$var reg 2 ! A $end
$var reg 1 " Reset $end
$var reg 32 # WD $end
$var reg 32 $ RQ $end
$var reg 1 % Clk $end
$var reg 2 & TestVector $end
$var reg 1 ' Rd $end
$var reg 1 ( TestBit $end
$var reg 2 , Q $end
$var reg 1 ) Wr $end
$upscope $end
$scope module reg1 $end
$var reg 2 ! A $end
$var reg 1 " Reset $end
$var reg 32 # WD $end
$var reg 32 $ RQ $end
$var reg 1 % Clk $end
$var reg 2 & TestVector $end
$var reg 1 ' Rd $end
$var reg 1 ( TestBit $end
$var reg 1 + Q $end
$var reg 1 ) Wr $end
$upscope $end
$upscope $end
$upscope $end
```
where as the correct set is:
```text
$scope module test_sim_intbv_width_1 $end
$var reg 2 ! A $end
$var reg 1 " Reset $end
$var reg 32 # WD $end
$var reg 32 $ RQ $end
$var reg 1 % Clk $end
$var reg 2 & TestVector $end
$var reg 1 ' Rd $end
$var reg 1 ( TestBit $end
$var reg 1 ) Wr $end
$var reg 32 * ClkCount $end
$scope module hw_inst $end
$var reg 2 ! A $end
$var reg 1 " Reset $end
$var reg 32 # WD $end
$var reg 32 $ RQ $end
$var reg 1 % Clk $end
$var reg 2 & TestVector $end
$var reg 1 + lq1 $end
$var reg 2 , lq2 $end
$var reg 1 ' Rd $end
$var reg 1 ( TestBit $end
$var reg 1 ) Wr $end
$var reg 3 - rbd $end
$scope module ao2 $end
$var reg 2 & Q $end
$var reg 2 , D $end
$upscope $end
$scope module ao1 $end
$var reg 1 ( Q $end
$var reg 1 + D $end
$upscope $end
$scope module reg2 $end
$var reg 2 ! A $end
$var reg 1 " Reset $end
$var reg 32 # WD $end
$var reg 1 % Clk $end
$var reg 2 , Q $end
$var reg 1 ) Wr $end
$upscope $end
$scope module reg1 $end
$var reg 2 ! A $end
$var reg 1 " Reset $end
$var reg 32 # WD $end
$var reg 1 % Clk $end
$var reg 1 + Q $end
$var reg 1 ) Wr $end
$upscope $end
$upscope $end
$upscope $end
```

josyb, 2015-03-15:
It occurred to me that, basically being a bit lazy, I started to declare all signals _globally_ (in the if \__name__ == '\__main__': section) which explains the behaviour. I revert to copy/paste the SIgnals from/to the testbench and convert functions.

jandecaluwe, 2015-04-11:
No issue then?

josyb, 2015-04-11:
Not really, would be an embellishment though and save copy/paste for the lazy engineers.

closed 2015-04-11 by josyb