myhdl/myhdl #18: test

jandecaluwe, 2015-03-02:


josyb, 2015-03-04:
I don't have that much free time myself, but I find this a useful feature (in my case necessary as in my work set-up I need it). 
I studied the toVHDL() process by adding tracing/logging and looking at the code with astviewer. It is quite complex. I managed one improvement but other attempts to 'fix' the code didn't bring the expected result. I spent several days on this already.
I too wondered whether I hadn't better written a wrapper in VHDL around the 'unsigned' conversion? And automated that process. But that would not be 'beautiful', would it?
BTW we used the numeric_ports = False on a project last year (6704 generated lines of VHDL code), and that went fine. It was only because of MyHDL that we were able to do that project in a short time. If we had to simulate the project in VHDL it would have taken much, much longer. This 'converted' me.
It is only in my latest projects that these few remaining issues popped up (Sigasi/Modelsim flags them and in last year's project we never really looked at the generated code, Quartus compiled it, so everything was fine).
In my infamous ControlStatus module ModelSim flags four issues: two I already described in #14 and #15. The third has to do with the intbv(0[1:] assignment to a bool(). I fixed that in my local production MyHDL. The fourth is when assembling a list of Signal\(intbv(0)[>1:] into a single vector.
I haven't looked into that yet, as I got stuck with #14 and #15.
To make this long story short: I volunteer.

josyb, 2015-03-08:
I already posted a PR to fix the third issue I mentioned just before.
In the mean time I continued on the three remaining issues and I was able to solve them.
But at the moment I can't create a PR for these as GitHub will combine all commits and things look a bit messy then. But I pushed them onto my fork, so you can have a look at the changes over there? I also feel that the way I solved #14 and #15 isn't that 'clean' as I created an additional 'Flag'-variable 'astinfo' to join 'SigAss, isLhs,and company'.
The fourth issue I solved by inspecting multiple levels of the 'lhs' object, and was then a simple fix.

jandecaluwe, 2015-05-25:
I have given this more thought and I am now convinced that this feature is broken and cannot be saved. To support it, we would have to dig further and further in the conversion code - complicating matters significantly for something that is only applicable to ports. I want to deprecate this.

As an alternative, I propose a `toVHDL.std_logic_ports` feature, which is `False` by default. When `True` it would separate signals from their corresponding ports and add type conversions at the top level. The advantage is it keeps issues nicely separated, and we can keep the numeric interpretation of signals in the convertor, simplifying the code. Also, it can be extended later for other types of ports if need be. The only issue I see is that the delta cycle behavior is different - but I think it is unlikely that this is a problem in practice.

I have just pushed a changeset with support for this, which just one small test, to experiment with the proposed feature.

josyb, 2015-05-25:
The beauty of the proposed solution lies in its simplicity ...
I tried it on my ControlStatus.py module and it is flawless.

jandecaluwe, 2015-07-16:
Implemented in 0.9.0.

closed 2015-07-16 by jandecaluwe