myhdl/myhdl #34: Creation of constants during conversion.

cfelton, 2015-03-17:
This is intended to be a feature enhancement request (MEP lite).

A common issue experienced by many new MyHDL users is how constants are converted.  More commonly than not, these constants are contained in a structure and managed in elaboration code versus literals used directly in the MyHDL generators.

The below is a variation of an example that was presented on the mailing-list:

The user was expecting (hoping) that `rom[0]` would be replaced with the constant value contained in `rom[0]`.  Since the `rom` is constant and the index is constant it is reasonable that `rom[constant_index]` be replaced in the converted code with a literal.  

Although the below example is a *toy example* it is a common design pattern, where the Python structures are used to manage constants, variables, and signals.  There are alternative methods
to -kinda- achieve the desired effect:

   * locally reference the constants outside of the MyHDL generators.

   * internally reference with a variable (this will generate an embedded case which the synthesizer will optimize away).

In addition, [initial value support](http://dev.myhdl.org/ideas/initial-values.html) might also provide a solution but the direct replacement seems ideal.  The enhancement request is intended to only address the `tuple(int)` or `tuple(intbv)` with a constant index.  The enhancement request is playing with the idea of direct constant conversion.

```python
from __future__ import print_function

from random import randint
from myhdl import *

def m_top_const(clock, reset, x, y, a, b, N=10):
    # a tuple of constant ints
    rom = tuple([(randint(-19,23), randint(0,127),) 
                 for _ in range(N)])

    v = [Signal(x.val) for _ in range(N-1)]
    glist = [None for _ in range(N)]

    # create multiple instances
    glist[0] = m_constants(clock, reset, x, a, b, rom[0])
    for ii in range(1,N):
        # append the constant module to the list of generators
        glist[ii] = m_constants(clock, reset, v[ii-1], a, b, rom[ii])
        
    # combine all the outputs into one output
    @always_seq(clock.posedge, reset=reset)
    def rtl():
        psum = 0          # sorta declaring the variable
        psum = psum + x   # initial value of the variable
        for ii in range(1,N):
            psum = psum + v[ii-1]
        y.next = psum

    # attach for whitebox testing
    m_top_const.rom = rom

    return glist, rtl

def m_constants(clock, reset, x, a, b, rom):
    # use fixed indexes from the ROM
    @always_seq(clock.posedge, reset=reset)
    def rtl():
        x.next = a*rom[0] + b*rom[1]

    return rtl

def test_constant():
    clock = Signal(bool(0))
    reset = ResetSignal(0, active=0, async=True)
    x = Signal(intbv(0, min=-233, max=544))
    y = Signal(intbv(0, min=-2*18, max=1111*13))
    a = Signal(intbv(0, min=-19, max=17))
    b = Signal(intbv(0, min=-3, max=37))

    # a test function
    def _test():
        # instantiate the design under test
        tbdut = m_top_const(clock, reset, x, y, a, b)
        rom = m_top_const.rom

        @always(delay(3))
        def tbclk():
            clock.next = not clock

        @instance
        def tbstim():
            reset.next = reset.active
            yield delay(33)
            reset.next = not reset.active
            
            yield clock.posedge
            yield clock.posedge
            
            psum = 0
            for vals in rom:
                psum = psum + (a*vals[0] + b*vals[1]) 
    
            assert psum == y

            raise StopSimulation

        return tbdut, tbclk, tbstim

    
    Simulation(_test()).run()
    print("** Simulation Successfull **")
    toVHDL(m_top_const, clock, reset, x, y, a, b)
    toVerilog(m_top_const, clock, reset, x, y, a, b)

if __name__ == '__main__':
    test_constant()
```

josyb, 2015-04-25:
Chris,
I worked on a fix for this. I reworked your [example](https://gist.github.com/josyb/fd599e6557a63284dfb0#file-myhdl_issue43_jb) too.
Most notable difference is that the ListOfConstant must be a list not a tuple as this will interfere with the ROM-inferral.
You can find the MyHDL source code changes on my 'ListOfConstants' branch
I verified that the conversion compiles for both VHDL and Verilog, although with warnings for the latter.

Can you take a look and comment?

Regards,

Josy

cfelton, 2015-04-28:
@josyb at a quick glance the modification looks good: straightforward and to the point.  Let me play with it and create some tests (test/conversion/general) then I can comment some more.

josyb, 2015-04-28:
@cfelton it should be relatively easy to extend this to n dimensions, I'll give that some attention. And will start making the tests too.

jandecaluwe, 2015-07-16:
I think it is best to support indexed constants in a general way (not making a difference on whether the index if constant or not.) Would #106 solve the issues?

cfelton, 2015-07-16:
@jandecaluwe, general solution makes sense.  Yes, the proposed seems probable.