myhdl/myhdl #51: Support for verilator

mrvkino, 2015-04-05:
Do you think you will add support for verilator in the future? http://www.veripool.org/wiki/verilator

Is there a way for me to add it (easily) in the meantime?

cfelton, 2015-04-05:
Does verilator support a PLI/VPI interface?  If not Cosimulation support is unlikely but you can use [convertible testbenches](http://docs.myhdl.org/en/latest/whatsnew/0.6.html#conversion-of-test-benches).  Convertible testbenches is what is commonly used with GHDL (open-source VHDL simulator).

Why use verilator instead of Icarus?

mrvkino, 2015-04-05:
From what I understand, yes!
http://www.veripool.org/projects/verilator/wiki/Manual-verilator
From a fork of the main project, but can still give a good idea.
https://github.com/jeremybennett/verilator/blob/bitsplit/include/verilated_vpi.h
The source code should give a better idea of the support for VPI . http://www.veripool.org/ftp/verilator-3.870.tgz

The reason is verilator is being actively developed and has support for systemverilog up to 2012. Icarus seems to be dormant and support only part of systemverilog 2005.

cfelton, 2015-07-09:
@mrvkino It looks like some, at some point, had some method to wrap the verilator output (C++ representation) and interface it to MyHDL.  This is awhile ago and you will need to contact the author to see if they are available:
http://thread.gmane.org/gmane.comp.python.myhdl/1549/focus=1557

You can try the verilator VPI interface with the existing VPI code (I am sure one of the developers will try when we get a chance) and see if it works with the current implementation.