myhdl/myhdl #52: shadow-of-shadows and user defined code conversion

cfelton, 2015-04-05:
A possible error was reported on the mailing-list where a shadow-of-shadow was used and passed to user defined code.  The following is the example provided:

```python
import myhdl
print(myhdl.__version__)
from myhdl import *
 
def add_example(CLK, IN_1, IN_2, SUM_OUT):
 
     SUM_OUT.driven = 'reg'
     IN_1.read = True
     IN_2.read = True
 
     in1_part = IN_1(10,0)
     in2_part = IN_2(10,0)
 
     part_of_in1_part = in1_part(5,2)
     part_of_in2_part = in2_part(5,2)
 
     #This always block isn't exactly like the verilog code.
     @always(CLK.posedge)
     def adding():
         SUM_OUT.next = IN_1 + IN_2
 
     return adding
 
add_example.verilog_code = \
"""
 always @(posedge $CLK) begin:
     $SUM_OUT[11-1:0] <= $in1_part + $in2_part;
     $SUM_OUT[32:28] <= $part_of_in1_part + $part_of_in2_part
 end
 """
 
clk = Signal(bool(0))
in1 = Signal(intbv(0)[32:])
in2 = Signal(intbv(0)[32:])
sum_out = Signal(intbv(0)[32:])
toVerilog(add_example, clk, in1, in2, sum_out)
```
https://gist.github.com/cfelton/7bb0cbd1657f5c49d2f2

The above example converts unexpectedly:

```
 always @(posedge CLK) begin:
     SUM_OUT[11-1:0] <= in1[10-1:0] + in2[10-1:0];
     SUM_OUT[32:28] <= 0 + 0
 end
```

jandecaluwe, 2015-04-05:
Do we even want to support "shadow of shadows"? Is there a relevant use case?

cfelton, 2015-04-05:
I'm not sure, the only use-case I can think of is where shadow-signals are passed to lower modules and in the lower module a shadow is also desired.

```python

def uppermod(clock, reset, x, y):
    fullstatus = Signal(intbv(0)[32:])
    substatus = fullstatus(16,0)

    inst = lowermod(substatus)
    # ...
    return inst

def lowermod(status):
    subsubstatus1 = status(4,0)
    subsubstatus2 = status(8,4)
    # ...
    return rtl
```

cfelton, 2015-04-05:
I should add, I don't know if the example above works or not (will test soon) or if the shadow-of-shadow is only an issue with user-defined code.

cfelton, 2015-04-05:
[A complete example](https://gist.github.com/cfelton/128486678fdd96166372#file-upperlower-v-L27) similar to the above upper/lower demonstrates a similar issue without user-defined-code.

Still on the fence if this really needs to be supported or not.  If not, it will be a common pitfall where someone uses a ShadowSignal instead of assigning to a new Signal.

cfelton, 2015-05-22:
This was originally reported in #10, I have recommended #10 be closed and this issue track the shadow-of-shadows.

virantha, 2021-06-27:
Please consider at least flagging an error if the user tries to shadow a shadow.  I've been banging my head against this issue (seeing None propagated into case statements and  zeros/false in generated verilog) for quite a few days, and only stumbled on this issue by accident while trying to come up with a small test case to post.  Right now, this just silently fails in the generated verilog (even though the simulation seems to do what you'd expect).

josyb, 2021-06-27:
PR #359 Solves this Shadow of Shadow problem. Unfortunately @jck, @cfelton are *hesitant* to merge it.

virantha, 2021-06-29:
Thank you @josyb , I will try applying the patch.  

I'm surprised that this is not of a higher-concern. 

If I start with a 32-bit signal at the top level, and the second level is two processing blocks of 16-bits each, which are then further divided into 8-bit blocks at the third level, how would you get an 8-bit signal to the lowest level without the kludge of combinationally assigning the input to some dummy Signal at every step before slicing?

josyb, 2022-04-28:
@virantha Did you try the patch? Any feedback?