myhdl/myhdl #59: Missing signal in generated VHDL

cbeighley, 2015-04-10:
MyHDL code below:

```
#!/usr/bin/env python
from myhdl import *
 
MAX = 16
 
def myhdl_b1(i, o):
 
  c = intbv(4,min=0,max=MAX)
 
  @always_comb
  def rtl():
    o.next = i + c
 
  return rtl
 
i, o = [Signal(modbv(min=0,max=MAX)) for i in range(2)]
 
toVHDL(myhdl_b1, i, o)
```

I expected this to get compiled into a VHDL module where c is declared between the component declaration and architecture section. However, what ends up happening is c is never declared.

See here for VHDL output:

```
-- File: myhdl_b1.vhd
-- Generated by MyHDL 0.9dev
-- Date: Fri Feb 13 13:15:38 2015
 
 
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;
use std.textio.all;
 
use work.pck_myhdl_09.all;
 
entity myhdl_b1 is
    port (
        i: in unsigned(3 downto 0);
        o: out unsigned(3 downto 0)
    );
end entity myhdl_b1;
 
 
architecture MyHDL of myhdl_b1 is
 
 
 
 
 
 
begin
 
 
 
 
 
o <= (i + c);
 
end architecture MyHDL;
```

The workarounds for this seem to be either declaring constants as integers, in which case they get used as literals in the generated VHDL, or declaring them as signals and assigning them values in a generator. It might be workable for the former to be the only supported way of doing it, but toVHDL should to throw an error rather than compiling broken VHDL.

jck, 2015-04-11:
NOTE: This example would fail to convert in 0.8
```
~/t/myhdl-issues ❯❯❯ tox issue_59.py                                                                                                                                                          
py27-stable runtests: PYTHONHASHSEED='2748229724'
py27-stable runtests: commands[0] | python ../../../issue_59.py
<class 'myhdl._intbv.intbv'>
Traceback (most recent call last):
  File "../../../issue_59.py", line 18, in <module>
    toVHDL(myhdl_b1, i, o)
  File "/home/jck/tmp/myhdl-issues/.tox/py27-stable/lib/python2.7/site-packages/myhdl/conversion/_toVHDL.py", line 158, in __call__
    genlist = _analyzeGens(arglist, h.absnames)
  File "/home/jck/tmp/myhdl-issues/.tox/py27-stable/lib/python2.7/site-packages/myhdl/conversion/_analyze.py", line 167, in _analyzeGens
    raise ConversionError(_error.UnsupportedType, n, info)
myhdl.ConversionError: File ../../../issue_59.py, line 9: Object type is not supported in this context: c
ERROR: InvocationError: '/home/jck/tmp/myhdl-issues/.tox/py27-stable/bin/python ../../../issue_59.py'
py27-dev runtests: PYTHONHASHSEED='2748229724'
py27-dev runtests: commands[0] | python ../../../issue_59.py
py34-dev runtests: PYTHONHASHSEED='2748229724'
py34-dev runtests: commands[0] | python ../../../issue_59.py
___________________________________________________________________________________________ summary ___________________________________________________________________________________________
ERROR:   py27-stable: commands failed
  py27-dev: commands succeeded
  py34-dev: commands succeeded

```

It is related to this commit: https://github.com/jandecaluwe/myhdl/commit/c2021eb8c773626a35063a3925f35de0bafcc4e7
That error check needed to be removed since it caused valid interface code to break. We need to decide whether to implement support for this usecase or reintroduce an exception.

This example works fine in an always_seq block, It seems like it wouldn't be too hard to support always_comb as well.

jck, 2015-04-11:
It looks like always_seq behaviour is incorrect. It does not respect the initial value.

This code:
```
#!/usr/bin/env python
from myhdl import *

MAX = 16

def myhdl_b1(clk, rst, i, o):

  c = intbv(4,min=0,max=MAX)

  @always_seq(clk.posedge, rst)
  def rtl():
    o.next = i + c

  return rtl

clk = Signal(False)
rst = ResetSignal(0, 0, True)
i, o = [Signal(modbv(min=0,max=MAX)) for i in range(2)]

toVHDL(myhdl_b1, clk, rst, i, o)
```

Converts to:
```
-- File: myhdl_b1.vhd
-- Generated by MyHDL 0.9.dev0
-- Date: Fri Apr 10 23:50:00 2015

library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;
use std.textio.all;

use work.pck_myhdl_090.all;

entity myhdl_b1 is
    port (
        clk: in std_logic;
        rst: in std_logic;
        i: in unsigned(3 downto 0);
        o: out unsigned(3 downto 0)
    );
end entity myhdl_b1;

architecture MyHDL of myhdl_b1 is

begin

MYHDL_B1_RTL: process (clk, rst) is
    variable c: unsigned(3 downto 0);
begin
    if (rst = '0') then
        o <= to_unsigned(0, 4);
    elsif rising_edge(clk) then
        o <= (i + c);
    end if;
end process MYHDL_B1_RTL;

end architecture MyHDL;
```