myhdl/myhdl #60: Problem using tuple for synthesizable MyHDL

cbeighley, 2015-04-10:
MyHDL code below:

```
#!/usr/bin/env python
from myhdl import *
 
W = 4
 
def myhdl_b3(in1, s, out1):
 
  t = (2,5,9,14)
 
  @always_comb
  def rtl():
    out1.next = in1 + t[s]
 
  return rtl
 
in1, out1 = [Signal(modbv()[W:]) for i in range(2)]
s = Signal(intbv()[2:])
 
toVHDL(myhdl_b3, in1, s, out1)
```
It seems like from this page, this should be valid MyHDL.

http://docs.myhdl.org/en/latest/manual/conversion.html

cbeighley, 2015-04-10:
Error message when running above code:

```
Traceback (most recent call last):
  File "myhdl_b3.py", line 19, in 
    toVHDL(myhdl_b3, in1, s, out1)
  File "/usr/local/lib/python2.7/dist-packages/myhdl/conversion/toVHDL.py", line 204, in __call_
    _convertGens(genlist, siglist, memlist, vfile)
  File "/usr/local/lib/python2.7/dist-packages/myhdl/conversion/_toVHDL.py", line 462, in _convertGens
    v.visit(tree)
  File "/usr/lib/python2.7/ast.py", line 241, in visit
    return visitor(node)
  File "/usr/local/lib/python2.7/dist-packages/myhdl/conversion/_toVHDL.py", line 1225, in visit_Module
    self.visit(stmt)
  File "/usr/lib/python2.7/ast.py", line 241, in visit
    return visitor(node)
  File "/usr/local/lib/python2.7/dist-packages/myhdl/conversion/_toVHDL.py", line 1607, in visit_FunctionDef
    self.visit_stmt(node.body)
  File "/usr/local/lib/python2.7/dist-packages/myhdl/conversion/_toVHDL.py", line 1433, in visit_stmt
    self.visit(stmt)
  File "/usr/lib/python2.7/ast.py", line 241, in visit
    return visitor(node)
  File "/usr/local/lib/python2.7/dist-packages/myhdl/conversion/_toVHDL.py", line 873, in visit_Assign
    self.visit(rhs)
  File "/usr/lib/python2.7/ast.py", line 241, in visit
    return visitor(node)
  File "/usr/local/lib/python2.7/dist-packages/myhdl/conversion/_toVHDL.py", line 651, in visit_BinOp
    self.BinOp(node)
  File "/usr/local/lib/python2.7/dist-packages/myhdl/conversion/_toVHDL.py", line 703, in BinOp
    self.visit(node.right)
  File "/usr/lib/python2.7/ast.py", line 241, in visit
    return visitor(node)
  File "/usr/local/lib/python2.7/dist-packages/myhdl/conversion/_toVHDL.py", line 1372, in visit_Subscript
    self.accessIndex(node)
  File "/usr/local/lib/python2.7/dist-packages/myhdl/conversion/_toVHDL.py", line 1423, in accessIndex
    self.visit(node.value)
  File "/usr/lib/python2.7/ast.py", line 241, in visit
    return visitor(node)
  File "/usr/local/lib/python2.7/dist-packages/myhdl/conversion/_toVHDL.py", line 1231, in visit_Name
    self.getName(node)
  File "/usr/local/lib/python2.7/dist-packages/myhdl/conversion/_toVHDL.py", line 1324, in getName
    self.raiseError(node, _error.UnsupportedType, "%s, %s" % (n, type(obj)))
  File "/usr/local/lib/python2.7/dist-packages/myhdl/conversion/_misc.py", line 150, in raiseError
    raise ConversionError(kind, msg, info)
myhdl.ConversionError: in file myhdl_b3.py, line 12:
    Object type is not supported in this context: t, type 'tuple'
```

cbeighley, 2015-04-10:
Workaround, assigning tuple values to a local variable:

```
#!/usr/bin/env python
from myhdl import *

W = 4

def myhdl_b3_workaround(in1, s, out1):

t = (2,5,9,14)

@always_comb
  def rtl():
    t_intbv = [intbv()[4:] for i in range(4)]
    for i in range(4):
      t_intbv[i][:] = t[i]
    out1.next = in1 + t_intbv[s]

return rtl

in1, out1 = [Signal(modbv()[W:]) for i in range(2)]
s = Signal(intbv()[2:])

toVHDL(myhdl_b3_workaround, in1, s, out1)
```

cbeighley, 2015-04-10:
Generated VHDL for workaround solution:

```
-- File: myhdl_b3_workaround.vhd
-- Generated by MyHDL 0.9dev
-- Date: Fri Feb 13 14:26:25 2015

library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;
use std.textio.all;

use work.pck_myhdl_09.all;

entity myhdl_b3_workaround is
    port (
        in1: in unsigned(3 downto 0);
        s: in unsigned(1 downto 0);
        out1: out unsigned(3 downto 0)
    );
end entity myhdl_b3_workaround;

architecture MyHDL of myhdl_b3_workaround is

begin

MYHDL_B3_WORKAROUND_RTL: process (s, in1) is
    type t_array_t_intbv is array(0 to 4-1) of unsigned(3 downto 0);
    variable t_intbv: t_array_t_intbv;
begin

for i in 0 to 4-1 loop
    case i is
        when 0 => t_intbv(i) := "0010";
        when 1 => t_intbv(i) := "0101";
        when 2 => t_intbv(i) := "1001";
        when others => t_intbv(i) := "1110";
    end case;
end loop;
out1 <= (in1 + t_intbv(to_integer(s)));

end process MYHDL_B3_WORKAROUND_RTL;

end architecture MyHDL;
```

jck, 2015-04-11:
Thanks for the bug report! I'm surprised no one has ever run into this before. My guess is that HW guys are very conservative when it comes to memories since synthesis tools are not too great at inferring memory.

Here is a much cleaner workaround:
```python
from myhdl import *

W = 4

def myhdl_b3(in1, s, out1):
    t = (2,5,9,14)

    @always_comb
    def rtl():
        x = t[s]
        out1.next = in1 + x

    return rtl

in1, out1 = [Signal(modbv()[W:]) for i in range(2)]
s = Signal(intbv()[2:])

toVHDL(myhdl_b3, in1, s, out1)
```

Which converts to:
```vhdl
-- File: myhdl_b3.vhd
-- Generated by MyHDL 0.9.dev0
-- Date: Sat Apr 11 00:24:03 2015

library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;
use std.textio.all;

use work.pck_myhdl_090.all;

entity myhdl_b3 is
    port (
        in1: in unsigned(3 downto 0);
        s: in unsigned(1 downto 0);
        out1: out unsigned(3 downto 0)
    );
end entity myhdl_b3;

architecture MyHDL of myhdl_b3 is

begin

MYHDL_B3_RTL: process (s, in1) is
    variable x: integer;
begin
    case to_integer(s) is
        when 0 => x := 2;
        when 1 => x := 5;
        when 2 => x := 9;
        when others => x := 14;
    end case;
    out1 <= resize(unsigned(signed(resize(in1, 5)) + x), 4);
end process MYHDL_B3_RTL;

end architecture MyHDL;
```

josyb, 2015-04-11:
This is also touched upon in the Gmane discussion: http://news.gmane.org/gmane.comp.python.myhdl which then provoked issue #34. I also created a gist: https://gist.github.com/josyb/af6b7809d976b2559ae4
And I am a HW guy ;-)
Ideally the VHDL code should look like:
```VHDL
type a_t is array (0 to 3) of natural range 0 to 15 ;
-- alternatively the (0 to 3) can be replaced by (natural range <>)
-- and the type could be unconstrained
constant t : a_t := (2, 5, 9, 14);

MYHDL_B3_RTL: process (s, in1) is
begin
    out1 <= resize(unsigned(signed(resize(in1, 5)) + t(s)), 4);
end process MYHDL_B3_RTL;

```

jandecaluwe, 2015-04-11:
Actually, this is [documented behavior](http://docs.myhdl.org/en/latest/manual/conversion.html#supported-types).

The historical reason is that tuple support was introduced for ROM inference. The template for ROM inference was based on what was found in synthesis manuals for FPGA vendors - an inlince case statement. But this only works for a specific case where the tuple access is the lhs.

I agree of course this is too limited. Ideally, we want to map the tuple to a constant or a function so that it can be used without restrictions. But that means the access is not longer inlined and we have to be sure the synthesis tools support this. I wouldn't know why not, but we have to check. I also don't want to introduce special cases (inlining or not) as workarounds for back-end tools.

Definitely something for 1.0, but I don't want to change this for 0.9.

jandecaluwe, 2015-07-16:
An implementation for #106 would solve this.

closed 2015-07-16 by jandecaluwe