myhdl/myhdl #61: [RFC] Use extended/escaped identifiers naming attribute references?

jck, 2015-04-11:
We're currently replacing periods with underscores and performing rudimentary collision detection. However, I don't think this technique is very robust and wont scale well when we start to support crazier things (like dicts).

I think we should use escaped identifiers for verilog[1] and extended identifiers for VHDL[2] to simplify the conversion logic. 

It would also prevent bugs like #33 in the future.

[1] http://verilog.renerta.com/source/vrg00018.htm
[2] http://www.seas.upenn.edu/~ese171/vhdl/vhdl_primer.html#_Toc526061347

jandecaluwe, 2015-04-11:
I understand, but are we sure all our target backend tools support this?

jmgc, 2015-04-11:
Dear all,

Why not use records in VHDL and structs in Verilog? For the first, all the VHDL93 should be able to work with them, in case of Verilog I am not an expert. But probably, different solutions can be used depending on the language/simulator.

Best,

Jose M.

jandecaluwe, 2015-04-11:
VHDL records are too limited, plain Verilog does not have structs.

jmgc, 2015-04-12:
Sorry, but why do you say that records are too limited. I have used them, and their behaviour is like structs in C. Which is the functionality you miss?

Thanks,

Jose M.

jck, 2015-04-14:
Quartus, Modelsim and Synopsys tools support this.

Based on what I can find online [1,2], Xilinx and Cadence tools also support this.

[1] http://www.xilinx.com/support/answers/1535.html
[2] https://lost-contact.mit.edu/afs/ict.kth.se/pkg/cadence/ic610/06.14.506/doc/compvhdluser/glossary.html

cfelton, 2015-04-14:
@jck, is the idea to simply add the "extended identifier" escapes at the end, a post-processing transformation?  

Is so, what is the risk if post-processing "name transformation" (i.e. replace unsupported characters) was done instead of inserting the escape `\`?  There are other "name transformations" that could be useful: keyword collisions, unsupported characters (VHDL and leading underscore), expanded names, etc.

My initial reaction was the same as Jan', concerned about interoperability.  Mainly due to my inexperience with extended identifier.  I did some quick tests with Icarus and GHDL and they both seem to support extended identifiers as well.

With the extended identifier, name expansion will look like this?

```python
myobj.x.next = yourobj.y + silly.samsobj.z
```
```verilog
\myobj.x = \yourobj.y + \silly.samsobj.z ;
```
```vhdl
\myobj.x\ = \yourobj.y\ + \silly.samsobj.z\ ;
```

jck, 2015-04-14:
Yes, each conversion module would have a write_name function which would take care of avoiding invalid names. That function can apply all the checks required for the target HDL.

jck, 2015-04-14:
I think we should still prefer underscores for top level ports. One reason is that qsys uses underscores in the naming scheme for automatically detecting interfaces.

cfelton, 2015-11-11:
@nturley has created [MEP113](http://dev.myhdl.org/meps/mep-113.html) that expands on the ideas proposed in this issue.