myhdl/myhdl #70: Installing with pip for python 3 problem

benreynwar, 2015-05-04:
Presumably the problem is that pip isn't grabbing the correct code.
Quite possibly this is something I'm doing wrong on my end but I thought I'd post just in case, since probably most people aren't using it with python3 yet.

```
ben@limner:~/Code/communityshare$ sudo pip3 install myhdl
[sudo] password for ben: 
Downloading/unpacking myhdl
  Downloading myhdl-0.8.1.tar.gz (579kB): 579kB downloaded
  Running setup.py (path:/tmp/pip_build_root/myhdl/setup.py) egg_info for package myhdl
    Traceback (most recent call last):
      File "<string>", line 17, in <module>
      File "/tmp/pip_build_root/myhdl/setup.py", line 15
        print versionError
                         ^
    SyntaxError: invalid syntax
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
```

jck, 2015-05-04:
v0.8 does not support python 3.

benreynwar, 2015-05-04:
That would explain it.  Sorry for bugging you.

cfelton, 2015-05-04:
@benreynwar you can use the `master` branch with python3 or use pip to install from testpypi https://testpypi.python.org/pypi/myhdl/0.9.dev1

  >> pip install -i https://testpypi.python.org/pypi myhdl

jck, 2015-05-04:
testpypi deletes packages periodically, I suggest installing dev versions directly from github.
```pip install https://github.com/jandecaluwe/myhdl/archive/master.zip```

cfelton, 2015-05-04:
@jck Thanks, didn't realize this is the case with testpypi.  testpypi is not really for distributing test/development version but for testing distributions :)

Is there a preference over the above pip install versus:

```
pip install git+https://github.com/jandecaluwe/myhdl
```

Or is it simply two ways to do the same thing?

jck, 2015-05-04:
testpypi is for testing if your package is all right before distributing it on the real pypi.

AFAIK, there is no difference between pip install git+https and pip installing the zip. The zip is just slightly faster than git.

closed 2015-07-16 by jandecaluwe