myhdl/myhdl #72: create gitter.im room

jck, 2015-05-06:
https://gitter.im is a nice chatroom integrated with github. I think it will provide a nice addition to the issue tracker and promote discussion. example: https://gitter.im/neovim/neovim

cfelton, 2015-05-06:
I agree, this might not be to bad because you can hail with github user name (e.g. @jck) and reference github issues.  General discussion can be moved to gitter and reference and issue.

Possible downsides: there will be multiple venues for discussion (might not actually be a downside); gitter is more like IRC, there is only one thread, all the conversations will be intermingled, I guess if that is a problem the mailing-list can be used.

josyb, 2015-05-12:
IMHO it would be OK (even great) for 'chatting' but any 'serious' discussion needs a better solution. We can fall back to the issues, like we're doing here, but that's not the right place for general discussions, I'd say.

cfelton, 2015-05-12:
@josyb, gitter would be for general discussions and explorations whereas the issues would be concrete/specific issues.  The benefit of gitter is the github integration, you can tag issues (e.g. #72) in gitter.  If someone had a general questions about an issue/PR it could be discussed on gitter and not the issue feed.

I would vote, gitter is the last public communication channel, otherwise we are getting spread thin.  We would have the following as official communication channels:

1.  mailing-list: been there from the beginning
2. IRC @freenode: used by many, required by GSoC
3. github issues/pr
4. gitter: low entry for new users, only requires a browser, github integration.

There are reasons foreach but I don't think we should go beyond the above list.  The first two simply have momentum.

josyb, 2015-05-12:
The Gmane mailing list is a bit painful, or perhaps I have the wrong view/link to it (http://news.gmane.org/gmane.comp.python.myhdl) ? 
For gitter you have to login to github first as well.

cfelton, 2015-05-12:
@josyb the web interface isn't the best.  The mailing-list is best used with a mail reader like thunderbird (IMO, see below).  Yes, I think gitter would mainly be useful for github users new to myhdl that don't have much experience with the other options of for experienced user that want to link to an issue/PR.

[image]

jck, 2015-05-16:
gitter can be accessed through irc (irc.gitter.im) in addition to the webapp. We can even run a gitter <-> freenode bridge if need be.

jandecaluwe, 2015-07-16:
Done.

closed 2015-07-16 by jandecaluwe