myhdl/myhdl #73: Simulation failures after simulation suspend

cfelton, 2015-05-07:
Currently there are two ways to run a simulation: run to completion or run a certain duration:

Run to completion
```python
sim = Simulation(top())
sim.run()
```

In the above example, the simulation will run until no more events or `StopSimulation` is raised (or an error occurs).

When the simulator runs to completion it does a `finalize` and resets all the `Signals` under simulation control back to the initial value.  The user state is the same as it entered.  New simulation instances can be created and additional simulations can be run.

Run for a duration
```python
sim = Simulation(top())
sim.run(100)
# the simulator is paused/suspended

# resume and run some more
sim.run(100)  
```

Running for a duration allows a user to run a number of simulation steps, pause, and resume.

There is a bug when the simulator is paused and a new simulation instance is created with reused signals, e.g. signals that are in a global state in the user space.

There were two proposed solutions:

1. Always `initialize` when a new instance is created. The
   `initialize` would be similar to `finalize` in that the
   simulator guarantees all user `Signals` are in an initial
   state (will be redundant if the user has no global signals).

2. Add a `quit` method to the simulator to end simulations
   that have been suspended (vs. stopped).  And don't allow
   new simulation instances if the simulator is in a run
   or suspend state (new instances only valid if none exist
   or stopped).

The second solution was preferred as discussed in #67.

A [previous example](https://gist.github.com/hgomersall/3aa0fe0eaec408555a7d) demonstrates when user global signals are used and the simulator is suspended subsequent simulation instances and runs can cause unexpected (incorrect) results.

jandecaluwe, 2015-05-08:
I think a PR for #2 shouldn't be too hard.

jandecaluwe, 2015-07-16:
Superseded by the proposal in #104.

closed 2015-07-16 by jandecaluwe