myhdl/myhdl #75: Define a numeric default value for tristate signals

jmgc, 2015-05-21:
When converting a tristate signal to an hdl, if it's numeric value is requested, for example using int, an error is generated as the default value is None. To avoid this problem, a metavalue warning could be generated, and a default value of 0 could be provided. This can be done in the _Signal class modifying the __int__(), __long__(), __float__() and __float__() functions to take care of this case.

jandecaluwe, 2015-05-25:
Using an undefined value in a numeric operation is an error. I don't see what's the benefit of masking that.

jmgc, 2015-05-25:
The reason is that if you try to use a conversion from an intbv tristate signal to int, as the default value is None, the result is that it cannot be converted, nor verified (at least with toVHDL) because it detects a conversion to int from a None value.

On the other hand, this solution gives a warning indicating that the value is unknown, but it defaults to 0, so the conversion can be done.

As a matter of fact, this behavior is equivalent to the one in vhdl numeric package.

jandecaluwe, 2015-05-25:
Attempting to convert an unknown points a real design error which is problematic in hardware. It should be corrected by design, and not masked by converting an error into a warning.

MyHDL's intbv is *not* based on the vhdl numeric packages on purpose.

closed 2015-05-25 by jandecaluwe