myhdl/myhdl #82: Interface conversion broken in the master branch

cfelton, 2015-05-27:
I have a [project](https://github.com/cfelton/minnesota) that fails with the latest code on the master branch.  According to git bisect the bug was introduced in 08519b4 [rename attribute ref in the AST before analysis](https://github.com/jandecaluwe/myhdl/commit/08519b452f153885d3cc43038111b7779e2bb2f1).

Warning: this issue simply describes  an observed failure.  Other than using `git bisect` to determine when the bug was introduced, no additional debugging as been completed or a standalone test case created.  Also, the code linked above passes travis-ci because it clones a previous version of the `master` branch (i.e. before 08519b4, it clones 88cb76b).

jmgc, 2015-05-27:
I detected also this failure (see #64). As far as I know @jck is working on solving it.

Meanwhile, I solved the problem substituting:

```python
        name = node.value.id + '_' + node.attr
        new_name = next(s for s in _suffixer(name) if s not in self.data.symdict)
        self.data.symdict[new_name] = attrobj
        self.data.objlist.append(new_name)
```

by

```python
        new_name = node.value.id + '_Dot_' + node.attr
        if new_name not in self.data.symdict:
            self.data.symdict[new_name] = attrobj
            self.data.objlist.append(new_name)
        else:
            pass
```

Things seem to work well enough, though as I've said, it is a workaround.

cfelton, 2015-05-27:
@jmgc, thanks for connecting the issues and the suggested workaround.  I don't need any of the upstream changes, using 88cb76b works for me (I just need to be careful between `mn` work and `myhdl` work, probably time to tame virtualenv :).

@jmgc the [gist link](https://gist.github.com/0fb34c528b9ab59a0256.git) to your test example seems to be dead?

jmgc, 2015-05-27:
Ups, I added a .git after the link. The good one is:

https://gist.github.com/jmgc/0fb34c528b9ab59a0256

Thanks

cfelton, 2015-05-27:
@jmgc, @jck here is a simplified test the produces the error: https://gist.github.com/cfelton/a6b6457b844c5ab9778f

In the near future I should be able to incorporate this into our test suite.

@jck tangential but relevant, is there an automate method to list the number of tests that are skipped on the myhdl repo landing page?  I realize the `py.test` summary indicates the number test skipped but is there a method for travis-ci to include the number skipped in the myhdl travis-ci banner?

jmgc, 2015-05-27:
@cfelton, thanks for the effort.

Just a small comment, it would be good to remove the line:

```python
intf1.sig3.next = concat(intf1.sig3[7:1], sdi)
```

It creates a close loop inside the always_comb.

cfelton, 2015-05-28:
@jmgc good catch and thanks.  That's what I get for trying to do this over lunch :)

commit 7fea7bdd (2015-06-28): avoid renaming attribute references twice

This commit fixes bugs introduced by 08519b4
Fixes #82
closed 2015-06-28 by jandecaluwe