myhdl/myhdl #94: Release 0.9

jck, 2015-07-03:
There are no outstanding bugs related to interface conversion, and the documentation has been updated. Can we release 0.9?

cfelton, 2015-07-03:
+1 for 0.9 release.

cfelton, 2015-07-03:
Doing a quick review of the issues, the following should be discussed if they should be fixed for 0.9:
#2, #59, #73 (questionable ones #24, #25, #70).

jandecaluwe, 2015-07-05:
Ok, @all thanks for the hard work. I think quality/features have improved significantly.

I propose to do the release on what we have now, and transfer all remaining issues/PR beyond the release (as maintenance fix or new feature).

However, I still need to do the following:
- add documentation for ConcatSignal() interface enhancement
- add documentation for the toVHDL.std_logic_ports feature recently introduced

I'll do that shortly.

Also, should I consider to number the release as 0.9.0, more in line with conventions? With the infrastructure we have now, it may be easier to minor releases on a more regular basis.

Objections, feedback?

cfelton, 2015-07-05:
@jandecaluwe sounds good to me, agree with 0.9.0 revision numbering.  If needed the more frequent minor releases sounds good.

Also agree, we need to have all the documentation before a release.  Since we have the readthedocs pointing to the latest github we can (and should) always be updating the extend and quality of the documentation (I will try and heed my own advice :)

cfelton, 2015-07-05:
One of the things I have neglected was moving the `fixbv` PR from bitbucket to github.  Quite awhile ago I created a PR for the `fixbv` on bitbucket but we moved before any comments etc. on the PR were generated.  I manually moved the `fixbv` to github but was lazy and never created a PR on github.

There are a couple individuals using the `fixbv` but they have to use my fork.  It would be nice to have a conversation on the `fixbv` and determine if we want it (`fixbv`) based on the current design and a target release.

cfelton, 2015-07-06:
It looks like I need to retract the "readthedocs points to the master branch" on github (I was either mistaken, the webpage was updated, or the webpage was downdated).

jck, 2015-07-06:
I prefer to use readthedocs's default method[1] of managing versions rather than the way myhdl's rtd is setup now. The 'latest' version points to git master, and 'stable' automatically points to the latest release based on git tags (it would be 0.8.1 for us).

For example: http://pip.readthedocs.org/en/stable/ and http://pip.readthedocs.org/en/latest/

https://docs.readthedocs.org/en/latest/versions.html

jck, 2015-07-06:
BTW, we also need to figure out what the next major version is going to be(1.0 vs 0.10)

jandecaluwe, 2015-07-07:
I restored the rtd setup to default. However, I changed the pointers on the main website and in the README file to stable. I think that makes sense.

I propose to simply move to 1.0 next.

jck, 2015-07-13:
@jandecaluwe It looks like 0.9 is not yet pip installable. You need to upload the sdist to pypi.

jandecaluwe, 2015-07-13:
@jck Right, I forgot the upload command. Just fixed that. Thx.

closed 2015-07-13 by jck