myhdl/myhdl #95: [enhancement] toVHDL name collision checks

cfelton, 2015-07-07:
Users often run into the scenario when the converted VHDL is invalid because of name collisions.  This typically occurs because VHDL is not case sensitive and Python is.  The toVHDL converter should produce warnings for signal, variable, and port names that conflict and/or are invalid (e.g. dash in the name '-' or starting with an underscore '_').

Example of an incorrect conversion:

```python
def pipe(xi, xo, clock, reset, N=8):

    buffer = [Signal(xi.val) for _ in range(N)]

    @always_seq(clock.posedge, reset=reset)
    def rtl():
        buffer[0].next = xi
        for n in range(1, N):
            buffer[n].next = buffer[n-1]
        xo.next = buffer[N-1]

    return rtl
```

[complete code example](https://gist.github.com/cfelton/cddeebe2f96b00b53ed5)

hgomersall, 2018-09-07:
Should this include reserved words? Or is that a different bug?

cfelton, 2018-09-07:
In this original enhancement suggestion I included reserved words, they could be separated if they are easier to resolve one by one.  This might have been addressed in #160.