myhdl/myhdl #96: Using parameters for delays doesn't work with simulator vsim

d-roma, 2015-07-09:
I'm new at myhdl so most probably I'm doing something wrong but from the rs232_tx example I think what I'm trying should be possible to do. 

I have defined a constant with an integer value. Then, I use this constant for the delay function. 

```python
HALF_PERIOD = 5

@instance
def clockGen():
    yield delay(HALF_PERIOD)
    ...
```

Everything seems to work fine with the testbench in python, but when I try to test the vhdl, using the vcom simulator (documentation says vsim but in the _verify module it's defined as vcom) I get the following error:

> ** Error: testResetValues.vhd(53): near "ns": expecting ';' 

The reasons is that the resulting vhdl has the following code:

```vhdl
constant HALF_PERIOD: integer := 5;
...
    wait for HALF_PERIOD ns;
```
As far as I known this is not supported for vhdl. Either the constant should be of type time instead of integer and the time unit inserted at the definition or the value should be directly replaced after the wait for.

cfelton, 2015-07-09:
@d-roma Thanks for reporting this, I think this is a known issue but hasn't been recorded anywhere. 

Here is a past mailing-list thread that discusses a fix: http://thread.gmane.org/gmane.comp.python.myhdl/3863/focus=3865

d-roma, 2015-07-09:
You're righ! As reported by http://thread.gmane.org/gmane.comp.python.myhdl/3863/focus=3865
changing line 1151 of _toVHDL.py from 

```python
            self.write(" ns")
```

to 

```python
            self.write("*1.0 ns")
```

Solved the issue for me. Thanks! Perhaps it would be good to commit this change, I don't can see any case where this could get in conflict with anything else.

jandecaluwe, 2015-07-10:
Is there a reason why it has to be 1.0 ns instead of just 1 ns?

d-roma, 2015-07-10:
No, no one, it also works fine with just 1 ns. Even if at some point the multiplying constant changes from integer type to real, it will still work using 1 ns. So, for vcom it seems completely indifferent using just 1 or 1.0. But probably using just 1 ns will ensure being compatible with every simulator, since the VHDL standard always uses integer values for physical types.

cfelton, 2015-07-10:
@jandecaluwe, @d-roma I was working off @josyb suggestion in the original mailing-list thread.  We want to use "1' instead of "1.0" based on the integer is the common value for VHDL physical types?

cfelton, 2015-07-10:
Updated the PR #97 to use "1".  

Separate issue but I imagine "ns" should actually come from the `timescale` attribute?

jandecaluwe, 2015-07-12:
Using timescale attribute - nice feature for 1.0

closed 2015-07-12 by jandecaluwe