myhdl/myhdl #98: TristateSignal Conversion Error and Verilog missing inout

cfelton, 2015-07-09:
When `TristateSignal`s are used as top-level ports the signal is not converted to an `inout` when the target is Verilog.  This appears to work in VHDL but not in Verilog. 

In addition PR #74 appeared to break [code that use to convert](https://gist.github.com/cfelton/6119313).  I used `bisect` to verify the revision where the code stopped converting f965132.

Failing code snip:
```python
def top(sda, scl, sda_i, sda_o, scl_i, scl_o):
     sda_d, scl_d = sda.driver(), scl.driver()
     @always_comb
     def hdl():
         sda_i.next = sda
         sda_d.next = False if not sda_o else None 
         scl_i.next = scl
         scl_d.next = False if not scl_o else None
     return hdl
```

It is not clear (at this point) why the conversion is failing.  The conversion for the code snip only fails for VHDL conversion.

josyb, 2015-07-11:
It will convert (for VHDL) if you change the ternary construct:
 ```python
       sda_d.next = False if not sda_o else None
```
into:
```python
        if sda_o:
            sda_d.next = None
        else:
            sda_d.next = False
```

josyb, 2015-07-11:
It looks like the AST traversal errs:
This construct breaks at encountering the 'None' 
```python
sda_d.next = False if not sda_o else None
```
```text
line: sda_d <= '0' when (not bool(sda_o)) else stdl( 
```
and if we swap places:
```python
sda_d.next = None if sda_o else 0
```
```text
line: sda_d <= stdl(
```
Note that here the 'stdl(' is wrong too, as the expected first object in  the eventual ' a when b else c' statement is a 'Z'.
#74 corrected (tried to) the invalid (others => 'Z') when the RHS had a width of 1 (bool or intbv[1:]). If we have an ternary construct, this width information is missing.

jmgc, 2015-07-11:
Dear @josyb,

I think this will be useful inside procedures, but not in the concurrent part. That's the reason why VHDL2008 would be the quickest solution, but it is not available right now.

Best

josyb, 2015-07-11:
@jmgc, dear Jose,

actually the ' a <= b when c else d ;' is a valid concurrent statement in VHDL ( a conditional signal assignment statement). MyHDL will convert the original code into a series of concurrent statements.
The issue here is that MyHDL has a problem converting the python 'b if c else d' construct.

Regards,

Josy

jmgc, 2015-07-11:
I think that we are saying the same with different words, or I am missing something.

Best,

Jose M.

jmgc, 2015-07-11:
Dear Josy,

I have revisited the issue, and I do not think the problem is the if, else, but the way None is treated in toVHDL. The size information is lost, as you said previously, but it is clearer in this case, because we do not use None for algebraic operations.

In this sense, I think the Tristate has to be revised, at least in toVHDL.

Does it make sense to you?

Jose M.

josyb, 2015-07-11:
Dear Jose,

The issue is not 'None', because if you make it a *sequential statement*, as I said in my first answer, MyHDL properly recognises the 'None' value and converts that into a *'Z'* for std_logic targets, and into  to *(others => 'Z')* for std_logic_vectors. So I feel the conversion of python ternary construct has to be inspected.

Regards,

Josy

cfelton, 2015-07-11:
Not sure if it is simply the ternary, haven't inspected the conversion code yet but recently one of the students had to change the following to a ternary for the conversion `toVerilog` to succeed.  In otherwords the conditional failed with a None and the ternary worked with a None for Verilog target.

```python
if sDataDir_r == OUTPUT_C :
    sd_intf.driver.next = sData_r
else :
    sd_intf.driver.next = None
```

There seems to be something funky going on, not sure what.

jmgc, 2015-07-11:
Dear Josy,

Let me rephrase it. When a Name Constants like True or False are used, everybody assumes that it is just one bit that is going to be modified. But, in case of None, it is used for both one bit and a vector of bits. As a result, it is not possible to use the same functions for True, False and None, but we are doing it, ending up with errors.

Moreover, you can not change one bit to None in a vector, it's an all or nothing behaviour.

Also, internal functions in toVHDL, like inferCast or inferBinaryOpCast, take into account the case of True and False, but not the None.

I do not say that the ternary operator does not need a revision, although it will require a thorough revision on how and when procedures are created. But I am quite sure that the None is a singular case, that requires a revision on its own.

Best

jmgc, 2015-07-11:
Dear all,

I have modified the code and test to solve both aspects, Tristate signals and ternary operators, making use of your suggestions. It will take a pair of days to upload it, as I have to solve my git problem.

Best

josyb, 2015-07-11:
A ternary expression like:
```python
        tt.next = sl_o if slv_en else sl
```
will convert OK.
Only when 'None' is used, it will not convert.
```python
        sl_d.next = False if not sl_o else None
```
The python ternary statement is categorized as 'IfExp'. _toVHDL and _toverilog have a function to handle this 'IfExp'. And, as Jose points out, the inferCast(), in _toVHDL, has a problem here, as it provokes an unnecessary 'stdl()' procedure. But also the 'None' doesn't get recognised and raises the exception.
In so far Chris may be right that #74 broke the conversion.  
I undid that change locally, and then it converts but the conversion is no good:
```VHDL
		sl_d <= '0' when (not bool(sl_o)) else stdl((others => 'Z'));
```
as we expect:
```VHDL
		sl_d <= '0' when (not bool(sl_o)) else 'Z';
```
And then also the other construct:
```python
        if sl_o:
            sl_d.next = None
        else:
            sl_d.next = False
```
will produce illegal VHDL code:
```VHDL
		if bool(sl_o) then
			sl_d <= (others => 'Z');  -- should be <= 'Z';
		else
			sl_d <= '0';
		end if;
```

Chris, I don't see a problem in the converted Verilog code (I'm not a Verilog expert though)
```Verilog
assign tt = slv_en ? sl_o : sl;
assign sl_d = (!sl_o) ? 1'b0 : 'bz;
```

josyb, 2015-07-11:
To some point this discussion is futile, as Jose points out the ternary VHDL construct only works inside a process starting with VHDL-2008. And vendor support for VHDL 2008 is not that great, e.g. Altera's Quartus II doesn't support the ternary statement inside a process. I doubt it Xilinx' ISE will.
Makes me wonder why I spent so much time on this :(
But at some point this needs fixing to please the ASIC guys? Who use Verilog most if the time, and the conversion to Verilog seems to work fine.

jmgc, 2015-07-11:
Dear Josy,

I have added support for not concurrent on VHDL93. Only I need to upload it to GitHub, and of course pass the quality check.

Best

josyb, 2015-07-11:
Jose,

what you propose in your PR #100 seems very much fine. I wish I studied your code changes before stepping into this discussion  :(
I too once had a similar problem with git because I pulled other commits in my branch etc. I then deleted both my github and local repositories and started all over, taken care to not pull anything into the branch where I am working.
I made a small drawing in March:
[image]
Maybe that helps a bit?

Regards,

Josy

jmgc, 2015-07-11:
I will take a look.

Thanks for your help,

Jose M.

jandecaluwe, 2015-07-12:
As I felt we were drifting away I have looked at this and pushed a commit that tries to resolve the issue. This is fragile as we try to fix something for which we don't have VHDL support yet, and some related fixes in the past didn't have a test, so that things that used to work may break again (in which case they should now be added as a test first.)

Anyway, I'll wait for 2 days are so and if no urgent new things pop up, finally do the 0.9.0 release.

jandecaluwe, 2015-07-16:
Addressed in 0.9.0, better solution proposed in #107.

closed 2015-07-16 by jandecaluwe