myhdl/myhdl #105: [1.0 feature] Initial values

jandecaluwe, 2015-07-16:
**Issue** Enable initial values for signals and variables

**Task** It's time to enable initial value support. Use `no_initial_values` as a parameter to `toVHDL()` and `toVerilog()` to disable/enable this. Default should be enabled.

**Guideline** This is best done in atomic well-defined parts. Start with the simple case of non-composite VHDL signalls & Verilog regs.

**See also**  http://dev.myhdl.org/tasks/initial-values-support.html

**Complexity estimation** Medium

Famguy, 2015-10-28:
Hello! I would like to work on this. Could anyone please tell me more about it?

cfelton, 2015-10-28:
@Famguy have you reviewed the [see also](http://dev.myhdl.org/tasks/initial-values-support.html) link above?

Famguy, 2015-10-28:
Yes @cfelton I did go through it. From what I gathered from the feature description, we need to add a support for initial signal values in the Verilog/VHDL scripts generated (for eg: the corresponding Verilog syntax being, `input d = 4'b1010`). This would involve modifying the `_toVerilog.py` and `_toVHDL.py` files to set the initial values for all inputs/wires/reg to a default value (0 maybe) or user defined ones in the generated scripts. Please do correct me if I'm wrong.

cfelton, 2015-10-29:
@Famguy yes, that is the general approach.  You can look at [this](https://github.com/jandecaluwe/myhdl/compare/master...cfelton:initial_value_support) to see the minimal changes to get started. 

One of the first things you want to do is create a test to verify initial values.  You should be able to user the [convertible testbench](http://docs.myhdl.org/en/stable/whatsnew/0.6.html#conversion-output-verification) approach (tests in conversion/general).  Once you have a test you can start working on the implementation.  These two links should help with the test creation: [contributing ](http://dev.myhdl.org/guide/guide.html) and [writing tests](http://dev.myhdl.org/guide/guide_tests.html).

Once you have a test we can discuss the specifics of the initial values, if needed.

cfelton, 2015-10-29:
@Famguy also see this issue, comments on contributing for 1.0 https://github.com/jandecaluwe/myhdl/issues/110

minghuadev, 2017-07-05:
The original "see also" link is broken. And I don't see how this feature would work. Besides, looked at cfelton's example page on bitbucket. 

The question on stackoverflow https://stackoverflow.com/questions/44922574/python-myhdl-package-how-to-generate-verilog-intial-block .

The version I'm using is myhdl 0.9.0. Will the feature go into 1.0 only?

minghuadev, 2017-07-05:
I see the last commit under #150 with hash 87784ad is not on 0.9.0 branch. Will wait till the release of 1.0 rolls into the pip repository. Or is there a way to update manually with pip command?

minghuadev, 2017-07-05:
Ok, figured out to install it with `pip install -e <path-to-git-clone>`. And did two more changes to the source code to get it work: [1] put the initial signal into a list, and [2] set `toVerilog.initial_values=True` before calling `toVerilog`. Thanks for the great work on myhdl!