myhdl/myhdl #106: [1.0 feature] General support for indexed constants

jandecaluwe, 2015-07-16:
**Info** http://dev.myhdl.org/tasks/indexed-constants.html

**Complexity estimation** This is probably a hard task. It has multiple facets, as it involves knowledge about VHDL/Verilog synthesis and access to synthesis tools. The task may depend on prior improvements of the conversion code to be done right. *Don't let this be your first task*, be sure you are familiar with the conversion code first.

josyb, 2015-07-17:
For VHDL this could be implemented without the _function call_ as described in http://dev.myhdl.org/tasks/indexed-constants.html.
A construct like this:
```VHDL
	type a_integer is array( 0 to 23) of integer;
	constant rom : a_integer := (  4, 5, 6, 7, 0, 1, 2, 3,
									2, 3, 0, 1, 6, 7, 4, 5,
									1, 0, 3, 2, 5, 4, 7, 6) ;
. . .
	output_siglist(n) <= input_siglist( rom( romaddress ));

```
is a perfect replacement for the currently generated _case_ statement. It will work for both static and dynamic indexing.
I have only verified this with Quartus II 15.0, but I put in a _Mort Subite_ that ISE and Vivado will happily accept the same (after all, it is valid VHDL).

josyb, 2015-08-01:
Update: the above construct will not infer a _formal_ ROM. It will be synthesised as a MUX-lookup table.
To use (B)RAM as ROM the _existing case construct_ is still necessary as this gets recognised by the vendor synthesis tools. On note: Altera needs either a registered address input or a registered output. I have no idea if that would be recognised as a ROM by the other vendors.

josyb, 2015-09-03:
I have some experimental code that simulates and converts (both VHDL and Verilog)
```python
# put the coefficients in a 2D array
    COEFF = [[ hdlutils.tofixedpoint(v, 9, SCALING, 'Signed') for v in [   0,  0.299,      0.587,     0.114   ]], # y
             [ hdlutils.tofixedpoint(v, 9, SCALING, 'Signed') for v in [ 128,  0.5,       -0.418688, -0.081312]], # cb
             [ hdlutils.tofixedpoint(v, 9, SCALING, 'Signed') for v in [ 128, -0.168736,  -0.331264,  0.5     ]], # cr
            ]
    cc.append( calc(9+SCALING, SCALING, COEFF[0], Clk, pl_Enables, bq.red, bq.green, bq.blue, ycrcb.y))
    if DO_CrCb:
        cc.append( calc(9+SCALING, SCALING, COEFF[1], Clk, pl_Enables, bq.red, bq.green, bq.blue , ycrcb.cr))
        cc.append( calc(9+SCALING, SCALING, COEFF[2], Clk, pl_Enables, bq.red, bq.green, bq.blue , ycrcb.cb))
```
In the converted code we can find these constants as integers:
```VHDL
ST_BayerQuad2YCrCb_cc_6_calc: process (Clk) is
begin
    if rising_edge(Clk) then
        if bool(pl_Enables(0)) then
            cc_6_r1c1 <= signed(resize(resize(cc_6_r1, 19) * (-173), 29));
            cc_6_g1c2 <= signed(resize(resize(cc_6_g1, 19) * (-339), 29));
            cc_6_b1c3 <= signed(resize(resize(cc_6_b1, 19) * (512), 29));
        end if;
```
Would this be something to proceed with?

jandecaluwe, 2015-09-04:
Note that this issue is not about valid VHDL or synthesizablity - those are necessary conditions. The purpose is to add generality in modeling without giving up the synthesis optimizations available now, including using specialized ROM structures which I must assume are more efficient than plain logic.

If VHDL constant tables don't cut it, one possibility may be to keep the case statement but wrap it into a function. That would work for Verilog also.

On the other hand, if the ROM inference feature requires a very specific template, we might be forced to implement a general solution in the most elegant way for the general case, and perform dedicated inlining of the case statement for a specific template. That would mean detecting the always block that can be a candidate for ROM inferencing.

josyb, 2015-09-04:
The above example adds indexed constants with fixed indexing.
For dynamic indexing we keep the existing tuple-construct. The tuple-construct only allows one dimension, but that is a prerequisite to infer a ROM anyway. I don't think that encapsulating a case-statement inside a function will work as it won't conform to the manufacturers' required template?

In my experimental [myhdl.Array()](https://github.com/josyb/myhdl/tree/m1D) code we could also support generally indexed constants. Referring to #60 the code would look like this:
```python
def myhdl_b3(in1, s, out1):

    t = Array( [2,5,9,14], intbv(0)[W:])
    
    @always_comb
    def rtl():
        out1.next = in1 + t[s]
    
    return rtl
```
and _would_ then generate this VHDL:
```VHDL
entity myhdl_b3 is
	port(
		in1  : in  unsigned(3 downto 0);
		s    : in  unsigned(1 downto 0);
		out1 : out unsigned(3 downto 0)
	);
end entity myhdl_b3;

architecture MyHDL of myhdl_b3 is
	type a4_u4 is array(0 to 3) of unsigned(3 downto 0);
	constant t : a4_u4 := ("0010", "0101", "1001", "1110");

begin
	out1 <= (in1 + t(to_integer(s)));

end architecture MyHDL;
```

I haven't explored this in the _toVerilog code (yet).

josyb, 2015-09-21:
I studied how to Verilog would handle this and apparently Verilog doesn't do array constants. IMnsHO sticking to convert to Verilog and VHDL on arm's length is hindering progress of MyHDL. See also #124
where I (try to) make a case to adopt SystemVerilog as an upgrade to Verilog. Note that the Verilog standard and the SystemVerilog standard have merged into one in 2009 (almost seven years ago!).

jandecaluwe, 2015-09-21:
If a useful feature is broadly supported by design tools, including free tools and at least one tool that can be used in our travis test infrastructure, there is no reason not to make use of it. It is purely a practical matter.

josyb, 2015-09-22:
SystemVerilog is well supported by Altera and Xilinx, directly in their synthesisers, Lattice relies on Synplify Pro.
For simulation: the FOSS Icarus Verilog simulator seems to be extending SystemVerilog support: http://iverilog.wikia.com/wiki/Release_Notes_Icarus_Verilog_10. Both Altera and Xlilinx support SystemVerilog, Altera with a free ModelSim Mixed-language simulator, Xilinx with its own Vivado Simulator.
Of course VHDL is well covered with GHDL, and the _included_ simulators from Altera and Xilinx.

Making the transition to support SystemVerilog, is as you say not a straightforward task, again as you say, we have something like  n * m  + p * r cases to test. Looks like I'll have to invest in a Linux box after all.
Is there someone I can team up with? Or shall I start submitting incremental PRs? Or do you want to issue a MEP first?

jandecaluwe, 2015-09-26:
I you mean a MEP for the transition to SystemVerilog, no that's not what I have in mind. I envisage an organic transition based on features, and that actually should be straightforward. The ideas is to make use of the fact that SystemVerilog has promised to be compatible evolution of Verilog.

For example, we could start with the indexed constants features as follows:
1. Investigate what tool switches are required to support them
2. Apply them to our test environment
3. Check if test suite still passes (and expect it to do so)
4. Implement indexed constant support

josyb, 2015-09-27:
I too prefer the organic transition.
At the moment we only need to _upgrade_ Verilog to SystemVerilog for the _new Array_ type, and the _indexable Constants_ ( which either are an _Array_  or a plain multidimensional list of intbv or integer). Another future expansion like the conversion of records/structures would also mandate SystemVerilog.

1. a. One way is to detect use of these constructs and then write the output to a _.sv_ file and issue an _information message_ about this.   
  b. The other way is to augment toVerilog's _standard_ attribute and then issue a _warning / error_ message when a construct is not convertible to that standard.  
  ```assert self.standard in ['1995',  '2001', '2005', 'SV2005', 'SV2009', 'SV2012']```  
I don't think we need  all choices for the time being.  
Note that the standard attribute isn't actually used  in the_toVerilog code. We will have to add at least one more attribute to specify packed or non-packed arrays. 
You can see this all in my experimental _m1D_ branch

jandecaluwe, 2015-09-27:
Why  would we need to support multiple standard versions? Can't we simply require people to upgrade to a more recent tool version (or set certain switches)?

josyb, 2015-09-27:
If you say _switches_ are you referring to _command line switches_ or to the likes of _\_toVHLD.std_logic_ports_? 
I found the _.standard_ attribute in \_toVerilog and got overboard with adding versions, I admit, but I already hinted the choices could/should be reduced. We probably best restrict ourselves to SV2005. In the _m1D_ branch I only test if the target standard is greater then or equal to 'SV2005' to find out whether _packed arrays_ are preferred. 
I have no good idea whether forcing people to switch to SytemVerilog, by producing only .sv output only, is acceptable. That's why I also proposed 1.a. where we constrain the output to SystemVerilog if we detect features which are not supported in Verilog 2001. But unfortunately things may be not so clear cut as some vendor may say it supports 2001, which in theory supports multidimensional arrays, but restrict this to the (basic) memory declaration like: ```reg [7:0] mema[0:255];```. And I don't know enough of SystemVerilog to make the right choice so I tend to go for 1.b.   
Input from someone versed in SystemVerilog will be very welcome.

jandecaluwe, 2015-09-28:
I meant command line switches.

Iff the only change would be to change a Verilog file extension, I don't think there is a problem with that (under the all-important assumption that this is broadly supported by free tools, including our test environment.) Of course, not in 1.0 maintenance. Otherwise, we have to review everything as it would mean SystemVerilog is not really a superset of Verilog.

FInally, I propose to restrict ourselves to the well understood and agreed-upon indexed constant feature as a proof of concept.

josyb, 2015-09-28:
I propose to do like this:
1. Augment the _.standard_ attribute to give a choice between ```'2001'``` and ```'SV2005'```. This will give us the opportunity to run (some or all) the standard Verilog tests as SystemVerilog files too. This will show us how the simulators (vlog and iverilog) cope with the name change. And we then have validated the test environment.
2. A _limited_ indexed constants feature as proof of concept. This will be a temporary branch _IndexedConstants_ (in my forked repository?)

jandecaluwe, 2015-10-03:
I would do it as follows:

1. Write a small test case in SystemVerilog that uses indexed constants in a representative way
2. Check that command switches required to make it run with icarus (and ideally vlog), as well as file extension requirements if any.
3. Apply those same changes to our test environment / file extensions, without any other change.
4. Check that the test suite runs.
5. If if does, apply those changes in development, warn users through communication channels, update documentation.
6. If does not, do nothing as yet and revisit the issues.
7. After migration, implement indexed constants in conversion for Verilog/VHDL.

In this way, we can migrate without having to support multiple standards, which conceptually should not be necessary.

rollingstone, 2019-01-03:
> The above example adds indexed constants with fixed indexing.
> For ROM inferral we keep the existing tuple-construct. The tuple-construct only allows one dimension, but that is a prerequisite to infer a ROM anyway. I don't think that encapsulating a case-statement inside a function will work as it won't conform to the manufacturers' required template?
> 
> In my experimental [myhdl.Array()](https://github.com/josyb/myhdl/tree/m1D) code we could also support dynamically indexed constants. Referring to #60 the code would look like this:
> 
> ```python
> def myhdl_b3(in1, s, out1):
> 
>     t = Array( [2,5,9,14], intbv(0)[W:])
> 
>     @always_comb
>     def rtl():
>         out1.next = in1 + t[s]
> 
>     return rtl
> ```
> 
> and _would_ then generate this VHDL:
> 
> ```vhdl
> entity myhdl_b3 is
>     port(
>         in1  : in  unsigned(3 downto 0);
>         s    : in  unsigned(1 downto 0);
>         out1 : out unsigned(3 downto 0)
>     );
> end entity myhdl_b3;
> 
> architecture MyHDL of myhdl_b3 is
>     type a4_u4 is array(0 to 3) of unsigned(3 downto 0);
>     constant t : a4_u4 := ("0010", "0101", "1001", "1110");
> 
> begin
>     out1 <= (in1 + t(to_integer(s)));
> 
> end architecture MyHDL;
> ```
> 
> I haven't explored this in the _toVerilog code (yet).

Hi,

myhdl.Array seems to be removed from package. could you please share the code of Array. Thanks

josyb, 2019-01-03:
myhdl.Array is experimental code only and doesn't form part of the official releases.

rollingstone, 2019-01-03:
> myhdl.Array is experimental code only and doesn't form part of the official releases.

Thanks. Would it be possible to share the unofficial version of the code? it would be very helpful. I am just getting into MyHDL. The link that you shared is not working anymore. Thanks

josyb, 2019-01-03:
The description has moved to http://dev.myhdl.org/resources/tasks/indexed-constants.html.

The unofficial code has diverged too much from the official release.

rollingstone, 2019-01-03:
@josyb Thanks