myhdl/myhdl #107: [1.0 feature] Better support of ternary operator

jandecaluwe, 2015-07-16:
**Task** Support Python's ternary operator in a better way in VHDL.

**Description** Verilog has a direct equivalent for the ternary operator, but for VHDL it depends on the version and the context. As a consequence, current VHDL support is fragile and unverfiable with the current test infrastructure. In this task, investigate whether it is possible to support the ternary operator using a function, so that it works for all VHDL versions in all contexts. The idea would be to put an overloaded function in the `pck_myhdl` package, and used that to compile `IfExp` nodes.

**Complexity estimation** Medium

josyb, 2015-07-17:
I have written several overloaded _ternary()_ functions, and use (used as it has been a while since I've written new VHDL code) them about everywhere, often even nested. I've thrown together bits and pieces from my (legacy) code base:
```VHDL

...
entity showternaryusage is
	port (
		...
		PixInfo        : in  std_logic_vector(ternary(WIDTH_PIXINFO > 0, WIDTH_PIXINFO - 1, 0) 
downto 0);
		
          );
end entity;
architecture a of showternaryusage is

	constant ALIGNMENT_PATTERN : std_logic_vector(WIDTH_DSENSOR - 1 downto 0) := ternary(WIDTH_DSENSOR = 10, "0000011111", "000000111111");
	constant WIDTH_BURST_EXTENDED : natural := ternary( widthr(MAX_BURST) < 9 , 
														8 , 
														ternary( widthr(MAX_BURST) < 13 ,
															12 ,
															16
															)) ;	

	signal dpgpixinfo         : std_logic_vector(ternary(WIDTH_PIXINFO > 0, WIDTH_PIXINFO - 1, 0) downto 0) := (others => '0') ;

begin

	sel		<=	ternary( (LdStart = '0') and (StepEn = '1') , 1 , 0) +
				ternary( (LdStart = '0') and (StepEn = '0') and (IncEn = '1') , 2 , 0) 
				;
	process( xvals , ycounter_Q , dsop)
		begin
			for i in 0 to NUM_SYMBOLS - 1 loop
				if (i = 0) then
					ythenxvals(0) <= ternary(dsop = '1' ,  std_logic_vector( to_unsigned( ycounter_Q , WIDTH_SYMBOL)) , xvals(0)) ;
				else
					ythenxvals(i) <= xvals(i) ;
				end if ;
			end loop ;
			alignmentdata := ternary( ALIGN_ON_SYNC_CHANNEL = "YES" , ddr_combine(srsynch , srsyncl) , ddr_combine(srdh(0) , srdl(0)) );

		end process ;

	addrcntr : upcounter2
		generic map (
			COUNT_MAX	=>	MAX_ADDRESS ,
			ONE_SHOT	=>	ternary( WRAP_AROUND = "NO" , "YES" , "NO" )
			) 
		port map (
			...
		) ;

sm : entity work.packdownsm
		generic map(
			WIDTH_D      => WIDTH_D,
		...
		)
		port map(
			Clk         => Clk,
			...
			DSoF		=> ternary(INCLUDE_SOF = 1 , D(INCLUDE_SOF + WIDTH_D - 1) , '0') ,  -- tricky ...
			...
		);
end a;
```				
So it is definitely possible to use such a _ternary_ function in every VHDL context.

jandecaluwe, 2015-07-18:
I have pushed a proposed implementation in the master branch for review and further testing.

josyb, 2015-07-18:
I would have preferred the C-style (condition, if_true, if_false) over the Python/VHDL form, because _nesting_ comes out clearer, but the harm is already done in the Python code and the produced VHDL code only serves its purpose ... 
But I wonder why you enhanced the
 ```python
s = (others => 'Z')
```
 in _getname()_ into  
 ```python
assert hasattr(node.vhd, 'size')   
s = (%d downto 0 => 'Z')" % (node.vhd.size-1)
```

josyb, 2015-07-18:
While updating my local MyHDL development tree I found my code to differ a bit:
```python
     elif n == 'None':
            if node.vhd.size == 1:
                s = "'Z'"
            else:
                s = "(others => 'Z')"
```
with yours:
```python
       elif n == 'None':
            if isinstance(node.vhd, vhd_std_logic):
                s = "'Z'"
            else:
                assert hasattr(node.vhd, 'size')
                s = "(%d downto 0 => 'Z')" % (node.vhd.size-1)
```
So I decided to test the differfence with @cfelton's code from #98 (enhanced/modified to test tristating a std_logic_vector too) . It made it clear to me that the ```(others => 'Z')```  would be an invalid argument for the ```tern_op()``` call.  So that's the answer.
But the ```(n downto 0 => 'Z')``` is flagged by ModelSim with this warning:
```text
Vcom: (vcom-1514) Range choice direction (downto) does not determine aggregate index range direction (to).
```
I managed to get rid of the warning by substituting a string of spelled out 'Z', ```"ZZ...Z"``` in place of the  ```(n downto 0 => 'Z')```.

I replaced
```python
 s = "(%d downto 0 => 'Z')" % (node.vhd.size-1)
```
with 
```python
s = "\"{:{fill}{align}{count}}\"".format('', count = node.vhd.size, fill = 'Z', align = '<')
```
or shorter:
```python
s = "\"{0:Z<{1}}\"".format('', node.vhd.size)
```
I then also verified that the changed code works for ```intbv()[1:]```

And now I understand that replacing ```if node.vhd.size == 1:``` with ```if isinstance(node.vhd, vhd_std_logic):``` specifically avoids a conflict with an ```intbv(()[1:]```.

jandecaluwe, 2015-07-18:
As this is in the development master branch, there is no problem in changing things. I like the suggestion about the ternary operator signature, and I implemented it.

jandecaluwe, 2015-10-10:
FIxed in 9820fca

closed 2015-10-10 by jandecaluwe