myhdl/myhdl #108: [1.0 feature] Fixed point

jandecaluwe, 2015-07-16:
**Task** Fixed point type support

**Description** A proposal for this is in http://dev.myhdl.org/meps/mep-111.html, and there is also a reference implementation. What is not clear to me (the BDFL) is how much of the resizing/alignment is automated. I think it is mandatory to automate as much as possible.

**Complexity estimation** Guru level + BDFL involvement. We have to carefully decide on and implement the type hierarchy, resizing & alignment issues, conversion support.

cfelton, 2015-07-17:
The original proposal envision little converter (compiler) support but required functions with elaboration.  I will try to explain the idea and it is worth discussing if this is the path we want or if we want the converter to do more.

Note, the following picks up where the [MEP111](http://dev.myhdl.org/meps/mep-111.html) and the [mailing-list conversations](http://search.gmane.org/?query=fixed-point+thoughts&author=&group=gmane.comp.python.myhdl&sort=relevance&DEFAULTOP=and&xP=Zfixbv&xFILTERS=Gcomp.python.myhdl---A) left off.  Those new to the conversation might want to review.

### Resizing with functions that can elaborate

The approach mentioned in MEP112 was to use a `resize` function to accomplish the alignment, rounding, and saturation.

Given the example:

```python 
    fx3 = resize(fx2 + resize(fx1, fx2), fx3)
```

In the above the `resize` function arguments are:

```
def resize(val, fmt, round_mode='convergent', overflow_mode='saturate'):
```

The resize function will resize `val` to match `fmt`'s format using the methods defined by `round_mode` and `overflow_mode`.  My current thought is this would require a convertible function that can elaborate, something like:

```python
def resize(w, x, y, z):
    # all the elaboration stuff
    @func_comb
    def_resize():
        # logic to perform the operation
    return _resize()
```

This would require the converter to determine if the function being converted contained an elaborated function and then only convert the elaborated function.

This is the original envisioned approached mentioned in MEP111.

### Resizing automajically

Another approach would be to have no resize function and have the converter do the work:

```python 
   fx3 = fx2 + fx1
```

The converter would need to do considerable more work.  The `fixbv` objects would contain the attributes to guide the resize.  For example, a `fixbv` object would have a set `round_mode`.  In the above example the rounding would be guided by `fx3`'s attributes.
   
For reference:
My first attempt at the [fixbv](https://github.com/cfelton/myhdl/blob/mep111_fixbv/myhdl/_fixbv.py) and [tests](https://github.com/cfelton/myhdl/blob/mep111_fixbv/myhdl/test/core/test_fixbv.py)

jandecaluwe, 2015-07-19:
If I understand it well, with proposal 1, the user has to do resizings/alignment. While that is perfect as a workaround, I think users will eventually expect full automatic support like I think proposal 2 suggests.

I am not saying that that is easy or even feasible. But I think that is the road we have to explore. The way forward is probably to think first about the user interface for modeling, and *not* worry about conversion - what would be the ideal fixbv type from a user perspective look like?

cfelton, 2015-07-21:
@jandecaluwe fully automated path sounds great!  The current design (user interface and api) proposed in MEP111 is applicable to both paths.  With any design, we can toss it around until it is fully polished.

With the current proposal in MEP111, similar to `intbv`, the format of the `fixbv` is defined by the properties of the representation, in this case range and resolution.  The ability to define the format by bit-widths is also supported in the proposal.

#### Definition by properties:

```python
fx1 = fixbv(0, min=-8, max=8, res=.001)
```
There is a lot going in in this example but I am limiting the conversation to the user interface for now (lots of interesting stuff to discuss when it comes to fractional representation).  This type of domain constraints (vs. physical constraints) supports higher abstraction (but not too high).  Based on these properties the number of integer and fractional bits (format) are determined.

There are two additional arguments not shown: `round_mode` and `overflow_mode`.  

```python
fx1 = fixbv(0, min=-8, max=8, res=.001
            round_mode='convergent`,
            overflow_mode='saturate')
```

The *modes* define what happens when a value is assigned, this is true for the initial value and any subsequent assignment:

```python
    fx1.val = math.pi
    fx1.val = fx1.max + .1 
```

In the above two examples the assigned value will be rounded and the overflow handled by the *modes* defined during instantiation (note `.val` is used instead of `[:]`).

#### Definition by bit-width
Also proposed in MEP111 is the ability to define the format via the bit-widths:

```python
fx2 = fixbv(0)[16, 0, 15]
```

In the proposals I chose this explict definition of the format where the tuple represents (full bit width (WL), number of integer bits (IWL), number of fractional bits (FWL),) where WL = IWL + FWL + 1.

#### Operations
These just work :)

```python
fx1 = fixbv(math.pi, min=-8, max=8, res=0.001)
fx2 = fixbv(0, min=1, max=1, res=2**-8)
fx3 = fixbv(0, min=-100, max=100, res=.125)

fx1.val = (fx2 * fx3) + fx1
```

There are a bunch of details here why this would just work but those can be discussed.  From a user interface perspective this is straightforward.  This is not what is proposed in MEP111, will update.  This is straightforward for modeling and will be some work for conversion.

#### Access
In addition to defining the format and performing operations a user will want to access the integer and fractional portions, currently I have proposed these as methods

```python
integer_part = fx1.int()
fractional_part = fx1_frac()
```

Thinking about this using properties would be a better approach

```python
integer_part = fx1.int
factional_part = fx1.frac
```

And assigning them should also be possible

```python
    fx1.int = 1
    fx1.frac = .06125
```
Which also brings up the question, should `fx1.val` be used instead of `fx1[:]`, I think so, I think when the bits are accessed it is no different than accessing `intbv` bits, you are on your own - bits are formatless.

```
bv1 = intbv(0)[16:])
fx1 = fixbv(0, min=-8, max=8, res=2**-12)
# ...

# this should work (danger zone, loose context)
bv1[:] = fx1[:]
fx1[:] = bv1[:]

# whereas this shouldn't (necessarily)
bv1[:] = fx1   # fx1.val
fx1.val = bv   # depends on the value
```
And when in a `Signal`, `fx1.next`. 

#### Updating the MEP
I will update the MEP to reflect the fully automated path and add the details required for automated conversion, as well as feedback from this thread.

smldis, 2016-02-13:
Hi I walked through this quickly:
The fx1.val syntax seems weird to me:
```
fx1 = fixbv(math.pi, min=-8, max=8, res=0.001)
fx2 = fixbv(0, min=1, max=1, res=2**-8)
fx3 = fixbv(0, min=-100, max=100, res=.125)

fx1.val = (fx2 * fx3) + fx1
```

What about:
Creation of new var fx1 with explicit representation
`fx1= fixbv((fx2 * fx3) + fx1, min=-8, max=8, res=0.001)`

Creation of new var fx1 with implicit representation
`fx1 = (fx2 * fx3) + fx1`

Where resulting representation is defined by rounding and overflow policies of each single fixbv:
In case of:
```
fx1 = fixbv(math.pi, min=-8, max=8, res=0.001, 
            round_policy='convergent' ,overflow_policy='saturate')
fx2 = fixbv(0, min=1, max=1, res=2**-8)
fx3 = fixbv(0, min=-100, max=100, res=.125)
```

In this case `(fx2 * fx3)` is a default policy multiplication, then the result must be added to `fx1`. Now we have a a policy mismatch.
In this case we can pick the most restrictive policy and in this case should be saturate for overflow,  we use this policy if just one of the two arguments wants saturation.

Therefore we can either build a policy table that decides wich policies to pick in case of policy mismatch or we can force the user to set policies in order to avoid mismatches like matlab do:
Example for multiply: http://www.mathworks.com/help/fixedpoint/ref/mpy.html

Am I introducing a lot of implementation difficulties? Since other software already do this kind of things right we can look forward to an implementation that is supposed to be usefull just by comparing the different apis, no?

Let me know,
smldis