myhdl/myhdl #109: [1.0 feature] Support for `global` declaration

jandecaluwe, 2015-07-16:
**Task** Support Python's `global` declaration

**Description** To realize the vision behind MyHDL, support for the `global` declaration is mandatory. It makes it possible to assign to variables within generators and still let those variable keep global state. In HDL terms, it enables register inferencing from variables - the ultimate semantic power in synthesizable modeling. Unfortunately, the `global` declaration is only available in Python 3K

**Complexity estimation** Guru level for several reasons. One reason is the Python 3K dependency. Therefore, this depends on a plan for the eventual fork between Python 2K and Python 3K, diverging but at the same time keeping them in sync as long as possible.

josyb, 2015-07-17:
I'm definitely not a Python guru but I have seen ```global```  being used in Python 2.7.  I just checked https://docs.python.org/3.4/reference/simple_stmts.html: it has the same wordings for both 2.7 and 3.4 Or is it that using the ```global``` declaration inside a ```generator``` is not possible in 2.7?

jandecaluwe, 2015-07-17:
I knew  something was wrong but not where it was ... thanks for making me realize it's not `global` but `nonlocal` that I wanted to talk about ;-)

jck, 2015-07-18:
Can I see an example Python 3 code which demonstrates how nonlocals will be used?

jandecaluwe, 2015-07-19:
It's all still only in my head ;-)

For an introduction of the problem I would like to solve, see:

    http://www.myhdl.org/examples/jc2.html#alternative-design

At this point one cannot use the `always`decorator and support variables that also keep state (meaning they will be implemented as a register.) With `nonlocal` that would become possible.

This is important because experienced VHDL users use this semantic power all the time. It's available in Verilog also, but there guidelines stand in the way. In any case, it enables powerful modeling techniques that are synthesizable but only available in event-driven HDLs, such as MyHDL.

This is a vast subject, and I will have to do things in the right order and write a MEP about it.

As I was thinking again about this, another possibility would be to support interfaces with variable members, solving the "nonlocal" access by attribute access. The advantage of that approach would be that it would work for Python 2K also. Something to think about further.

jandecaluwe, 2015-07-21:
I rephrased it to keep all options open.

jck, 2015-07-21:
I think we can implement this as a py3k only feature without complicating the code base too much.

jck, 2015-07-22:
The good news is that nonlocals already work for modelling with Python 3 I think.
https://gist.github.com/jck/ab2a6a033393c8854ec4

jandecaluwe, 2015-07-23:
@jck Sure, the opposite would have been surprizing!

I found it interesting that the reasoning I am making coincides nicely with the original proposal for `nonlocal`: https://www.python.org/dev/peps/pep-3104/

When I look at your reworked code, I like the style. MyHDL has an advantage here over Verilog/VHDL that the variables that keep state (and become registers in synthesis) are clearly distinguished. (This is often a source of confusion to inexperienced HDL coders.) So I guess the best way forward is as a unique P3K feature. Perhaps it it still possible to support this (with appropriate messages fro Py2K) from a single sourcebase?

In any case I need to write a MEP (even if the implementation is available earlier ;-).