myhdl/myhdl #110: [1.0 fixes] How to fix issues for 1.0

jandecaluwe, 2015-07-17:
The goal is to minimize the open issues in the Issue Tracker.

* Pick an issue you would like to fix. Check if it really refers to an issue, and suggest how to deal with the issue otherwise.

* If it is a real issue, design test that should work but fails

* Add it to the `myhdl/test/bugs`. Use existing tests there as a template for the details.

* If you don't plan to fix it yourself, that is OK. Add a `pytest.xmark.fails` directive to indicate the expected failure, otherwise travis will fail on it. Raise a PR that reference the issue number, so people will see the test was added.

* Otherwise, fix the issue and raise a PR when completed, again referencing the issue number.

* Thanks!

josyb, 2015-07-27:
small typo: ```@pytest.mark.xfail```  and include ```import pytest``` in your example

jandecaluwe, 2015-07-27:
Yes, hope it's ok now.

ironmaniiith, 2016-03-04:
Guidelines link gives 404, you can change it to [http://dev.myhdl.org/guide/guide.html](http://dev.myhdl.org/guide/guide.html) in the description.

jandecaluwe, 2016-03-05:
Yes, thx.

ironmaniiith, 2016-03-05:
The project seems really interesting to me, @jandecaluwe can you give me an idea from where can I start contributing to it and what are the top most requirements for now. I would be happy to get involved with the project :smile:.
I have good knowledge of  python, vhdl and verilog, that would be sufficient I guess.?

jandecaluwe, 2016-03-07:
http://dev.myhdl.org/guide/guide.html