myhdl/myhdl #121: [RFC] enhancement verilog testbench without cosim

Thomasb81, 2015-08-17:
Hello

Problem:
I target a design for FPGA. I have know a couple of verilog module, build on a bottom-up approach.
I face strange behaviour when running the application :  sound synthesizer . And I was not able to find the issue with directed testbench. So I want convert my existing verilog code to myhdl. 
This will allow me drive the dut with more realistic case. To solve timing issue I make effort to pipe-line my design and manage to use FPGA primitive : 

```
def mul_reg(clk,a,b,p):
    a_q = Signal(intbv(min=0,max=2**16))
    b_q = Signal(intbv(min=0,max=2**16))

    @always(clk.posedge)
    def logic():
        a_q.next = a
        b_q.next = b
        p.next = a_q * b_q
    return logic

mul_reg.verilog_code = \
"""
MULT18X18SIO #(
   .AREG(1),
   .BREG(1),
   .PREG(1),
   .B_INPUT("DIRECT")
) mul_reg_inst (
.A({2'b00,$a}),
.B({2'b00,$b}),
.CEA(1'b1),
.CEB(1'b1),
.CEP(1'b1),
.CLK($clk),
.RSTA(1'b0),
.RSTB(1'b0),
.RSTP(1'b0),
.BCIN(18'h00000),
.BCOUT(),
.P($p)
);
"""
```

The primary objective is to validate the myhdl model against the vendor primitive. But the simulator vendor does not provide vpi interface for co-simulation.
I understand that that people having access to Mentor or Cadence simulator have capability to generate vendor primitive library will use vpi/cosim.

An initial attempt to work-arround the simulator vendor limitation is to generate VCD trace at testbench level, post-process the trace to create to rebuild a testbench. However the current solution is far to be automatic :  The dut still need to be instantiate. In the VCD, there is not anymore port direction so statement need to remove : the dut and the testbench should not drive the same wire ... So this task is quite boring and need to be perform each time I change my test-bench. 

Solution:
The VCD spec is part of verilog spec. Myhdl implement it's traceSignal feature by instrumenting the simulator in order to print the update signal value for all signal monitored into the VCD file.
Same idea can be apply to a verilog test bench. The tool could generate testbench file similar to the one create for the cosim when running the function `toVerilog()`. But instead of calling vpi function the initial begin / end statement, will have a calendar of value generated from the myhdl simulation : 
```
initial begin
  clk = 1'b1;
  #185
  clk = 1'b0;
  #190
  clk = 1'b1;
  #195
  clk = 1'b0;
  #200
  clk = 1'b1;
  a = 16'b100000000000000000;
  b = 16'b10000000000;
  result = 16'b1000000000;
  #205
  clk = 1'b0;
  a = 16'b100000000000;
  #210
  clk = 1'b1;
  result = 16'b10000000000;
  #215
  clk = 1'b0;
  b = 16'b1000000000000;
  #220
  clk = 1'b1;
  result = 16'b100000000000;
  #225
  clk = 1'b0;
  a = 16'b10000000000000;
  #230
  clk = 1'b1;
end
```
In details
A copy of traceSignal class can be use as a template.
In Signal class similar to printVCD a printVerilog need to be implemented.
Difficult part is how can to generate dut instantiation code...
I do not understand very well why traceSignal() and toVerilog() are exclusive function...

Thomas

cfelton, 2015-08-17:
You can convert testbenches but you will be limited to what the testbench can do:
http://docs.myhdl.org/en/stable/whatsnew/0.6.html#conversion-of-test-benches

In addition @hgomersall has a package that will capture the VCD from a myhdl simulation and use it to drive the Vivado simulator:
https://github.com/hgomersall/Veriutils

Thomasb81, 2015-08-17:
Well,
The initial intention was not to generate testbench. Just to have an easy way to do a play back. So limitation are acceptable. If checker is needed, It seem still possible to write it as verilog code.

About Veriutils, it probably does the job. Unfortunately it seems not easily affordable to me for the moment...

hgomersall, 2015-08-18:
I'd be very keen to work with you to get Veriutils to be useful here. I put quite a bit of effort in to solve exactly this problem, so
 i'm sure we can work through any problems.

Thomasb81, 2015-08-18:
Hi Henry,

Today i use ISE tool chain because my device are not supported in the new toolchain.
It is not clear for me what need to be change in your library to be compatible with ISE : template, command line, interface etc ...

Thomas

hgomersall, 2015-08-18:
The main stuff that would need doing is a tcl template for driving ISE (see the Vivado one [here](https://github.com/hgomersall/Veriutils/blob/master/vivado/templates/simulate.tcl.tmpl)). Currently all the templates are set up for Vivado, but that's just because that's what I've used. Vivado is hard coded as the vendor sim, but again, that was just an implementational short circuit because I had no need for others. It would be easy to make it user selectable (probably with a function argument or something, instead of `vivado_verilog_cosimulation` it would be something like `vendor_verilog_cosim(..., tool='vivado')` with a few mods to make that work.

I'm keen to use @cfelton's [gizflo](https://github.com/cfelton/gizflo) to replace my rudimentary interaction with Vivado. ISE would come for free then. To be clear, the hard or nice bit about Veriutils is not the interacting with other tools, but the auto generation of verification test benches and then doing that verification, along with a few neat utilities (random numbers, initialisation, clock etc).

closed 2021-12-01 by Thomasb81