myhdl/myhdl #122: Constant values inside recursive functions do not work in VHDL conversion

burnpanck, 2015-08-21:
Applies to MyHDL 0.9.
If a constant value is given as an argument that is then assigned to a signal in an enclosed process, conversion to VHDL will generate a VHDL constant for that variable. If this function is recursive, the same name should be bound to different values, but the generated VHDL code will use the same name with one single value for all generated processes.

Here's an MWE:
```
inp = Signal(intbv(0,0,2**8))
nz = Signal(False)
idx = Signal(intbv(0,0,8))

def toplevel(inp,nz,idx,n):
    if not n:
        @always_comb
        def p():
            nz.next = inp[0]
            idx.next = 0
        return p
    inz = Signal(False)
    iidx = Signal(intbv(0,0,n+1))
    ret = toplevel(inp,inz,iidx,n-1)
    @always_comb
    def p():
        if inp[n]:
            nz.next = True
            idx.next = n
        else:
            nz.next = inz
            idx.next = iidx
    return ret,p

toVHDL(toplevel,inp,nz,idx)
```
Here, `n` is the constant that has a different value on each recursive level.
Note that when `idx.next = n` is replaced by `idx.next = n-1`. then the name `n` will not be used, but replaced by it's value in the generated VHDL.

cfelton, 2015-08-21:
@burnpanck you can change the code to `idx.next = int(n)` and it will work as you want.

burnpanck, 2015-08-21:
Ok, thanks for the workaround! I actually just did what I noted at the very end: offset the constant by one. In any case, I think this is a serious bug that should be addressed: It leads to incorrect code that does not follow the python source. such bugs are particularly annoying, as the user will always first think that his own code has a bug rather than suspecting the translation to be faulty.

cfelton, 2015-08-21:
@burnpanck It isn't a bug, the converter can't infer the designers intent.  Are they using `n` as a variable or do they (the designer) expect the literal value of `n` at the point it is used in the generator instantiation.  The `int(n)` actually follows one of the Python mantras: being explicit.  In this case it is clear that you want the literal value of `n` at conversion and not `n` as a variable.

It might be reasonable to entertain a conversation, if the converter can be enhanced to infer the intended use (variable or literal) but I don't know if it is possible, and if it is possible it probably is not trivial.

burnpanck, 2015-08-21:
I see what you mean, but I disagree. The fact that `n` is an argument to the elaboration function should make it clear that this is not global _constant_ that is meant to be changed by hand in the VHDL afterwards, but it really is a local parameter that happens to be simple constant value. The behaviour I see is completely unexpected: The `n` that gets passed to the different levels of  `toplevel` are all different, but the generated code behaves as if it was the same. The generated code behaves differently than the simulated code if simulated using MyHDL. I cannot see any other design intent other than when I write `n`, it shall mean whatever object is denoted by `n` at that point! If it happens to be a signal, then that means the value of that signal, which is not a constant, but still the value of the signal I denoted by that name. If it happens to be a constant, then I mean the value of that constant. The fact that it has the same name as another constant does not make it the same.
Whatever other design intent one could imagine, then it should apply to all instances of `n`. In the current situation `signal.next = n`, that would then be understood as assigning the unique global constant by the name `n`. But `signal.next = n+1` it is understood as assigning the incremented constant value of a local parameter. If that is not a bug, then it is at least completely inconsistent.

josyb, 2015-09-02:
I assume that the simulation is OK.
Then it is a conversion issue - only ;)
If we look at one of the generated processes:
```VHDL
	toplevel_ret_p : process(ret_iidx, inp, ret_inz) is
	begin
		if bool(inp(3)) then
			inz  <= '1';
			iidx <= to_unsigned(n, 3);
		else
			inz  <= ret_inz;
			iidx <= resize(ret_iidx, 3);
		end if;
	end process toplevel_ret_p;
```
We can see that the _n_ in ```if inp[n]:``` gets correctly recognised but goes wrong in ```idx.next = n```
I looked at the ast tree with pyastviewer.py and at my trace.log and can see that the actual passed value for _n_ is a _real_ which in case of the Subscript (inp[n]) gets converted into an _int_ but not when used as the assignment value (= n). This explains why the _cast_ ```int( n )``` proposed by Chris cures the ailment.

jandecaluwe, 2015-09-19:
First: iff simulation is works fine, and successfully converted code doesn't match it, there is a bug. (X- initalization issues notwithstanding.) So yes, there is a bug here.

The culprit is some hacking of mine at a time when MyHDL was less popular, to make converted VHDL more "beautiful" by trying to infer symbolic constant names, instead of simply using numeric values. That was an attempt to please a customer. However, that code is brittle and ugly and it's VHDL only. Moreover, as seen here, it leads to bugs which are really unacceptable as recursion during elaboration should work.

I propose to take all related code out of the MyHDL codebase, making the code smaller and cleaner again. **The implication is that symbolic constants will no longer be in the converted VHDL code, making it lower-level and more difficult to review**. You cannot have everything.

If this type of functionality is important, we will have to consider alternative, probably more explicit approaches and rethink matters from the ground up.

I'll wait a week or so for possible feedback before going ahead with this.

josyb, 2015-09-19:
I'm perfectly happy with _naked_ integers and binary bitstrings in stead of symbolic constants (although I might have a reputation of being a pit _picky_). The named constants are not consistently converted e.g. if you call on a function multiple times only the first gets a symbolic constant while the others get a _naked_ integer (it is actually a bit more complicated).

It would be nice to have symbolic constants though because it makes the VHDL, as you say, _more beautiful_, and hence _more_ acceptable to the customer. But there are other things we can do to improve the _beauty_ of the generated VHDL, e.g. formatting it like the Sigasi editor does. But other enhancements like _Array_ and _indexed constants_ have priority as they will improve productivity with _more beautiful_ Python code (which is were our interests lie).

jandecaluwe, 2015-09-26:
This was fixed by improving a significant amount of doubtful code. My favorite kind of fix ;-) Symbolic constants in VHDL output are gone, but the result is a cleaner MyHDL and more in  line with the philosophy: Verilog/VHDL are useful back-ends, nothing more.

closed 2015-09-26 by jandecaluwe

cfelton, 2015-09-26:
@jandecaluwe sounds great, I agree with the approach, Verilog/VHDL useful backends (my feedback is late of course :)   

@burnpanck, sorry for the confusion, it should have been evident that distinguishing uses isn't a big obstacle.  And it should have been obvious because the Verilog conversion worked for the test-case.

commit b28ef3e2 (2015-09-26): Fix #122
commit beed676d (2015-09-26): Fix #122 by removing support for named constant in VHDL