myhdl/myhdl #123: [bug] Inconsistent Hierarchy returning list of generators

cfelton, 2015-08-21:
A subtle error occurs, in some scenarios, when a list of generators is returned.  The following captures the slight differences

```python
def mod1():
    # ...
    return gen1

def mod2():
    # ...
    return [gen2]

def top():
    mod_insts = []
    mod_insts += [mod1]   # this works
    mod_insts += mod2     # this doesn't work 
```

This appears to only occur when a single generator is being returned.  [This test](https://github.com/cfelton/myhdl/blob/hier_list_error/myhdl/test/bugs/test_issue_123_2.py) demonstrates the error.

jandecaluwe, 2015-09-30:
Mm, I fear this is not straightforward.

The problem is that hierarchy extraction expects the whole hierarchy tree to be reconstructable from the return values. It does look at object ids for that purpose. But when you extend a list with another one, the latter object is gone. So the return value of mod2 is no longer found in the return value of the higher level module that instantiates it. In hardware terms, list.extend flattens out the hierarchy.

At this point, my best suggestion is to make the error message clearer. There are a couple of ways to get an "inconsistent" hierarchy; the user should get extensive feedback on what may cause it. And suggestions, like: use list.append instead of list.extend to built instance return values.

BTW: this is a good task for a student/beginner: find/list all the different cases in which you can get hierarchy inconsistencies, and suggest a help message for users.

cfelton, 2015-10-01:
Ugh, that is a tough one ... 

When I originally reported this I took part of a day to add some additional [debug information](https://github.com/jandecaluwe/myhdl/compare/master...cfelton:issue_123_hier_list_error)  (the original effort was [here](https://github.com/jandecaluwe/myhdl/compare/master...cfelton:hier_list_error) but the git history was junk and I had to clean it up).  I took a swag at adding a `dump_hierarchy` function, it produces something like:

    Hierarchy for "fft_permute_top"
       :fft_permute_top(128) [ 80, 192]
           returns rtl_s2p, rtl_permute
         +fft_permute(192)  has 0, 2, 2
             returns rtl_permute

Also added some tests, the following is a dump for one of the **list.extend** failures:

    Hierarchy for "list_of_interfaces_top_2"
    Hierarchy Error: <gen list?> <class 'list'> (68300424) missing
    Inconsistent hierarchy - are all instances returned ?
       :list_of_interfaces_top_2(232) [656, 304, 328, 720, 488, 984, 600]
           returns rtl, rtl, rtl, rtl, rtl, rtl, rtl
         :mod_with_interface_2(424) [600]
             returns rtl
         :mod_with_interface_2(120) [984]
             returns rtl
         :mod_with_interface_2(208) [488]
             returns rtl
         :mod_with_interface_2(168) [720]
             returns rtl
         :mod_with_interface_2(752) [328]
             returns rtl
         :mod_with_interface_2(840) [304]
             returns rtl
         :mod_with_interface_2(816) [656]
             returns rtl

    Hierarchy Error: <gen list?> <class 'list'> (58747208) missing

I realize the dump is a little cryptic, I figured it needed to be compact, especially for larger designs.  I would need to add documentation (to the manual?) if we decided this was useful.   

Since this isn't really a bug (current limitation?) I should move the tests I added to `core` (?) and catch the error.  In the future, as you suggest, someone can walk through all the cases and propose useful messages.  If we want this feature I can create a PR when complete.

jandecaluwe, 2015-10-03:
It's time for reflection ;-)

As you know, the design goal of MyHDL is to be minimalistic, reuse standard Python as much as possible, and only introduce specific features when really necessary. I think that has serve us well.

However, there are times when the concept fails and this is one. If you flatten hierarchy in a module by using powerful Python features, there is no way hierarchy extraction can be done, and the problems occur mysteriously late in the design process. Not ideal.

I  have been thinking about this and I believe there is good solution: bite the bullet and introduce another decorator `myhdl.module` for function intended to describe hardware from which hierarchy can be extracted (e.g. for tracing signals, and for conversion). Nothing would change for general modelling, but a backward-incompatiblity would be introduced for such modules.

The benefit could be great. Readiblity would probably be better, as we clearly see where the "hardware" module functions are. The decorator could verify the return value and check if is "well-formed" (i.e. all return instances are found in the namespace), flagging any issues upfront.

A big potential benefit is that we could probably drastically the hierarchy extraction process. It is the part I like/trust the least currently. Hacking the profiler is "nice magic" but it has also fundamental drawbacks (people have complained about this.) The decorator could decorate the return value with all info necessary for hierarchy extraction.

Another benefit is that this opens the possibilty for some interesting checks on the hardware module function that can avoid common errors.

This is all just premature thinking out of my head, food for thought, feedback welcome.

cfelton, 2015-10-03:
I just created PR #132, I had created it before the latest comments but had not submitted it.  Wanted to add a quick note since it overlap the `@myhdl.module` suggestion.

cfelton, 2015-10-03:
At a first glance the `@myhdl.module` proposal sounds reasonable.  I like the how it will explicitly define a myhdl module versus other functions.  From a users perspective the additional overhead is minimal and it seems like we will get a lot of benefit on the analysis and conversion.  Backward incompatibility seems ok because porting should be straightforward (minimal task).

Does this mean the functions decorated with the `@myhdl.module` will not need to return the generators?

jandecaluwe, 2015-10-04:
> Does this mean the functions decorated with the @myhdl.module will not need to return the generators?

Initially I thought to leave the return value as it is and only implement checks. However, as the decorator is going to change the return value anyway (returning a dedicated object instead) it is probably less confusing to not have a return value at all. The meaning of the decorator would then be: "construct an appropriate return value from module instances and local generators". An then everything would be correct by construction.

To make porting easier we could tolerate the return value initially and just warn that it is being ignored if not `None`.

Another thought: if the decorator uses a specific type for the return value, things like `list.extend` would not work automatically, while `list.append` would. So probably a lot of errors would be caught upfront, even without special checks by the decorator.

jck, 2015-10-06:
I think it is possible to avoid using the profiler without introducing a new decorator.

The instance decorator can take care of building the symbol table. After that, the return value(of the top level module) should have sufficient information for the conversion process.

jandecaluwe, 2015-10-07:
@jck I am not following here. The functions containing instantiations and generators are modules in the hierarchy. Hierarchy extraction infers this hierarchy for name expansion during conversion, and for the hierarchy in waveform tracing. I don't see how you could do that with instance decorators only.

On a related note: what is your opinion, conceptually, of the return value of "module" functions? One could see it as a case of "explicit is better than implicit", or conversely, as error-prone overhead in our specific use case.

jck, 2015-10-07:
The instance decorators can get the code objects(and names) of the functions they are enclosed in by using inspect. Example: http://stackoverflow.com/questions/5067604/determine-function-name-from-within-that-function-without-using-traceback/5067654#5067654

I'm still thinking about the module decorator.

jandecaluwe, 2015-10-07:
Yes, but not some hardware-describing functions will be purely structural, that is, will only  contain "instances" (by calls to lower-level functions) and no local generators defined by instance decorators. But such functions are still part of the hierarchy.

jck, 2015-10-07:
oops, I didn't think about that case. The myhdl.module decorator seems like the best way to proceed.

I think local generators should still be explicitly returned. However, we can add support for using always/alwas_comb etc as context managers as well. In any function, logic inside these context managers is automatically a part of this 'module', and logic inside local generators need to be explicitly returned(useful for parametrization).

jck, 2015-10-07:
Example:

```
@myhdl.module
def Mux(z, a, b, sel):
    with always_comb:
        if sel == 1:
            z.next = a
        else:
            z.next = b
```

jck, 2015-10-07:
Actually, the context manager solution might be too complex to implement because I'm not sure if we'll have closures like the nested generators.

I like the idea of @myhdl.module meaning "construct an appropriate return value from module instances and local generators". However, how would someone do [conditional instantiation](http://docs.myhdl.org/en/stable/manual/structure.html#conditional-instantiation) in this case?

jandecaluwe, 2015-10-08:
Ok. Conditional instantiation proves that we do need to support return values properly. Case closed. So let us start from there.

The `myhdl.module` decorator would check return values to be "well-formed", and it would modify them into an annotated version suitable for e.g. hierarchy extraction. If I look at decorator tutorials, this is actually a common use case and so should be quite natural for people to accept. The immediate benefit is that all kinds of late mysterious errors with hierarchy extraction would be flagged upfront. Moreover, I believe readibility would be greatly enhanced. 

And an additional benefit is that we would be able to dump the profiler hacking and drastically simplify the hierarchy extraction code. All very good. Looks likes this a must-have feature.

In a later phase we *could* consider to treat the absence of a return value (`None`) as as an indication to build a sensible default one.  I am not sure, we can delay that decision. In any case I would like to deprecate the `instances()` function.

josyb, 2016-01-13:
Would this ```myhdl.module``` help me in simplifying the overhead in returning the generators in my *alternative* development approach as shown on https://gist.github.com/josyb/afd84c9a06fdec77f2fd?

josyb, 2016-01-13:
@jck I implement conditional instantiation as follows:
```python
SLOW, MEDIUM, FAST = range(3)

def top(..., speed=SLOW):
    ...
    ...
    if speed == SLOW:
        @always_xxx
        def slowAndSmall():
            ...
        return slowAndSmall
    elif speed == FAST:
        @always_xxx
        def fastAndLarge():
            ...
        return fastAndLarge
    else:
        raise NotImplementedError
```
This may be easier to cope with because the compiler will only see one path?
Quite often I (have to) use ```return myhdl.instances()``` to collect the generators (If there are several unrelated conditional instantiations involved).

jandecaluwe, 2016-01-14:
What do you mean with "simplifying the overhead"?

josyb, 2016-01-14:
```python
    reg1 = Register(WIDTH_D, Clk, D, Shift)
    reg1rtl = reg1.rtl()
    reg2 = Register(WIDTH_D, Clk, reg1.result, Shift)
    reg2rtl = reg2.rtl()
    reg3 = Register(WIDTH_D, Clk, reg2.result, Shift)
    reg3rtl = reg3.rtl()
    reg4 = Register(WIDTH_D, Clk, reg3.result, Shift, Q)
    reg4rtl = reg4.rtl()
    . . .
    return myhdl.instances()
```
For every class instantiation there must be a call to *collect* the generators. It would be nice if these actions ```regxrtl = regx.rtl()``` would happen magically (making the code even more beautiful).

jandecaluwe, 2016-01-15:
Amazing that this coding style even converts ;-)

One advantage of this style I can see is to experiment with various types of architectures. Therefore, I don't think MyHDL should have specific support for one particular architecture name.

Also, I believe we should be careful with "magic". It is tempting, it can be useful, but on the other hand it can hamper explicitness, making code obscure to a reviewer.

That having said, there is nothing that prevents you to add custom magic that is useful for your projects yourself. As you long as it is in the elaboration phase (like e.g. `instances`) you can do what you like without jeopardizing convertibility.

josyb, 2016-01-15:
It converts, and both VHDL and Verilog visually look good. Of course I'd better verify that with co-simulation, but I haven't set that up yet. (Have you seen how I formatted the VHDL code?- in Sigasi style)
I added the Verilog code to the gist:https://gist.github.com/josyb/afd84c9a06fdec77f2fd.
I only added minor tweaks to remove *_self* and *_rtl* from the generated names.
IMO the proposed ```@myhdl.module``` would  avoid that particular _rtl_ architecture name.
For now I will take your advice and write a helper function to collect the generators. I tried an enhanced ```myhdl.rtlinstances()``` but that is not straightforward as that changes the prefix naming.

jandecaluwe, 2016-02-15:
The draft MEP for the @module decorator is in http://dev.myhdl.org/meps/mep-114.html. Disqus has been enabled for feedback.

The implementation is in feature branch https://github.com/jandecaluwe/myhdl/tree/mep-114. Many tests have been migrated, and currently the regressions pass on all platforms.

jck, 2016-02-15:
Jan, if you create pull request from that branch to master, we can review the code inline there.

jandecaluwe, 2016-02-16:
I have created pull request #147 from the feature branch, for easier review.

smldis, 2016-02-16:
Hello, thank you all for your work on this awesome project.
I'm seeing a lot of effort going in the hierarchy part of the project, but I feel uncomfortable with the functional style used for modules.
What are the advantages of using functions instead of a module class?
With a class we could manage module parameters, hierarchy and generators easily.
Parameters for example could be used to set if that module is just a model and should not be converted, or they could provide information about results of module instatiation like number of pipeline stages or more specifically number of taps of an automatically generated filter.

jandecaluwe, 2016-02-16:
> What are the advantages of using functions instead of a module class?

The functional style has the smallest overhead for a user and should feel familiar to HDL designers.

cfelton, 2016-02-18:
@jandecaluwe I have reviewed [MEP114](http://dev.myhdl.org/meps/mep-114.html).  One useful addition would be to include a code snip.  I realize the `@module` is straightforward and mentioned but and example would be useful in the MEP:

```
from myhdl import module, Signal, intbv
@module
def somemodule(clock, reset, count):
    # ...
```

There are a couple small typos:

    toVHDL(func, *args, **kwars)  -> oVHDL(func(*args. **kwargs))

Also, when I tried switching a couple of my examples to use the `module` I ran into the following error:

      File "myhdl/myhdl/_module.py", line 71, in _getCallInfo
        callerrec = stack[4]
    IndexError: list index out of range

[line 71 in _module.py](https://github.com/jandecaluwe/myhdl/blob/mep-114/myhdl/_module.py#L71)

It seems the length of the stack is only 4 in [this case](https://github.com/cfelton/alt.hdl/blob/mep-114/examples/ex1_wprcnt/myhdl/useless_counter.py).  Tried it with Python versions 2.7.6 and 3.4.0.  It looks like the error and unexpected stack size was do to the fact that I, in this case, had the conversion code in the Python module (file).  The [test](https://github.com/cfelton/alt.hdl/blob/mep-114/examples/ex1_wprcnt/myhdl/test_useless_counter.py) runs fine and the conversion works, the code in questions needs to be modified to be more generic to handle the various stack sizes.

jandecaluwe, 2016-02-20:
Thanks for the feedback. Commit 9b5a613e55a6003b04480710a71d9d8a5a983bc2 should fix the issue. Also, I have updated http://dev.myhdl.org/meps/mep-114.html.