myhdl/myhdl #125: Usability of Interfaces

josyb, 2015-09-02:
Yesterday I decided to try using an Interface. 
I am currently experimenting with a StructType object to mimic the VHDL _record_ / SystemVerilog _struct_, and wanted to get a feel for the 'how and what' of the Interface.  
Some background: I use MyHDL to develop re-usable modules for use with Altera's Qsys. Qsys has two main interfaces, the Memory Mapped (MM) and the Streaming (ST) types. All of the modules have (at least) one input and (at least) one output ST-interface. So the ST was a good candidate for my first try.
I also have a utility to generate the necessary _hw.tcl file for Qsys to be able to recognise/parametrise the module. This utility also steers the conversion.
I added some code to generate the members for the ST-interfaces on demand.

I was kindly surprised, because it looked good. The ports were all there and had the correct names matching the advertised names in the corresponding _hw.tcl file.

So today I decided I thus had to upgrade all my existing modules to conform to my latest changes in the Qsys utility, not necessarily because of the interfaces but in anticipating using interfaces I had already changed some port naming rules.
The first module mentioned above was a rather simple one and only had one level of functions, better said the top function was the only RTL function.
Now when trying/ working on a larger module where the top function only collects the RTL from lower modules I find that the names are 'mangled'.
This MyHDL code
```python
import myhdl

class fi( object ):
    ''' a simple interface '''
    def __init__(self):
        self.Data = myhdl.Signal( myhdl.intbv()[8:])
        self.Ready = myhdl.Signal( bool(0) )
        self.Valid = myhdl.Signal( bool(0) )
        
                
def mangler( Clk, Reset, D, Q):
    ''' a silly module to mangle the input data '''
    @myhdl.always_comb
    def mangler():
        Q.Data.next[:4] = D.Data[4:]
        Q.Data.next[4:] = D.Data[:4]
        Q.Valid.next = D.Valid
        D.Ready.next = Q.Ready
    
    
    return mangler

def topfi(Clk, Reset, D, Q):
    ''' the top level '''
    
    mnglr = mangler(Clk, Reset, D, Q)
    
    return mnglr

def convert():
    
    Clk = myhdl.Signal( bool(0) )
    Reset = myhdl.ResetSignal(0, active=1, async=True)
    D = fi()
    Q = fi()
    
    myhdl.toVHDL( topfi, Clk, Reset, D, Q)
    

if __name__ == '__main__':
    ''' '''
    convert()
```

produces the following VHDL:
```VHDL
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;
use std.textio.all;

use work.pck_myhdl_10.all;

entity topfi is
	port(
		Clk           : in  std_logic;
		Reset         : in  std_logic;
		mnglr_Q_Ready : in  std_logic;
		mnglr_Q_Valid : out std_logic;
		mnglr_Q_Data  : out unsigned(7 downto 0);
		mnglr_D_Ready : out std_logic;
		mnglr_D_Valid : in  std_logic;
		mnglr_D_Data  : in  unsigned(7 downto 0)
	);
end entity topfi;
-- the top level 

architecture MyHDL of topfi is
begin
	TOPFI_MNGLR_MANGLER : process(mnglr_D_Valid, mnglr_D_Data, mnglr_Q_Ready) is
	begin
		mnglr_Q_Data(8 - 1 downto 4) <= mnglr_D_Data(4 - 1 downto 0);
		mnglr_Q_Data(4 - 1 downto 0) <= mnglr_D_Data(8 - 1 downto 4);
		mnglr_Q_Valid                <= mnglr_D_Valid;
		mnglr_D_Ready                <= mnglr_Q_Ready;
	end process TOPFI_MNGLR_MANGLER;

end architecture MyHDL;
```
where I expected / need this:
```VHDL
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;
use std.textio.all;

use work.pck_myhdl_10.all;

entity topfi is
	port(
		Clk           : in  std_logic;
		Reset         : in  std_logic;
		Q_Ready : in  std_logic;
		Q_Valid : out std_logic;
		Q_Data  : out unsigned(7 downto 0);
		D_Ready : out std_logic;
		D_Valid : in  std_logic;
		D_Data  : in  unsigned(7 downto 0)
	);
end entity topfi;
-- the top level 

architecture MyHDL of topfi is
begin
	TOPFI_MNGLR_MANGLER : process(D_Valid, D_Data, Q_Ready) is
	begin
		Q_Data(8 - 1 downto 4) <= D_Data(4 - 1 downto 0);
		Q_Data(4 - 1 downto 0) <= D_Data(8 - 1 downto 4);
		Q_Valid                <= D_Valid;
		D_Ready                <= Q_Ready;
	end process TOPFI_MNGLR_MANGLER;

end architecture MyHDL;
```
I used the current 1.0-dev master branch to check this (not my experimental code).
Your thoughts?

cfelton, 2015-09-02:
@josyb this is a current limitation.  The top-level port names will be a result of the python compiler flattening the design.  This has been discussed as a future enhancement (preserving the top-level port names) but it currently isn't on the list of things.

josyb, 2015-09-03:
I would call it a _bugfix_ rather than a _future enhancement_.
Looking at the workaround (excuse me for rudely copying from my actual work instead of expanding the above example, but I am under pressure):
```python
def ST_combine2( Clk, Reset,
                 D0Data, D0SoP, D0EoP, D0Error, D0Valid, D0Ready,
                 D1Data, D1SoP, D1EoP, D1Error, D1Valid, D1Ready,
                 QData, QSoP, QEoP, QError, QValid, QReady,
                 ):
    ''' the Qsys wrapper
       this is what is seen by Qsys
    '''
    # start the collection
    rtl = []
    # make interfaces
    D0 = AvalonInterface.ST()
    # addSignals makes CloneSignals of the input Signals
    D0.addSignals( {'Data': D0Data, 'SoP': D0SoP, 'EoP': D0EoP, 'Error': D0Error, 'Valid': D0Valid } )
    # but the outputs need a Signal proper, which later will be _assigned_ to the original output
    D0.makeSignals( {'Ready': D0Ready})
    
    D1 = AvalonInterface.ST()
    D1.addSignals( {'Data': D1Data, 'SoP': D1SoP, 'EoP': D1EoP, 'Error': D1Error, 'Valid': D1Valid } )
    D1.makeSignals( {'Ready': D1Ready})

    Q = AvalonInterface.ST()
    Q.addSignals( {'Ready': QReady})
    Q.makeSignals( {'Data': QData, 'SoP': QSoP, 'EoP': QEoP, 'Error': QError, 'Valid': QValid } )
    
    # delegate the action to the module that **does** handle the 'interfaces'        
    rtl.append( st_combine2( Clk, Reset, D0, D1, Q))

    # but now we have to do some **extra** assigning
    if QReady is not None:
        @myhdl.always_comb
        def ready():
            D0Ready.next = D0.Ready
            D1Ready.next = D1.Ready
        rtl.append( ready )
        
    if QData is not None:
        @myhdl.always_comb
        def data():
            QData.next = Q.Data
        rtl.append( data )

    if QSoP is not None:
        @myhdl.always_comb
        def packets():
            QSoP.next = Q.SoP
            QEoP.next = Q.EoP
        rtl.append( packets )  
          
    if QError is not None:
        @myhdl.always_comb
        def error():
            QError.next = Q.Error
        rtl.append( error)
         
    if QValid is not None:
        @myhdl.always_comb
        def valid():
            QValid.next = Q.Valid
        rtl.append( valid )

        
    return rtl

def st_combine2( Clk, Reset, D0, D1, Q):
    """ this can handle interfaces  
         and is the one we will use when building a system in MyHDL
    """
...
```
Most of the nitty-gritty work is done in the AvalonInterface ST-class and in my Qsys-generator module to reduce the typing effort. And I can copy/paste from this or better make a template for it in Eclipse.

nturley, 2015-11-10:
This is super easy. The problem is here:

https://github.com/jandecaluwe/myhdl/blob/9dffa041305d66ff3a85392ae29de11952b70d18/myhdl/conversion/_analyze.py#L1244

You want all arguments to be name + '_' + attr, however that only happens if it hasn't been used. If it has been used, it keeps the name it was assigned. This was probably because they didn't want to screw up references to the signal by changing it's name. Thankfully, there is really good symbol table thing that I don't understand that keeps it working just fine.

I made the change and this is what your code produced.

``` python
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;
use std.textio.all;

use work.pck_myhdl_10.all;

entity topfi is
    port (
        Clk: in std_logic;
        Reset: in std_logic;
        Q_Ready: in std_logic;
        Q_Valid: out std_logic;
        Q_Data: out unsigned(7 downto 0);
        D_Ready: out std_logic;
        D_Valid: in std_logic;
        D_Data: in unsigned(7 downto 0)
    );
end entity topfi;
-- the top level 

architecture MyHDL of topfi is
begin

TOPFI_MNGLR_MANGLER: process (D_Data, Q_Ready, D_Valid) is
begin
    Q_Data(8-1 downto 4) <= D_Data(4-1 downto 0);
    Q_Data(4-1 downto 0) <= D_Data(8-1 downto 4);
    Q_Valid <= D_Valid;
    D_Ready <= Q_Ready;
end process TOPFI_MNGLR_MANGLER;

end architecture MyHDL;
```

josyb, 2015-11-10:
Thanks Neil!
I already did something in that area to support nested interfaces. The code is on another machine I don't have access to at this moment.
May I suggest you make that branch and start testing.  I believe you are quite right that this will work OK.  I will merge that modification into my local MyHDL code too  and check this in one of my more complicated designs.

cfelton, 2015-11-10:
@nturley @josyb, we are specifically talking about **top-level port interface name preservation**, correct?  We might consider changing the title of this issue to the specific problem/requested enhancement.

If you run the test suite with the changes suggested above, one of the tests in `test/conversion/general` fails, Unfortunately, I doubt the change is this simple (I didn't debug, simply ran the tests).  

Personally, I would like to see this issue re-written as an *enhancement* request to preserve top-level port name with justification and uses cases.  With the feature clearly defined we could create some tests that currently fail and then entertain how to implement the feature.  I imagine this renaming can happen with non-interfaces as well (although I can not make it happen).  This enhancement may or may not fix #134.  

The top-level port renaming should probably be listed in the [known limitations](http://docs.myhdl.org/en/stable/manual/conversion.html#known-issues) for the current documentation.

I could be in the minority here thinking this is an enhancement, if the consensus is to define port renaming as a bug then we need to limit the conversation to the port renaming and generate tests to add to the `test/bugs`.

josyb, 2015-11-10:
@cfelton, which test failed? If it is only one, it is worth looking into.
I think the change may work: essentially it is applying the top-name instead of the derived names, just like the top-name for other top-level ports is preserved. If this happens early enough in the conversion the object will be known by that name?

Speaking for me personally, I'd classify it as a 'flaw' (avoiding the b-word), but it is not urgent to me. I have a workaround, but I hope to one day have a better alternative. (as we discussed on gitter, and after we have a solution for the multidimensional types)

cfelton, 2015-11-10:
@josyb it is one of the interface tests (`test_interface1.py`) and the name of the test is `test_name_conflict_after_replace`.

josyb, 2015-11-10:
@cfelton I have that one marked in my local code with a ```@pytest.mark.xfail```.   
@jck is testing his renaming of a generated name which conflicts with another top-level signal (in this case a_x). I have bypassed that renaming, because I feel that that conflict should be an error to be solved in the MyHDL RTL modules rather than later when synthesising  in the FPGA vendor's tools.

nturley, 2015-11-11:
I agree with @cfelton that it's an enhancement. I assume that currently myHDL makes no guarantees about signal names, just that they will be valid VHDL or verilog. So any invalid signal names is a bug and any improvements in signal names is an enhancement. Whether it should make guarantees or what those guarantees are is something to write up an MEP about (In fact, I just wrote one). Once an MEP is approved, then any deviations from the MEP specification changes from enhancements to bugs.

The hack I mentioned above (again, I reiterate the word hack) would fix the test case referenced for #134, but I doubt it would catch all collisions involving top level interface signals (indeed, apparently test_name_conflict_after_replace found one). I also agree that it's probably not just this simple (hence my warning that there might be weird side effects).

josyb, 2015-11-11:
@nturley  A rose by any other name would smell as sweet ... the limitation of interfaces having issues when used at the top level is not mentioned in the docs. Actually in MEP-107 it is clearly indicated: _Interfaces as ports, including top-level ports_.
The test ```test_name_after_conflict``` doesn't implicate that your  _patch_ is the wrong approach, only it now fails because it has been preempted and the targeted renaming of a_x into something like a_x_1 doesn't take place. But I find this renaming a little bit contrived and although this avoids an error while converting to V*, the user will have to take action sooner or later if this signal at the top level has been renamed. Hence my suggestion to not perform such renaming and have the user correct/change the conflicting names.