myhdl/myhdl #126: [Enhancement] 'Open' Signals

josyb, 2015-09-06:
When creating re-usable modules we create output signals, which quite often are not all used in an instantiation. In VHDL we have the _open_ keyword to tell the compiler about our intention. Of course in Python we would use _None_ to achieve the same. But the _None_ is propagated and we end up assigning something to _None_ which results in an Attribute Exception.
The following code illustrates this:
```python
def addresscounter( clk, reset, startValue, Length , initP, cntEn, 
                                 isStart, isEnd, Q, WRAP_AROUND = False ):

    lq = Signal(intbv(0)[len(Q):])

    @always_seq(clk.posedge, reset=reset)
    def counter():
        endval = int(startValue) + int(Length)
        if initP or cntEn:
            if initP :
                lq.next = startValue
                isStart.next = 1
                isEnd.next = 0
            else:
                if lq < (endval - 1):
                    lq.next = lq + 1
                    isStart.next = 0
                else:
                    lq.next = startValue
                    isStart.next = 1

                if lq == (endval - 2):
                    isEnd.next = 1
                else:
                    isEnd.next = 0

    @always_comb
    def assignoutputs():
        Q.next = lq
```
When we instantiate the module as follows:
```python
wraddresscounter = addresscounter.addresscounter(Clk, Reset, 0, DEPTH, wraddresscounterinitP, 
                                                 wraddresscountercntEn, None, None, 
                                                 wraddresscounter_Q, WRAP_AROUND= True)
```
We will get:
```text
AttributeError: 'NoneType' object has no attribute 'next'
```
There are two workarounds:
1. declare two (in this case) SIgnals and add them to the instantiation
```python
wraddresscounter = addresscounter.addresscounter(Clk, Reset, 0, DEPTH, wraddresscounterinitP, 
                                                 wraddresscountercntEn,  dummysig1, dummysig2, 
                                                 wraddresscounter_Q, WRAP_AROUND= True)
```
2. declare local Signals in the re-usable module and selectively assign them to the outputs
```python
def addresscounter( clk, reset, startValue, Length , initP, cntEn, 
                                 isStart, isEnd, Q, WRAP_AROUND = False ):

    lq = Signal(intbv(0)[len(Q):])
    lisstart = Signal(bool(0))
    lisend = Signal(bool(0))

    @always_seq(clk.posedge, reset=reset)
    def counter():
        endval = int(startValue) + int(Length)
        if initP or cntEn:
            if initP :
                lq.next = startValue
                lisstart .next = 1
                lisend.next = 0
            else:
                if lq < (endval - 1):
                    lq.next = lq + 1
                    lisstart .next = 0
                else:
                    lq.next = startValue
                    lisstart .next = 1

                if lq == (endval - 2):
                    lisend.next = 1
                else:
                    lisend.next = 0

    @always_comb
    def assignoutputs():
        Q.next = lq

    if IsStart is not None:
        @always_comb
        def assignis():
            IsStart.next = lisstart

    if IsEnd is not None:
        @always_comb
        def assignie():
            IsEnd.next = lisend

    return instances()
```
Both workarounds are equivalent (with 2. being  preferred), **but** both will end up with warning(s):
```
** ToVerilogWarning: Signal is driven but not read: lisstart
** ToVerilogWarning: Signal is driven but not read: lisend
```

I would like to get rid of the warning(s).  I discovered _/_markRead()_ in _Signal.py: if I change the code as follows:
``` python
    if IsStart is None:
        lisstart._markRead()
    else:
        @always_comb
        def assignis():
            IsStart.next = lisstart
```
the warnings are gone.
There are two _caveat_:
 1. The _markRead() method is not documented.
 2. The _open_Signals are still in the converted code, so later on the Synthesiser will complain anyway. 
 ```text
Warning (10036): Verilog HDL or VHDL warning at scfifo.vhd(68): 
      object "wraddresscounter_lisend" assigned a value but never read
 ```

So ideally the lines assigning to _open_ Signals should be removed from the converted code.

josyb, 2015-09-09:
A QaD approach would be to _post-edit_ the generated VHDL file and remove the _qualified not-read_ signals.
Qualifying a Signal for warning suppression / removal:
 1. add an attribute _suppresswarning to the _Signal class
 2. add a method suppresWarning() to set that atrribute to _True_
 3. in the user code, like above, call sig.suppresWarning() (replacing the call to _markRead() above)
 4. in _toVHDL.py, add code to selectively suppress the warning for the marked Signals, while at the same collecting them in a list.
 5. at the end of conversion, re-open the generated VHDL file and remove those lines containing one of the Signals in the collected list. They will appear only in the signal declaration and in lines where they are the LHS, so there is no danger in deleting those lines except perhaps that you may end up with _empty statements_ like:
```VHDL
if  bool( somesig ) then
else
end if ;
```
ModelSim nor Quartus II seem to mind. No warning, not even an informational message. 

for Verilog, I suppose a similar solution would be possible?

smldis, 2016-02-15:
Hello to contributors,
What about this?

```python
def addresscounter( clk, reset, startValue, Length , initP, cntEn, 
                                 isStart, isEnd, Q, WRAP_AROUND = False ):

    lq = Signal(intbv(0)[len(Q):])

    @always_seq(clk.posedge, reset=reset)
    def counter():
        endval = int(startValue) + int(Length)
        if initP or cntEn:
            if initP :
                lq.next = startValue
            else:
                if lq < (endval - 1):
                    lq.next = lq + 1
                else:
                    lq.next = startValue

    if isStart is not None:
        @always_seq(clk.posedge, reset=reset)
        def counter_isstart():
            endval = int(startValue) + int(Length)
            if initP or cntEn:
                if initP :
                    isStart.next = 1
                else:
                    if lq < (endval - 1):
                        isStart.next = 0
                    else:
                        isStart.next = 1

    if isEnd is not None:
        @always_seq(clk.posedge, reset=reset)
        def counter_isend():
            endval = int(startValue) + int(Length)
            if initP or cntEn:
                if initP :
                    isEnd.next = 0
                else:
                    if lq == (endval - 2):
                        isEnd.next = 1
                    else:
                        isEnd.next = 0

    @always_comb
    def assignoutputs():
        Q.next = lq

    return instances()
```

I just translated the code without inspecting it, I think it can be rewrited in something more readable if needed.

hgomersall, 2016-03-23:
Surely the correct way to handle this is explicitly - have a default kwarg set to None and handle the logic correctly inside the block? Having signals not set may or may not be a bug, and it should be picked up accordingly.

Nicolasp38, 2016-03-23:
Why not create a class for this : 
```
class Open():
    def __init__(self) :
        self.next = None
```
and use it like this :
```
wraddresscounter = addresscounter.addresscounter(Clk, Reset, 0, DEPTH, wraddresscounterinitP, 
                                                 wraddresscountercntEn, Open(), Open(), 
                                                 wraddresscounter_Q, WRAP_AROUND= True)
```
This way, no need to use a special coding inside addresscounter (simple is better).
Open() is explicit (explicit is better)
Also, the parser can be aware of unconnected ports/signals.

Nicolasp38, 2016-03-23:
Maybe Unconnected() is better than Open().

cfelton, 2016-03-23:
As I understand this issue, I agree with @hgomersall.  In this case using a Python approach to handle ports (arguments) you don't want to pass is sufficient.  Also, using [interfaces](http://docs.myhdl.org/en/stable/whatsnew/0.9.html#interfaces-conversion-of-attribute-accesses) solves most of these issues (and is cleaner!).

I do not support this enhancement proposal because there are cleaner Pythonic methods to handle.  Option one, use interfaces :).  Option two, in the ~~module~~ block port definition use default values and handle accordingly in the block:

```python
@myhdl.block
def myblock(
    # required ports
    clock, reset, datain, fault,
    # optional ports
    dataout=None,
    # parameters
    num_iterations=10
):
    if dataout is None:
        # dummy signal that is not used
        dataout = Signal(datain.val)
``` 

The example @smldis provided is another example of handling not defined ports or passed `None`.  I like defining which ports are optional by using the default as `None` in the argument list (port list). 

If someone wants an `Open` in their specific projects there is nothing that should prevent it but I don't think it is something that needs to be included in the myhdl package.

josyb, 2016-03-23:
@cfelton: you may want to humour me and show me an example how *interfaces* handle the unused/open/unconnected signal? I mean cure/lift the warning, both in conversion by MyHDL, and later when running the generated V* code through the FPGA Vendor tool.

Your snippet:
```python
if dataout is None:
        # dummy signal that is not used
        dataout = Signal(datain.val)
```
doesn't seem right to me, unless ```None``` also works as a bit-bucket? Even if it works, does it get rid of the warnings?

The issue I depicted is that I (try to) code the *general* solution for the problem at hand. And sometimes to keep the 'code' readable I do create signals which are eventually not used as their final destination/output port is ```None```
Note that the actual code is/can be quite more involved that the simplified code in the first post.
I have locally implemented the above *QaD* and post-edit the generated VHDL file to remove those unused assignments, generating warning-free code ... 
My view: *Warnings should not exist, you either (optionally) inform that you handled the case or you issue an error*. I treat them as an error and rework the code to remove them.

cfelton, 2016-03-23:
@josyb the warning part was much further down in the description, I did not notice it as part of the enhancement request.  What I described will not remove the warnings.  

I think unused signals in the complex, modular, and scalable systems, we are implementing, are unavoidable.  My preference would be to remove the warning and add a `report_unused_signals` function.  

This is a difficult balance, personally I find it wasteful (as in implementation time) in a modular design to make sure all signals are removed and/or driven to remove warnings, when I simply want them removed because they are unused :)

josyb, 2016-03-23:
Good code can't be rushed, and the time spent generating warning free code will be gained tenfold later.

A ```report_unused_signals``` would hide the warnings in the MyHDL conversion perhaps, but they will still show up later in the Vendor's tools.

I implemented what I describe in my second post, but only for toVHDL() ...

cfelton, 2016-03-23:
> Good code can't be rushed, and the time spent generating warning free code will be gained tenfold later.

That's too generic of a statement to make much use of.  Well crafted modular (reusable) code fits that description exactly but could require lots of work to remove all the unused warnings.  From my experience, if you spent that time creating tests versus trying to remove the unused warnings you would be better off.

cfelton, 2016-03-23:
@josyb and yes, specifically to the warnings *interfaces* make the scenario worse :)  Because, it is the tendency to define the interface fully but many uses only use a subset of the interface.

cfelton, 2016-03-23:
@josyb also, I would update the description so it is clear in the first sentence or so that this enhancement request is specific to managing unused signal warnings, when I came back to this issue I had the impression it was a request to deal with unused ports (which there are other options).

"When creating re-usable modules we create output signals, which quite often are not all used in an instantiation and cause **undesirable warnings.**  The following discusses options to deal with the unused outputs and proposes and enhancement that can assist in dealing with the unused signals."

I still think the correct solution is the one posted by @smldis or removing unused signal warnings :)

josyb, 2016-03-23:
@cfelton Changed the description. (Per your suggestion above. Does this mean you see some merit in this discussion?)
The solution by @smldis (3rd post) is certainly correct. Just that I'm not a fan of duplicating logic in multiple processes. Also I reduced the complexity of the example, specifically for this discussion. In the original code there is another block of code that is activated when WRAP_AROUND is ```True``` doubling the code size.

Nicolasp38, 2016-03-24:
I agree with @josyb . The solution by @smldis forces to change the code structure. It is much more verbose and much less readable (and maintainable). 
@cfelton I don't understand why you are so opposed to a Open/Unused solution. You talk about efficiency : code your module without worrying if all outputs will be used then use your module as you need to (and specify unused outputs).
To my opinion, the solution to assign the output only when it is not None is not understandable by newcomers and quite hard to understand (because it is due to the way the parser works). It is not "natural" : I assign a value to a signal but I have to check if it not None ? If I assign it's value it can't be None. So what's the trick ?

Nicolasp38, 2016-03-24:
What about "external" libraries like the Xilinx one ?
You recreate a Python version of a component for simulation/structure parsing and specify the VHDL or Verilog template. To guide the parser, you use "output.driven=True". It is not possible when assigning None to a port of a module instance.

josyb, 2016-03-24:
@Nicolasp38: @cfelton is not opposed, only not (yet) convinced :)
And currently @cfelton is not specifying unused outputs, he just lives with the warnings, focusing on tests instead.
Dealing with unused output ports like we do in VHDL is not straightforward (IMHO).

Nicolasp38, 2016-03-25:
@josyb I agree with you. I don't like to see warnings in logs. Some warnings can be considered as errors.

cfelton, 2016-03-25:
My position has soften (slightly) because I thought this originally had to do with managing unused port vs. the **warnings created by unused signals**.  I am still unconvinced, as systems become more complex, that we can efficiently manage unused signals manually.  Personally, when I don't use a signal I want the unused signal removed (it's unused) and I don't want to manually have to mark them all as unused.

My preferred approach would be to remove unused warnings and add a `inst.report_usused()` and allow users to treat it as an error `assert inst.report_unused() is None` if they prefer.  

OK, now what If you want them to be an error - what tools should be available to help manage the unused?  I don't know, because I favor a different approach, but the current proposal only addresses one type of unused signals so I don't think it is worth pursuing.  In addition, I still think there are elaboration methods, for the folks who prefer unused warnings to “take the time and craft their code" to circumvent the warnings.

I understand (I think I do) that the evolution of having no unused signals extends from the time when we created specific designs (i.e. non-modular or reusable) and we have tried to perpetuate this forward.  My opinion, as designs grow in complexity, interfaces with numerous signals, ~~modules~~ blocks supporting various status signals, and conditional generator blocks; that it will be to unwieldily to manually manage all the unused signals.  I feel, if your tests fail to detect the errors you have bigger issues.

VHDL and Verilog often provide the guidance for many topics but in this case I think looking towards Python approaches will serve us better.

Nicolasp38, 2016-03-29:
@cfelton This is exactly because designs become bgiger and bigger that we need a way to manage warnings. When you have 10 warnings in a design, you can check them each time you make a change to see if new ones have appeared or replaced another one. When you have hundreds or thousands of warnings, you can't do any check anymore. When you explicitly write that a signal is not used, then you don't have to worry about it anymore and you can focus on useful things.

josyb, 2016-03-29:
@cfelton 
> I am still unconvinced, as systems become more complex, that we can efficiently manage unused signals manually.

In my case I only have to treat unused signals at the lowest level: my *building blocks*, and while developing such *blocks* I always take care of possibly unused output ports (those possibly set to ```None```).

> My preferred approach would be to remove unused warnings and add a inst.report_usused() and allow users to treat it as an error assert inst.report_unused() is None if they prefer.

Silently dropping the warnings and making the user call up the report is the wrong way round. It rather should be: you don't care of warnings, so you turn it off, e.g. : ```dtc.warnings( driven_not_read = False)```

> OK, now what If you want them to be an error - what tools should be available to help manage the unused? I don't know, because I favor a different approach, but the current proposal only addresses [...]

The *driven, not read* is actually a *benign* warning. Some other warnings, like *read, not driven* are more severe (and should be an error, rather?) and have to be taken care of by changing the *hdl* code.
It is certainly possible to write the code in such a way that no *unused signals* are generated, as @smldis has shown, but this may come at the cost of poor readability and increased maintenance. 

> I understand (I think I do) that the evolution of having no unused signals extends from the time when we created specific designs (i.e. non-modular or reusable) and we have tried to perpetuate this [...]

I beg to differ... I have always created blocks with re-usability in mind, and I'm sure 99% of MyHDL aficionados will do so.Treating *unused signals* at the lowest level is easy as there shouldn't be that many.  The complex outer/higher level designs are mostly structural, so there shouldn't arise many *unused signals* either.
 In VHDL this is easy enough, as you use the explicit ```open``` keyword on component instantiation. But the obvious Python counterpart ```None``` doesn't work as such, and hence it is a little bit more work in MyHDL (using conditional generators to do the assignment). IMHO certainly manageable. A true equivalent of the VHDL ```open``` keyword, as @Nicolasp38 suggests, would be nicer, though.

>  I feel, if your tests fail to detect the errors you have bigger issues.

I'm not following here ...

Another remark: currently MyHDL writes warnings to **stderr** (painting the message in red)  so if you call the MyHDL *module* by an external program, this will likely fail.

smldis, 2016-03-29:
Hi all, it is great to see that this discussion is a source of a lot of practical considerations.
My "solution" was trying to use current MyHDL capabilities, the reason was philosophical: since MyHDL wants to be able to generate HDL code, we should always be able to conditionally generate a portion of Hardware. Something like open signals and unused generated code are kind of bad practice.

Many of you pointed out obvious practical problems (eg. code maintainability, warnings checking, etc..)
Let's start thinking about how to enhance MyHDL to deal with those problems.

For what I know MyHDL modules are functions with arguments, arguments can be signals, interfaces or other parameters for controlling the code generation.
I see a problem here, None or Open() (see @Nicolasp38 post) are not part of these categories.
I would propose a new element, let me use the name Connector ("please suggest a better name").
A Connector is a hardware element that can be an open circuit, a probe or other things. 
It can be used instead of a signal when the replaced signal is the wire between different modules.

Could this bare definition provide a starting point to address all the possible future problems? 
(We would have now modules, signals, interfaces and Connectors)

Regards

NicoPy, 2016-11-07:
Here is what I use (tested on a small VHDL project)  : 

``` python
class Open():
    def __init__(self):
        self.next = None
        self._name = ""
 
    # For conversion
    def __str__(self):
        return "open"

comp = my_component(i_clk, i_data, Open(), o_data)
```