myhdl/myhdl #127: Python 2 <-> 3 difference: Signal in multiple list is not supported

ThorimMB, 2015-09-21:
The code in https://gist.github.com/ThorimMB/00a3a5e51d97ea980015 raises a "Signal in multiple lists" conversion error on Python 3. It converts perfectly with Python 2.7 and compiles in Quartus:

File "bitonic.py", line 151, in convert: toVHDL(array8sorter, *sigs)
File "myhdl/conversion/_toVHDL.py", line 170, in __call__: siglist, memlist = _analyzeSigs(h.hierarchy, hdl='VHDL')
File "myhdl/myhdl/conversion/_analyze.py", line 128, in _analyzeSigs: raise ConversionError(_error.SignalInMultipleLists, s._name)
myhdl.ConversionError: Signal in multiple list is not supported: sort_merge_hi_merge_list_a(0)

I think it is not good if some parts of code only convert in a special Python version ;)

jandecaluwe, 2015-09-27:
Mm, looks like list comprehensions are now considered nested functions in Python 3. Sigh.

jandecaluwe, 2015-09-27:
Fixed by looking up cellvars in a more specific way, inspired by simular code for attribute transormation.

closed 2015-09-27 by jandecaluwe

cfelton, 2015-09-28:
@jandecaluwe, this has happened a couple times where the examples from the website break with new code/releases.  I think we should include the examples in our regression (travis-ci).  I think this would require creating/including the complete example and test in the site-myhdl repo.  Then in the regression script the site repo would be cloned and the examples ran.

If this sounds reasonable and no objections I can create an issue and try to get one of the newer contributors to take on the task (should be a straightforward task)?

jandecaluwe, 2015-09-29:
I am not in favour of adding external repos to the travis regression test suite. If a test from site-myhdl breaks, because of a migration or because an example was pushed with an issue, someone will still have to turn it into an issue and fix it. But until that happens, all other progress is stalled. Instead, I want to use the regression suite to guide my maintenance/development, and as a check to accept PRs.

If examples regularly fail, it means that our unit test suite is not strong enough and we should fix it with priority, but adding those issues to the unit tests. Actually, that is what I am focussing on when I have time. Perhaps understandably, not too many people seem interested in solving bugs (instead of working on new features), but I think improving quality is what is needed most first. We have been making good progress lately, using Python3 as a catalyst and not just a problem, but as can be seen from the present issue, subtle conceptual issues may still pop up.

Instead, I would encourage everyone to go over the examples and any other code, turn any issue into a clear description and a test case that is as small as possible, and report it here. That is great help, probably half the work. I always add those test cases to the unit tests. As this is a test that fails before and not afterwards, it is fully in line with test -driven development. We are sure that such a test is meaningful and that it tests something significant that has been overlooked before.

cfelton, 2015-09-29:
@jandecaluwe fair enough, I agree the more test the better and I agree the test suite should have full coverage.  It makes sense not to clutter the myhdl issues with example issues.

commit 0bc41cc0 (2015-09-27): Fix #127
Lookup cellvars in a more direct way