myhdl/myhdl #128: ConcatSignal error on negative values

gw0, 2015-09-29:
`ConcatSignal()` crashes on handling negative values. It would make sense that `ConcatSignal()` should not handle boundary checks at all.

Example code and error:

```bash
def bounds_err():
    a_list = [Signal(intbv(-1, min=-8, max=8)), Signal(intbv(-1, min=-8, max=8))]
    a_vec = ConcatSignal(*a_list)
    @always(delay(1))
    def foo():
        a_vec[1].next = 0
    Simulation((foo,)).run()
```

```
Traceback (most recent call last):
  File "xxx.py", line 89, in <module>
    bounds_err()
  File "xxx.py", line 88, in bounds_err
    Simulation((foo,)).run()
  File "zzz/venv/src/myhdl/myhdl/_Simulation.py", line 135, in run
    waiter.next(waiters, actives, exc)
  File "zzz/venv/src/myhdl/myhdl/_Waiter.py", line 189, in next
    clauses = next(self.generator)
  File "zzz/venv/src/myhdl/myhdl/_ShadowSignal.py", line 176, in genfunc
    newval[hi:lo] = a
  File "zzz/venv/src/myhdl/myhdl/_intbv.py", line 180, in __setitem__
    self._handleBounds()
  File "zzz/venv/src/myhdl/myhdl/_intbv.py", line 81, in _handleBounds
    (self._val, self._min))
ValueError: intbv value -1 < minimum 0
```

gw0, 2015-09-29:
I am not sure what exactly made it work, but in case someone needs a workaround he could use the following as a starting point.

1. I have been experimenting with the `fixbv` type from @cfelton, but needed to patch a few things it a little. Check out my commits in <https://github.com/gw0/myhdl>.

2. If you want to use a list of Signals individually (read-write), but pass them as input to the `Foo` model as a combined shadow signal (read-only), use:

    ```python
    a_list = [ Signal(fixbv(0.0, min=fix_min, max=fix_max, res=fix_res)) for _ in range(dim) ]
    a_vec = ConcatSignal(*reversed(a_list))
    foo = Foo(out, a_vec, dim)
    ```

3. If you then want to access (read) them individually again simply as `a_list[j]` in `Foo` do the following:

    ```python
    a_list = [ Signal(fixbv(0.0, min=fix_min, max=fix_max, res=fix_res)) for j in range(dim) ]
    for j in range(dim):
        a_list[j].assign(a_vec((j + 1) * fix_width, j * fix_width))
    ```

4. The above won't work if you want to output a combined signal from `Bar` model, because the shadow signal is read-only. In this case define it the other way around to have a combined Signal (read-write), but access them as shadow slices individually (read-only):

    ```python
    a_vec = Signal(intbv(0)[dim * fix_width:])
    a_list = [ Signal(fixbv(0.0, min=fix_min, max=fix_max, res=fix_res)) for j in range(dim) ]
    for j in range(dim):
        a_list[j].assign(a_vec((j + 1) * fix_width, j * fix_width))
    bar = Bar(a_vec, dim)
    ```

5. Accessing (writing) them individually in `Bar` is ugly (directly store bits at correct offset), but can be done with:

    ```python
    val = fixbv(12.3, min=fix_min, max=fix_max, res=fix_res)
    y_dcontext_vec.next[(j + 1) * fix_width:j * fix_width] = val[:]
    ````

cfelton, 2015-10-06:
@gw0 If I understand there is no issue with `ConcatSignal` and negative values, correct?

Example:
```
In [39]: for ii in range(-4, 4):
    ...:     x, y = [Signal(intbv(ii, min=-4, max=4)) for _ in range(2)]
    ...:     z1 = ConcatSignal(x, y)
    ...:     w = Signal(intbv((-1*ii)-1, min=-4, max=4))
    ...:     z2 = ConcatSignal(w, x, y)
    ...:     print("{} {}     = {}".format(bin(x, 3), bin(y, 3), bin(z1, 6)))
    ...:     print("{} {} {} = {}".format(bin(w, 3), bin(x, 3), bin(y, 3), bin(z2, 9)))
    ...:     
100 100     = 100100
011 100 100 = 011100100
101 101     = 101101
010 101 101 = 010101101
110 110     = 110110
001 110 110 = 001110110
111 111     = 111111
000 111 111 = 000111111
000 000     = 000000
111 000 000 = 111000000
001 001     = 001001
110 001 001 = 110001001
010 010     = 010010
101 010 010 = 101010010
011 011     = 011011
100 011 011 = 10001101
```
This issue can be closed. 

@gw0, you might want to follow/review #108.  It outlines some of the current progress with the `fixbv`.  I will be pushing the updates to [MEP111](http://dev.myhdl.org/meps/mep-111.html) and  changes to the `fixbv` soon (soon is relative :)

@gw0, thanks for pointing out a couple issues in the current `fixbv` example implementation, I will integrate those and extend the tests.  I am not sure about the workarounds you did for negative, for low-level bit manipulations everything should be handled by base `intbv`,

josyb, 2015-10-06:
@cfelton : @gw0's code definitely fails ...
In \_ShadowSignal.py line 176:
```   newval[hi:lo] = a```
could be changed to:
```    newval[hi:lo] = a if a >= 0 else 2**w + a```
to represent a negative value in its _unsigned 2's complement_ form.

Is it possible that your code does not fail because your are not simulating it and thus the ```_handlebounds()``` doesn't get called?

cfelton, 2015-10-06:
@josyb thanks, I misunderstood where the issue was occurring and overlooked the original example.  @gw0 what did you mean in the second comment "I am not sure what exactly made it work"?

cfelton, 2015-10-06:
@josyb A question I have, should a *slice* always be handled as 
unsigned or not?

The following fails as well:
```
w = intbv(-2, min=-8, max=8)
x = intbv(-3, min=-8, max=8)
y = intbv(0)[8:]
print(bin(w, 4), bin(x, 4), bin(y, 8))
y[8:4] = w
y[4:0] = x
print(bin(w, 4), bin(x, 4), bin(y, 8)
```

Is the above failure expected or not, and if so how does it
differ from concatenation?

The bit `concat(w, x)` fails also for negative values.  What 
is the overall policy of mixed (signed and unsigned) 
concatenation?  Is it this way for a reason (I don't know/recall)?

josyb, 2015-10-06:
@cfelton The mechanism for failure is the same as for concatenation. In concatenation we think in _bits_ (as we do when concatenating VHDL std_logic_vectors), but in MyHDL it passes via the integer (or natural) value stored in the intbv object. So your example ```y[8:4] = w``` fails because that value ends up being negative.
If we draw a parallel with VHDL: there we concatenate signed and unsigned objects into std_logic_vectors, or slice signed and unsigned objects from std_logic_vectors, as these std_logic_vectors  are used for _transport_ between IP-blocks (mandatory for Qsys, and very probably also for IP Integrator). But we need to cast to an from extensively ...

I recall reading about _unsignedness_ in concatenation: an extract from the docs
> In some cases it can be useful to interpret intbv objects in terms of “signed” and “unsigned”. Basically, it depends on attribute min. if min < 0, then the object is “signed”, otherwise it is “unsigned”. In particular, the bit width of a “signed” object will account for a sign bit, but that of an “unsigned” will not, because that would be redundant. From earlier sections, we have learned that the return value from a slicing operation is always “unsigned”.  

If we regard concatenation to be the opposite of slicing, its nature is _unsigned_ too.

Your code 
```python
w = intbv(-2, min=-8, max=8)
x = intbv(-3, min=-8, max=8)
y = intbv(0)[8:]

y[8:4] = w
y[4:0] = x
```
is actually identical to what happens inside ConcatSignal():
ConcatSignals tallies the bitwidths of the arguments and creates an _unsigned_ intbv of that width and then proceeds with the individual assignments.

Expect Jan to give a clearer explanation! ;)

gw0, 2015-10-07:
@cfelton I was experimenting and hacking things until they worked, therefore I am not sure what exactly made the negative values issue work and I unfortunately do not have much time to investigate it. You could look at what changes (few last commits) I made in my `mep111_fixbv` branch (https://github.com/gw0/myhdl/tree/mep111_fixbv).

Maybe explicitly "casting" any type into bits should be the way to go:

```python
y.next[8:4] = w[:]
y.next[4:0] = x[:]
x.next[2:0] = w[2:0]
```

cfelton, 2015-10-07:
@gw0, thanks for the clarification, wasn't sure if it started working for you in your modified code or a different example.  Note `fixbv` isn't released yet so expect issues and changes.  I will gladly incorporate others fixes modifications.

@josyb, @gw0, I think it needs to be decided if the slices are always handled as unsigned (setting and getting) or if then need a cast.  Currently, all the concats works the same, if you set a negative number it fails, maybe this is intended?

josyb, 2015-10-07:
@cfelton I think sticking to the docs that a slice is always unsigned is the correct way to go. A slice is a _neutral_ collection of bits, and _implicitly declaring_ these as unsigned is _harmless_. If the target is a signed signal you have to use the .signed() cast.
```python
ssig = [Signal( intbv( 0, min = -2**(n-1), max = 2**(n-1)) for _ in range(m)]
lv = Signal( intbv(0)[m*n:]
for i in range( m ) :
    ssig[i].next = lv[(i+1) * n : i * n].signed() 
```
----   
Concatenating Signals should always work, i.e. failing on negative values is a **bug** and should be repaired. There isn't a way to cast a signed intbv to an unsigned intbv. IMO ConcatSignal() and concat() should handle this under the hood.
@gw0's explicit cast [:] won't work as e.g. w[:] returns an new intbv set to the same (negative) value

jandecaluwe, 2015-10-10:
The original example is confusing and not suitable for inclusion in the bug database. You cannot assign to a ConcatSignal, and indexing it and then using it as target in a Signal assigment doesn't work either. The reported issue occurs earlier at initialization time. I have added a test case in the core regression suite and fixed it.

closed 2015-10-10 by jandecaluwe

GProtoZeroW, 2018-03-29:
Sorry Y'all but I am reopening this because `concat` does have an issue with concatenating two negative numbers that I discovered redoing the complex multiplier example. The issue appears to be that  if the leading term in `concat` is neg when concat forms it is setting a min of 0 when it really should have no min/max instead  just a bit width 

Exsamples:
```
Max=(2**(16-1)-1); Min=-2**(16-1)
print(Min, Max)

aPos=intbv(1, min=Min, max=Max)
aZero=intbv(0, min=Min, max=Max)
aNeg=intbv(-1, min=Min, max=Max)
print(f"""BitWidths:
aPos: {aPos._nrbits}
aZero: {aZero._nrbits}
aNeg: {aNeg._nrbits}
""")

bPos=intbv(89, min=Min, max=Max)
bZero=intbv(0, min=Min, max=Max)
bNeg=intbv(-89, min=Min, max=Max)
print(f"""BitWidths:
bPos: {bPos._nrbits}
bZero: {bZero._nrbits}
bNeg: {bNeg._nrbits}
""")

#aPosbPos
PP=concat(aPos, bPos)
print(f"""PP:
PP Val: {PP}
PP Signed: {PP.signed()}
PP Bits: {PP._nrbits}
PP Max: {PP.max}
PP Min: {PP.min}
Pass
"""
)

#aPosbNeg
PN=concat(aPos, bNeg)
print(f"""PN:
PN Val: {PN}
PN Signed: {PN.signed()}
PN Bits: {PN._nrbits}
PN Max: {PN.max}
PN Min: {PN.min}

"""
)

#aZerobNeg
ZN=concat(aZero, bNeg)
print(f"""ZN:
ZN Val: {ZN}
ZN Signed: {ZN.signed()}
ZN Bits: {ZN._nrbits}
ZN Max: {ZN.max}
ZN Min: {ZN.min}
Pass
"""
)
```
And the error:
```
#aZerobNeg
NN=concat(aNeg, bNeg)
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-81-a59978089a19> in <module>()
      1 #aZerobNeg
----> 2 NN=concat(aNeg, bNeg)

~/anaconda3/lib/python3.6/site-packages/myhdl/_concat.py in concat(base, *args)
     80 
     81     if basewidth:
---> 82         return intbv(val, _nrbits=basewidth + width)
     83     else:
     84         return intbv(val)

~/anaconda3/lib/python3.6/site-packages/myhdl/_intbv.py in __init__(self, val, min, max, _nrbits)
     60             raise TypeError("intbv constructor arg should be int or string")
     61         self._nrbits = _nrbits
---> 62         self._handleBounds()
     63 
     64     # support for the 'min' and 'max' attribute

~/anaconda3/lib/python3.6/site-packages/myhdl/_intbv.py in _handleBounds(self)
     79             if self._val < self._min:
     80                 raise ValueError("intbv value %s < minimum %s" %
---> 81                                  (self._val, self._min))
     82 
     83     def _hasFullRange(self):

ValueError: intbv value -89 < minimum 0
```
For DSP usage of myHDL concat should not have a min or max on its returned or it ignores the sign but jeeps the sign bit of the leading argument when it forms its result. And that if it is later sliced 
back up the signed values (including the signed MSB) should be recoverable either through enhancements to myHDL or a clearly documented means with confirmed Verilog/VHDL synthesis of those means

josyb, 2018-03-29:
@GProtoZeroW I (re-)studied the *_concat.py* code, and you are right it doesn't handle a leading negative signed *intbv*
So it still is a**bug** (as it was back in 2015).
Looks like very few people are doing *signed*

GProtoZeroW, 2018-04-13:
I redid and expanded the old complex multiplier example with a bunch of testing of the `concat` function showing its issue in handling negative numbers and conversion to verilog and vhdl.  The notebook is here https://github.com/PyLCARS/PythonUberHDL/blob/master/myHDL_DigitalSignalandSystems/ComplexMultiplier.ipynb it should serve as a benchmark/test bench to get `concat` working as needed

josyb, 2018-04-13:
If you change the code in _concat,py to:
```python
def concat(base, *args):

    if isinstance(base, intbv):
        basewidth = base._nrbits
#         val = base._val
#         val = base._val - (2 ** base._nrbits if base._val & (1 << (base._nrbits - 1)) else 0)
        val = (2 ** base._nrbits + base._val) & (2 ** base._nrbits - 1)
    elif isinstance(base, integer_types):
        if isinstance(base, bool):
            basewidth = 1
        else:
            basewidth = 0
        val = base
    elif isinstance(base, _Signal):
        basewidth = base._nrbits
        if isinstance(base._val, intbv):
            val = base._val._val
        else:
            val = base._val
    elif isinstance(base, str):
        basewidth = len(base)
        val = long(base, 2)
    else:
        raise TypeError("concat: inappropriate first argument type: %s"
                        % type(base))

    width = 0
    for i, arg in enumerate(args):
        if isinstance(arg, intbv):
            w = arg._nrbits
#             v = arg._val
            v = (2 ** arg._nrbits + arg._val) & (2 ** arg._nrbits - 1)
        elif isinstance(arg, _Signal):
            w = arg._nrbits
            if isinstance(arg._val, intbv):
#                 v = arg._val._val
                v = (2 ** arg._nrbits + arg._val_val) & (2 ** arg._nrbits - 1)
            else:
                v = arg._val
        elif isinstance(arg, bool):
            w = 1
            v = arg
        elif isinstance(arg, str):
            w = len(arg)
            v = long(arg, 2)
        else:
            raise TypeError("concat: inappropriate argument type: %s"
                            % type(arg))
        if not w:
            raise TypeError("concat: arg on pos %d should have length" % (i + 1))
        width += w
        val = val << w | v & (long(1) << w) - 1

    if basewidth:
        return intbv(val, _nrbits=basewidth + width)
    else:
        return intbv(val)
```
This problem will be gone.
```python
def printcs(name , v):
    print("{}: Val: {} Signed: {} Bits: {} Max: {} Min: {}  Pass".format(name, v, v.signed(), v._nrbits, v.max, v.min))

Max = (2 ** (16 - 1) - 1); Min = -2 ** (16 - 1)
print('Min {}, Max {}'.format(Min, Max))

aPos = intbv(1, min=Min, max=Max)
aZero = intbv(0, min=Min, max=Max)
aNeg = intbv(-1, min=Min, max=Max)
print("BitWidths: aPos: {} aZero: {} aNeg: {}".format(aPos._nrbits, aZero._nrbits, aNeg._nrbits))

bPos = intbv(89, min=Min, max=Max)
bZero = intbv(0, min=Min, max=Max)
bNeg = intbv(-89, min=Min, max=Max)
print("BitWidths: bPos: {} bZero: {} bNeg: {}".format(bPos._nrbits, bZero._nrbits, bNeg._nrbits))

printcs('PP', concat(aPos, bPos))
printcs('PN', concat(aPos, bNeg))
printcs('PZ', concat(aPos, bZero))

printcs('NN', concat(aNeg, bNeg))
printcs('NP', concat(aNeg, bPos))
printcs('NZ', concat(aNeg, bZero))

printcs('ZZ', concat(aZero, bZero))
printcs('ZP', concat(aZero, bPos))
printcs('ZN', concat(aZero, bNeg))
```

```
Min -32768, Max 32767
BitWidths: aPos: 16 aZero: 16 aNeg: 16
BitWidths: bPos: 16 bZero: 16 bNeg: 16
PP: Val: 00010059 Signed: 00010059 Bits: 32 Max: 4294967296 Min: 0  Pass
PN: Val: 0001ffa7 Signed: 0001ffa7 Bits: 32 Max: 4294967296 Min: 0  Pass
PZ: Val: 00010000 Signed: 00010000 Bits: 32 Max: 4294967296 Min: 0  Pass
NN: Val: ffffffa7 Signed: ffffffa7 Bits: 32 Max: 4294967296 Min: 0  Pass
NP: Val: ffff0059 Signed: ffff0059 Bits: 32 Max: 4294967296 Min: 0  Pass
NZ: Val: ffff0000 Signed: ffff0000 Bits: 32 Max: 4294967296 Min: 0  Pass
ZZ: Val: 00000000 Signed: 00000000 Bits: 32 Max: 4294967296 Min: 0  Pass
ZP: Val: 00000059 Signed: 00000059 Bits: 32 Max: 4294967296 Min: 0  Pass
ZN: Val: 0000ffa7 Signed: 0000ffa7 Bits: 32 Max: 4294967296 Min: 0  Pass
```

But I guess you will return later when slicing the concatenated intbv back into real and imaginary?

GProtoZeroW, 2018-04-13:
What does "But I guess you will return later when slicing the concatenated intbv back into real and imaginary?" mean. And what about the conversion issues?

josyb, 2018-04-13:
What you actually need is a ```[record | struct]``` to represent a complex value.
I guess you could also use a ```list of signals``` (in this case ```[Re, Im]```) to feed complex values internally from function to function? Then you would only have to do all this juggling for the top level ports.

GProtoZeroW, 2018-04-14:
@josyb , you and anyone are more than welcome to take those notebooks on PyLCARS and make them better. In fact, the whole idea was to create a better teaching resource for myHDL and other python tools for engineering. I mean it took me a year to find the adder example on Rosetta code that allowed me to figure out how to use myHDL. Hell, the end goal is to make python LCARS as seen in Star Trek.