myhdl/myhdl #134: Interface signal argument name collision

nturley, 2015-11-08:
When you pass a signal from an interface into a function, the signal will take the argument name, rather than keeping the original name that the interface gave it. If the function is called multiple times, then this will cause name collisions.

for instance, this
```python
from myhdl import *

class Bus:
    def __init__(self):
      self.a = Signal(bool(False))
      self.b = Signal(bool(False))

def invert(sigin, sigout):
    @always_comb
    def foo():
        sigout.next = not sigin
    return foo

def top(busin, busout):
    inverta = invert(busin.a, busout.a)
    invertb = invert(busin.b, busout.b)
    return inverta, invertb

busin = Bus()
busout = Bus()
toVerilog(top, busin, busout)
```
will produce this
```verilog
module top (
    sigout,
    sigout,
    sigin,
    sigin
);

output sigout;
wire sigout;
output sigout;
wire sigout;
input sigin;
input sigin;

assign sigout = (!sigin);
assign sigout = (!sigin);

endmodule
```

ideally, in this example the signals would be named busin_a, busin_b, busout_a, and busout_b. Another solution is just to run them through a uniquifier (sigin, sigin_0, sigout, sigout_0).

josyb, 2015-11-08:
If you change the code from a _combinatorial_ process into a _sequential_ process, the _double use_ of ```sigout``` and ```sigin``` is remedied, but you are confronted with another side-effect of the interface re-naming action, see my complaints in issue #125 

```python
from myhdl import *

class Bus:
    def __init__(self):
        self.a = Signal(bool(False))
        self.b = Signal(bool(False))

def invert(Clk, sigin, sigout):
    @always_seq(Clk.posedge, reset =  None)
    def foo():
        sigout.next = not sigin
    return foo

def top(Clk, busin, busout):
    inverta = invert(Clk, busin.a, busout.a)
    invertb = invert(Clk, busin.b, busout.b)
    return inverta, invertb

Clk = Signal( bool( 0 ))
busin = Bus()
busout = Bus()
toVerilog(top, Clk, busin, busout)
toVHDL(top, Clk, busin, busout)
```
```verilog
module top (
    Clk,
    inverta_sigout,
    invertb_sigout,
    inverta_sigin,
    invertb_sigin
);

input Clk;
output inverta_sigout;
reg inverta_sigout;
output invertb_sigout;
reg invertb_sigout;
input inverta_sigin;
input invertb_sigin;

always @(posedge Clk) begin: top_inverta_foo
    inverta_sigout <= (!inverta_sigin);
end

always @(posedge Clk) begin: top_invertb_foo
    invertb_sigout <= (!invertb_sigin);
end

endmodule
```

cfelton, 2015-11-08:
@josyb the renaming of top-level interface ports is a known limitation, any improvements will be treated as enhancements / new features.  The issue @nturley demonstrated is a bug and we should try and address the issue.

josyb, 2015-11-08:
@cfelton I just tried to help by pointing out that the _@always_seq_ hasn't got the issue and give @jck a clue.
Please forgive me that I kept repeating myself on the _renaming_ issue, but to me this limitation is a real show-stopper.

nturley, 2015-11-09:
Here's what I have so far: It's not the fact that it's sequential or combinatoric, it's actually the presence of other signals.  Line 92 of _analyze checks if there are any signals in the instances sigdict. If it doesn't detect any signals in the instance, it skips it (I'm guessing as an optimization). This example has no signals, only interfaces/classes, so both inverta and invertb act like they are on the top level of the hierarchy.

Proposed Solutions:
1. Uniquify every signal before adding it to the list (by appending an underscore to the name or something)
2. Remove the optimization.
3. search the interfaces for members of type signal.

1 seems like the easiest and safest solution. It could also fix issue 95.
3 might be useful research on a fix/enhancement for issue 125.

josyb, 2015-11-09:
I checked my trace.log and it is not that line 92 wreaking havoc ... as I see different names passing by depending or not whether the interface objects have accompanying Signals (as your remark ,that any Signal would do, probably holds).

cfelton, 2015-11-09:
@nturley small hint, if you use the "#" before the issue github will automatically include links to the issue or PR.  You referenced issue #95 and #125 in your comment.

nturley, 2015-11-09:
@cfelton thanks. I'm still new to this whole github thing.