myhdl/myhdl #138: toVerilog fails to detect that a signal is driven by user-defined code

xesscorp, 2015-11-17:
The code:

```python
from myhdl import *

def pass_thru(i, o):
    @always_comb
    def in_to_out():
        o.next = i
        
    return instances()

def top(i, o):
    ii = Signal(bool(0))
    inst = pass_thru(i, ii)
    @always_comb
    def logic():
        o.next = ii
        
    return instances()

if __name__ == '__main__':
    i, o = [Signal(bool(0)) for _ in range(2)]
    toVerilog(top, i, o)

```

generates the following (correct) Verilog code:

```verilog
// File: top.v
// Generated by MyHDL 0.9.0
// Date: Tue Nov 17 00:34:34 2015

`timescale 1ns/10ps

module top (
    i,
    o
);

input i;
output o;
wire o;
wire ii;

assign ii = i;
assign o = ii;

endmodule
```

But changing the code to use a user-defined block like so:

```python
from myhdl import *

def pass_thru(i, o):
    @always_comb
    def in_to_out():
        o.next = i
        
    return instances()

pass_thru.verilog_code = \
"""
SB_IO inp (
.PACKAGE_PIN($i),
.D_IN_0($o)
);
"""

def top(i, o):
    ii = Signal(bool(0))
    inst = pass_thru(i, ii)
    @always_comb
    def logic():
        o.next = ii
        
    return instances()

if __name__ == '__main__':
    i, o = [Signal(bool(0)) for _ in range(2)]
    toVerilog(top, i, o)
```

Results in the following (incorrect) Verilog code:

```verilog
// File: top.v
// Generated by MyHDL 0.9.0
// Date: Tue Nov 17 00:37:08 2015

`timescale 1ns/10ps

module top (
    i,
    o
);

input i;
output o;
wire o;
wire ii;

assign ii = 0;

SB_IO inp (
.PACKAGE_PIN(i),
.D_IN_0(ii)
);

assign o = ii;

endmodule
```

The ii signal is forced to 0 because the Verilog converter doesn't realize the SB_IO primitive will drive ii with the value found on the i input. The circuit synthesized from the Verilog does not function correctly.

cfelton, 2015-11-17:
@xesscorp The user-defined code needs the user to supply a little more information, as [described in the manual](http://docs.myhdl.org/en/stable/manual/conversion_examples.html#conv-usage-custom).  As you point out, the converter needs to know that the output of the module, that is being user-defined, is being driven. 

Adding `o.driven = True` or `o.driven = 'wire'` will resolve the issue.

xesscorp, 2015-11-17:
Yeah, sorry, missed that.

cfelton, 2015-11-17:
@xesscorp no problem.  @jandecaluwe this issue can be closed.

closed 2015-11-17 by jandecaluwe