myhdl/myhdl #143: Ram instanciation syntax enhancement

Nicolasp38, 2016-01-08:
When declaring a list for use as RAM, the following syntax works :
    ram = [Signal(intbv(0)[24:]) for i in range(addra.max)]

However, the following syntax, which is much more convenient and equivalent, does not work :
    ram = [Signal(intbv(0)[24:])] * addra.max

jandecaluwe, 2016-01-08:
They are not equivalent at all, the second syntax is definitely not what you want.

Nicolasp38, 2016-01-08:
Can you elaborate on this ?

jandecaluwe, 2016-01-08:
http://www.jandecaluwe.com/blog/its-a-simulation-language.html

NagabhushanS, 2016-01-14:
Exactly, the second statement instantiates the list with all the elements being the same object. So, if we  change any element, all the elements of the list will be modified.

closed 2016-01-14 by jandecaluwe