myhdl/myhdl #151: Broken link

kiddick, 2016-02-29:
On this page [Google Summer of Code 2016](http://dev.myhdl.org/gsoc/gsoc_2016.html) there is the broken link
>The following is a list of additional examples and projects that can be used as first exercises and references:

    *MyHDL Examples*
    MyHDL List of Resources

http://www.myhdl.org/examples/

cfelton, 2016-02-29:
Thanks for pointing this out.

This repository is specifically for the myhdl code base.  Side problems can be created on the [site repo](https://github.com/myhdl/site-myhdl-dev)

kiddick, 2016-02-29:
I've mentioned about that issue on site repo https://github.com/jandecaluwe/site-myhdl-dev-retired/issues/2

cfelton, 2016-03-01:
@kiddick that is the wrong repo, whey the "retired", the correct repo (as linked in the first comment) is https://github.com/myhdl/site-myhdl-dev

kiddick, 2016-03-01:
@cfelton but there is no section "issues" in forked repos on github :smile:

cfelton, 2016-03-01:
@kiddick good point, fair enough.  I have updated the link.  Will need to figure out future process.

@jandecaluwe this issue can be closed.

closed 2016-03-01 by jandecaluwe