myhdl/myhdl #161: [enhancement] limited top-level list-of-signal port conversion

cfelton, 2016-04-01:
# [enhancement] limited top-level list-of-signal port conversion

Currently, list-of-signals in the top-level is not supported for conversion.  This issue/enhancement proposes that a subset of a list-of-signals, essentially those whose overall dimension remain 1, support conversion.  In other words, the conversion compiler would convert list-of-signals of type bool (one-dimension) to a bit-vector in the target HDL. 

There are occasions when a bit-vector (list-of-signals of type bool) or a bank of tristates are desired and/or needed.  Currently, there is no conversion process to support individual control of a  `TristateSignal` with more than one bit.  One such scenario is described in [a seven-segment implementation](https://hackaday.io/project/7982-cat-board/log/27492-driving-an-array-of-led-digits-with-the-cat-board).

The proposal is to support conversion of top-level list-of-signal ports whose types are `bool`.

#### Example: top-level list-of-signal-of-bools
```python
def list_of_bools(x, y):
    nbits = len(x)
    @always_comb
    def assign():
        for ii in range(nbits):
            y[ii].next = x[ii]
    return assign

x, y = [[Signal(bool(0)) for _ in range(27)]
         for __ in range(2)]
toVerilog(list_of_bools, x, y)
toVHDL(list_of_bools, x, y)
```

#### Example: top-level list-of-signal-of-tristates-of-bools
```python
def list_of_tristate(tri, infromtri, outtotri):
""" drive the output when `outtotri` is 0 """
    nbits = len(tri)
    tdrv = [tri[ii].driver() for ii in range(nbits)]
    print(type(tdrv), tdrv)
    
    @always_comb
    def rtl_in():
        for ii in range(nbits):
            if tri[ii] == 0:
                infromtri.next[ii] = 0
            else:
                infromtri.next[ii] = 1
        
    @always_comb
    def rtl_out():
        for ii in range(nbits):
            if outtotri == 0:
                tdrv[ii].next = 0
            else:
                tdrv[ii].next = None

    return rtl_in, rtl_out

tri = [TristateSignal(bool(1)) for _ in range(8)]
ift = Signal(intbv(0)[8:])
ott = Signal(intbv(0)[8:])
toVerilog(tristate_bank, tri, ift, ott)
toVHDL(tristate_bank, tri, ift, ott)
```

In both of the above examples, the list-of-signals port would be converted (Verilog version) to `input wire [7:0] x, output reg [7:0] y` and `inout wire [7:0] tri`, respectively.

cfelton, 2016-04-01:
pinging those who have had an interested in this issue in the past: @josyb, @devbisme.

josyb, 2016-04-01:
@cfelton Did I? ;-)
You specifically target ```TristateSignal``` - which is something I don't use, or rather defer to the vendor's primitives.
I would more interested in the general solution, elaborating all  LoS, because now and then I have to write a wrapper to check the conversion of a *building block*.

cfelton, 2016-04-01:
@josyb you commented when I had the draft of this in a gist.  

That is fine if you use vendor primitives but you will still need an external top-level wrapper or explicit lines for the inouts.  This limitation, I think, warrants serious consideration. 

As for the LoS wrappers, you should be able to create a single top-level wrapper that can be reused to test single ~~module~~ block conversion and synthesis (that's what I do).  A wrapper is often needed, even when LoS is not used, if you want to limit the IO (Quartus has nice tools with the virtualio, ISE non, nor to I believe Vivado).

```python
conversion_wrapper(clock, reset, sdi, sdo, module_to_wrap)
```

josyb, 2016-04-01:
I hoped that you understood the *wink* as *of course* :)
I never have *inouts* :) E.g. for an I2C connection I use:
```vhdl
		SCl         : out std_logic;
		SDaOut      : out std_logic;
		SDaIn       : in  std_logic
```
You are quite right about the wrapper when lots of signals are involved. Must think about that again.

josyb, 2016-04-10:
Rereading previous comment:
At least not in MyHDL, of course my top-level VHDL files (can) have bidirectional signals. Until now I haven't used a MyHDL block as the top-level file for a project.
So I'd correct the previous comment into:
For open-drain signals:
```python
def somefunc( Clk, Reset,  ..., SCl, SDaIn, SDaOut)
```
And for tri-state:
```python
def someotherfunc( Clk, Reset, ..., BusIn, BusOut, EnableBusOut)
```

> can be reused to test single ~~module~~ block conversion

You don't to seem quite happy with ```block```, neither am I, but isn't life too short ...

cfelton, 2016-04-10:
>> can be reused to test single ~~module~~ block conversion
> You don't to seem quite happy with block, neither am I, but isn't life too short ...

@josyb it is not intended to be cynical but I use it as a transition - until it is more common.  The more places folks see the module-block relation the more ingrained it will become.

josyb, 2016-04-10:
Excuse me.But I would simply refer to it as block tout-court.