myhdl/myhdl #167: @block and lists in Python3

ThorimMB, 2016-05-14:
_Block._verifySubs() raises BlockError(InstanceError) in Python3 if lists are used:

myhdl.BlockError: rtl_0: subblock <listcomp> should be encapsulated in a block decorator

I checked the MyHDL code and searched for the problem (_block.py, line 170):
- isinstance(inst, (_Block, _Instantiator)) returns true for the list object
- inst.modctxt is false, so the error is raised

The error must not be raised for list comprehensions. I commented out the two lines and then simulation and conversion works well.

This problem also effects the Bitonic-Sort example, if it is modified to the new API.

jandecaluwe, 2016-05-15:
Is this specific to list comprehensions? I realize now that Python3 added as scope to those, so probably it is a stack handling issue. However, if it would apply to lists in general that would be puzzling.

Can you please provide your failing (small) example? That saves me lots of time guessing and I can immediately add them to the bug tests.

ThorimMB, 2016-05-15:
I took test_issue_127.py and inserted the @block statements for this test_issue_167.py:

[test_issue_167.py.txt](https://github.com/jandecaluwe/myhdl/files/264929/test_issue_167.py.txt)

jandecaluwe, 2016-05-15:
Ok, this is with list comprehensions and I believe I know the cause and how to solve it. But in your original post you also talk about lists in general, which suprizes me. Do you have such an example also?

ThorimMB, 2016-05-15:
Unluckily in my first post the name of the module `<listcomp>` was not displayed because of the Markdown code ...

I have the same problem in other modules, so I checked them again and they seem to have the same cause: returning a list of sub-blocks in a block-function which always raises the exception with `<listcomp>`.

I think _verifySubs() has to special deal list-comprehensions by checking the elements in the list instead of raising the exception.
In Python2 the list is handled by the function _flatten() which already leads to the check of the list's elements.

jandecaluwe, 2016-05-16:
I cannot reproduce the problem with general lists (as I expected), only with list comprehensions.

The `_flatten()` function is not the issue - it works the same over Python version. Rather, the problem is that Python3 introduces an additional anonymous functional scope for list comprehensions, which causes an additional level in the stack frame that `block` uses to  check valid usage.

I have solved this currently by checking for the special name `<listcomp`> and adapting the stack entries accordingly in Python3.

ThorimMB, 2016-05-16:
Works perfectly. Thank you!

closed 2016-05-16 by ThorimMB