myhdl/myhdl #173: Support for port mapping during conversion to Verilog and some general doubts

vineeshvs, 2016-05-29:
Hi,

1. Is 'port mapping' facility available in myHDL? When I tried creating instances of blocks it was making multiple copies of the full block with different names for variables.
2. Where can I get the latest manual for myHDL? Is the one given in http://old.myhdl.org/doku.php/doc:pdf the latest one?
3. Where can I post similar doubts? ( Is the following link the correct mailing list? https://lists.sourceforge.net/lists/listinfo/myhdl-list)

Thanks

jck, 2016-05-30:
2. http://docs.myhdl.org/en/latest/
3. http://discourse.myhdl.org

vineeshvs, 2016-06-01:
Thanks @jck 

1. Given below is what I mean by port-mapping in Verilog. 
`   test_module test_module_instance_1
     (
      .a_in(a_in_1),
      .b_in(b_in_1),
      .c_out(c_out_1),
      );`

`   test_module test_module_instance_2
     (
      .a_in(a_in_2),
      .b_in(b_in_2),
      .c_out(c_out_2),
      );`

Here I am trying to create two instances of the `test_module` in the code as shown above. Would that be possible?

 If I understand correctly, myHDL makes multiple copies of the full content of the `test_module` with different variable names