myhdl/myhdl #179: MyHDL decorators break use of from __future__ import print_function

rqou, 2016-07-17:
MyHDL decorators prevent proper use of ```from __future__ import print_function```. However, you will only notice this if you try to use some of the new capabilities of print such as the ```end``` argument. Minimal test case:
```python
from __future__ import print_function

from myhdl import *

@block
def Demonstration():
    clkhalfperiod = delay(10)
    @always(clkhalfperiod)
    def clk_gen():
        print(".", end="")

    return clk_gen

demo_inst = Demonstration()
sim = Simulation(demo_inst)
sim.run(1000)
```

Output:
```
$ python myhdl_print_future_broken.py 
Traceback (most recent call last):
  File "myhdl_print_future_broken.py", line 14, in <module>
    demo_inst = Demonstration()
  File "/home/rqou/code/myhdl-venv/local/lib/python2.7/site-packages/myhdl-1.0.dev0-py2.7.egg/myhdl/_block.py", line 125, in __call__
    self.srcline, *args, **kwargs)
  File "/home/rqou/code/myhdl-venv/local/lib/python2.7/site-packages/myhdl-1.0.dev0-py2.7.egg/myhdl/_block.py", line 158, in __init__
    self.subs = _flatten(func(*args, **kwargs))
  File "myhdl_print_future_broken.py", line 8, in Demonstration
    @always(clkhalfperiod)
  File "/home/rqou/code/myhdl-venv/local/lib/python2.7/site-packages/myhdl-1.0.dev0-py2.7.egg/myhdl/_always.py", line 85, in _always_decorator
    return _Always(func, args, callinfo=callinfo, sigdict=sigdict)
  File "/home/rqou/code/myhdl-venv/local/lib/python2.7/site-packages/myhdl-1.0.dev0-py2.7.egg/myhdl/_always.py", line 94, in __init__
    super(_Always, self).__init__(self.genfunc, callinfo=callinfo)
  File "/home/rqou/code/myhdl-venv/local/lib/python2.7/site-packages/myhdl-1.0.dev0-py2.7.egg/myhdl/_instance.py", line 102, in __init__
    tree = self.ast
  File "/home/rqou/code/myhdl-venv/local/lib/python2.7/site-packages/myhdl-1.0.dev0-py2.7.egg/myhdl/_instance.py", line 132, in ast
    return _makeAST(self.funcobj)
  File "/home/rqou/code/myhdl-venv/local/lib/python2.7/site-packages/myhdl-1.0.dev0-py2.7.egg/myhdl/_util.py", line 79, in _makeAST
    tree = ast.parse(s)
  File "/usr/lib/python2.7/ast.py", line 37, in parse
    return compile(source, filename, mode, PyCF_ONLY_AST)
  File "<unknown>", line 3
    print (".",end ="")
                   ^
SyntaxError: invalid syntax
```

josyb, 2016-07-17:
I flagged this problem on MyHDL:Discourse http://discourse.myhdl.org/t/conflict-print/99/5 too. But forgot to raise a formal issue. 
Current work-around is to not use the ```end=''``` stuff.
Another work-around is to guard the ```print()``` with an ```if __debug__:``` See [Excluding Code from Conversion](http://docs.myhdl.org/en/stable/manual/conversion.html?highlight=__debug__#excluding-code-from-conversion). This keeps the guarded code from being converted, and allows the use full Python power.

It has nothing to do with MyHDL decorators (IMO).
One option is to extend MyHDL's handling of ```print()```. But read the section on [Print Statement in *What's new in MyHDL 0.6*](http://docs.myhdl.org/en/stable/whatsnew/0.6.html?highlight=print#print-statement). 
Another is making up a new MyHDL keyword to replace ```print()``` and subsequently breaking code ...

rqou, 2016-07-17:
The example I posted has nothing to do with conversion. It seems to be caused by MyHDL's decorator logic trying to extract an AST for only one function by extracting that function's source code using ```inspect``` and trying to turn only that snippet into an AST (```_makeAST``` in ```_util.py```). This obviously would remove the ```from __future__ import print_statement``` line from the original source. Unfortunately, you cannot work around this by putting the import at the top of the decorated function because this import seems to be a special case in Python's parser and cannot be present more than once or not at the top of the file. A really terrible hack might be to somehow use introspection to detect the activation of py3 print and then adding ```from __future__ import print_statement``` before calling ast.parse in ```_makeAST``` (I haven't tested this and don't know if it actually works or not).

josyb, 2016-07-17:
MyHDL parses the code with conversion in mind ... Perhaps e-read the section: http://docs.myhdl.org/en/stable/whatsnew/0.6.html?highlight=print#print-statement.

rqou, 2016-07-17:
Guarding the print() with ```if __debug__``` is not sufficient to work around the error. It gives the same stack trace as above, implying that MyHDL can't even create an AST and hasn't gotten to trying to do conversion-related logic yet.

josyb, 2016-07-17:
That's not good ... let's wait for @jandecaluwe's view on this?
I currently don't use ```end=''```

jandecaluwe, 2016-07-17:
This was addressed in development 1761fc2a0e265bd522d91c4e4bc92c29c828d35c

rqou, 2016-07-17:
Sorry, but I'm pretty sure that fix is still broken. You have just traded "Py3-style print doesn't work" for "Py2-style print doesn't work." Example:
```python
from myhdl import *

@block
def Demonstration():
    clkhalfperiod = delay(10)
    @always(clkhalfperiod)
    def clk_gen():
        if __debug__:
            print ".",

    return clk_gen

demo_inst = Demonstration()
sim = Simulation(demo_inst)
sim.run(1000)
```
```
$ python myhdl_print_py2_broken.py 
Traceback (most recent call last):
  File "myhdl_print_py2_broken.py", line 13, in <module>
    demo_inst = Demonstration()
  File "/home/rqou/code/myhdl-test-venv/local/lib/python2.7/site-packages/myhdl/_block.py", line 125, in __call__
    self.srcline, *args, **kwargs)
  File "/home/rqou/code/myhdl-test-venv/local/lib/python2.7/site-packages/myhdl/_block.py", line 158, in __init__
    self.subs = _flatten(func(*args, **kwargs))
  File "myhdl_print_py2_broken.py", line 6, in Demonstration
    @always(clkhalfperiod)
  File "/home/rqou/code/myhdl-test-venv/local/lib/python2.7/site-packages/myhdl/_always.py", line 85, in _always_decorator
    return _Always(func, args, callinfo=callinfo, sigdict=sigdict)
  File "/home/rqou/code/myhdl-test-venv/local/lib/python2.7/site-packages/myhdl/_always.py", line 94, in __init__
    super(_Always, self).__init__(self.genfunc, callinfo=callinfo)
  File "/home/rqou/code/myhdl-test-venv/local/lib/python2.7/site-packages/myhdl/_instance.py", line 102, in __init__
    tree = self.ast
  File "/home/rqou/code/myhdl-test-venv/local/lib/python2.7/site-packages/myhdl/_instance.py", line 132, in ast
    return _makeAST(self.funcobj)
  File "/home/rqou/code/myhdl-test-venv/local/lib/python2.7/site-packages/myhdl/_util.py", line 81, in _makeAST
    tree = compile(s, filename='<unknown>', mode='exec', flags=flags)
  File "<unknown>", line 4
    print ".",
            ^
SyntaxError: invalid syntax
```
I believe I have a proper fix. PR incoming...

PR #182 (2016-07-18): Correctly fix _makeAST to preserve future flags
Examines the actual feature flags in use when compiling the function f
and passes those flags to compile(). Fixes #179.

Unfortunately, I don't have a unit test for this because a potential unit test can _only_ run with Python 2 and I'm not sure how well that would work.
commit d64ebb43 (2016-07-18): Correctly fix _makeAST to preserve future flags

Examines the actual feature flags in use when compiling the function f
and passes those flags to compile(). Fixes #179.
closed 2016-07-18 by jandecaluwe