myhdl/myhdl #196: @always_comb generator conversion to Verilog is inconsistent when using certain statements

parasyte, 2016-10-22:
I noticed the latest docs show that a simple `@always_comb` block can be converted to a Verilog `assign` statement with a `wire` type for the output: http://docs.myhdl.org/en/latest/manual/conversion_examples.html#a-small-combinatorial-design

The MyHDL and Verilog are reproduced here:

```python
from myhdl import block, always_comb, Signal, intbv

@block
def bin2gray(B, G):
    """ Gray encoder.

    B -- binary input
    G -- Gray encoded output
    """

    @always_comb
    def logic():
        G.next = (B>>1) ^ B

    return logic

if __name__ == '__main__':
    def convert(hdl, width=8):
        B = Signal(intbv(0)[width:])
        G = Signal(intbv(0)[width:])

        inst = bin2gray(B, G)
        inst.convert(hdl=hdl)

    convert(hdl='Verilog')
```

Verilog conversion:

```verilog
// File: bin2gray.v
// Generated by MyHDL 1.0dev
// Date: Fri Oct 21 23:37:14 2016

`timescale 1ns/10ps

module bin2gray (
    B,
    G
);
// Gray encoder.
// 
// B -- binary input
// G -- Gray encoded output

input [7:0] B;
output [7:0] G;
wire [7:0] G;

assign G = ((B >>> 1) ^ B);

endmodule
```

After some trial and error, I discovered that this example does not convert to the same Verilog if the `logic` function contains certain constructs. For example, if you add an [`if __debug__`](http://docs.myhdl.org/en/latest/manual/conversion.html#excluding-code-from-conversion) statement so it reads:

```python
    @always_comb
    def logic():
        '''test'''
        G.next = (B>>1) ^ B

        if __debug__:
            pass
```

The Verilog output will become:

```verilog
input [7:0] B;
output [7:0] G;
reg [7:0] G;

always @(B) begin: BIN2GRAY_LOGIC
    G = ((B >>> 1) ^ B);

end
```

The type of `G` is now `reg`, and the combinational `assign` statement is now an `always` block...

The same conversion will result if a `pass` statement is used directly:

```python
    @always_comb
    def logic():
        '''test'''
        G.next = (B>>1) ^ B
        pass
```

```verilog
input [7:0] B;
output [7:0] G;
reg [7:0] G;

always @(B) begin: BIN2GRAY_LOGIC
    G = ((B >>> 1) ^ B);
    // pass
end
```

I really don't understand the peculiarities between the different Verilog conversions, here. But it suggests that the use of `reg` in this example is going to be synthesized to a flip-flop instead of a direct connection to the input via a few gates.

FWIW, this issue does not occur with multiple *assignments* ... For example, adding a new output called `H` and assigning `H.next = B` ... The Verilog conversion in this case is just two `assign` statements with two wires; `G` and `H`.

josyb, 2017-04-06:
The ```reg``` does not imply the inference of a register. Processes need ```reg``` on their LHS. Here is an explanation of the difference between ```wire``` and ```reg```: https://inst.eecs.berkeley.edu/~cs150/Documents/Nets.pdf
MyHDL handles the first construct:
```python
    @always_comb
    def logic():
        G.next = (B>>1) ^ B
```
by generating ```assign``` statements and thus declare ```wire``` for the LHS argument. As soon as you add conditional code like:
```python
    @always_comb
    def logic():
        '''test'''
        G.next = (B>>1) ^ B

        if __debug__:
            pass
```
it will treat all as a process and convert it to an ```@always``` block, and thus declare the LHS as ```reg```

parasyte, 2017-04-06:
My point is that the `if __debug__` statement is not hardware logic. It's a Python statement that *excludes* its body from compilation. See the documentation linked in the original post.

The bug here is that, though the exclusion works (try with a body other than a `pass` statement) the presence of `if __debug__` incorrectly changes the process into combinatorial logic. Even though though *no such combinatorial logic exists*.

josyb, 2017-04-07:
I'm not sure I would call it a bug, as the converted code will work fine. 
There  are other situations where MyHDL will create a process rather than (a series of) assignments, like:
```python
@always_comb
def assign():
    for i in range(NUM_ASSIGNS):
        G[i].next = (B[i] >> 1) ^ B[i]
```