myhdl/myhdl #199: Broken VHDL / Verilog conversion with python3

MarioSchwalbe, 2016-11-04:
Hi, following the tutorial, I tried the bin2gray example and the simulation seems to work well. But conversion to either VHDL or Verilog raises an exception when running with the Python3 interpreter (Python 2 works). I guess that's not intended. For reference:
```python
#!/usr/bin/env python

from __future__ import print_function
from myhdl import *

def Bin2Gray(B, G, width):
	"""
		Gray encoder.
		B     -- input intbv signal, binary encoded
		G     -- output intbv signal, gray encoded
		width -- bit width
	"""
	@always_comb
	def logic():
		for i in range(width):
			G.next[i] = B[i+1] ^ B[i]

	return logic

def TestBench(width):
	B   = Signal(intbv(0))
	G   = Signal(intbv(0))
	dut = Bin2Gray(B, G, width)

	@instance
	def stimulus():
		for i in range(2**width):
			B.next = intbv(i)
			yield delay(10)
			print("B: " + bin(B, width) + " | G: " + bin(G, width))

	return dut, stimulus

if __name__ == '__main__':
	# check it
	sim = Simulation(TestBench(width=3))
	sim.run()

	# create it
	w = 3
	B = Signal(intbv()[w:])
	G = Signal(intbv()[w:])
	toVerilog(Bin2Gray, B, G, w)
	toVHDL(Bin2Gray, B, G, w)
```
Exception:
```
Traceback (most recent call last):
  File "Bin2Gray.py", line 45, in <module>
    toVerilog(Bin2Gray, B, G, w)
  File "/usr/local/lib/python3.5/site-packages/myhdl/conversion/_toVerilog.py", line 151, in __call__
    genlist = _analyzeGens(arglist, h.absnames)
  File "/usr/local/lib/python3.5/site-packages/myhdl/conversion/_analyze.py", line 164, in _analyzeGens
    v.visit(tree)
  File "/usr/local/Cellar/python3/3.5.2_3/Frameworks/Python.framework/Versions/3.5/lib/python3.5/ast.py", line 245, in visit
    return visitor(node)
  File "/usr/local/Cellar/python3/3.5.2_3/Frameworks/Python.framework/Versions/3.5/lib/python3.5/ast.py", line 253, in generic_visit
    self.visit(item)
  File "/usr/local/Cellar/python3/3.5.2_3/Frameworks/Python.framework/Versions/3.5/lib/python3.5/ast.py", line 245, in visit
    return visitor(node)
  File "/usr/local/lib/python3.5/site-packages/myhdl/conversion/_analyze.py", line 290, in visit_FunctionDef
    self.visitList(node.body)
  File "/usr/local/lib/python3.5/site-packages/myhdl/conversion/_misc.py", line 162, in visitList
    self.visit(n)
  File "/usr/local/Cellar/python3/3.5.2_3/Frameworks/Python.framework/Versions/3.5/lib/python3.5/ast.py", line 245, in visit
    return visitor(node)
  File "/usr/local/Cellar/python3/3.5.2_3/Frameworks/Python.framework/Versions/3.5/lib/python3.5/ast.py", line 255, in generic_visit
    self.visit(value)
  File "/usr/local/Cellar/python3/3.5.2_3/Frameworks/Python.framework/Versions/3.5/lib/python3.5/ast.py", line 245, in visit
    return visitor(node)
  File "/usr/local/lib/python3.5/site-packages/myhdl/conversion/_analyze.py", line 264, in visit_Call
    if node.starargs:
AttributeError: 'Call' object has no attribute 'starargs'
```
Tested on both: Mac OS X (10.12) and Linux (Ubuntu Xenial) with a recent MyHDL.

jck, 2016-11-05:
This has been fixed in master, you can use the dev version till 1.0 is released.

`pip install git+https://github.com/jandecaluwe/myhdl`

MarioSchwalbe, 2016-11-08:
Awesome. Thanks.

closed 2016-11-24 by jck