myhdl/myhdl #202: Exception printing intbv/modbv with Python3

MarioSchwalbe, 2016-11-09:
Hi,

I've found another minor Python3 inconsistency. The following code works with Python 2 but fails with Python 3. This is MyHDL 1.0dev:
```python
from __future__ import print_function
from myhdl import intbv

import sys

x = intbv(10)

print("{}".format(x))
if sys.version_info.major >= 3:
    print("{!a:>20}".format(x))
print("{!r:>20}".format(x))
print("{!s:>20}".format(x))
# Fails with Python3
print("{:>20}".format(x))
```
The last one raises:
```
$ python3 FormatBug.py
a
           intbv(10)
           intbv(10)
                   a
Traceback (most recent call last):
  File "FormatBug.py", line 20, in <module>
    print("{:>20}".format(x))
TypeError: non-empty format string passed to object.__format__
```
The same happens for `modbv` or `Signal` instances if one of the custom integer types is being used.

jck, 2016-11-09:
`intbv` does not define `__format__`. In Python 2, `object.__format__` converts the object to string automatically.

http://stackoverflow.com/questions/24170519/python-typeerror-non-empty-format-string-passed-to-object-format
http://bugs.python.org/issue7994

Maybe we could implement `__format__` for MyHDL's types?

MarioSchwalbe, 2016-11-09:
Oh, did not know that before. From the python docs:
> The conversion field causes a type coercion before formatting. Normally, the job of formatting a value is done by the __format__() method of the value itself. However, in some cases it is desirable to force a type to be formatted as a string, overriding its own definition of formatting. By converting the value to a string before calling __format__(), the normal formatting logic is bypassed.

What I get here is: With no conversion specification `__format__` gets used to print a python object according to its own understanding of formatting. Right? So, (a) since, `int`s can be formatted, it should also work for `intbv`, and (b) would be backwards-compatible. Both would be less confusing for newcomers. If it's not too complicated to implement...

But also not a big deal, as I usually convert to `int` anyway to get the decimal representation, rather than hexadecimal, which seems to be the new default in 1.0dev.

jck, 2016-11-12:
It isn't hard to implement. We just need to figure out what the right behavior _should_ be.

zhijieshi, 2021-10-02:
Still the same in v0.11.

In Python 3.9.x, if I print Signal like

```
        print("{:x}".format(counter))    # counter is Signal(intbv(0)[16:])
```

The error message is

```
TypeError: unsupported format string passed to _Signal.__format__
```

I would think `d`, `b`, `x`, and `X` are commonly used tags for all intbv, modbv, and bool. 

It also would be great to print bool as 0 or 1, instead of False/True, by default.

josyb, 2021-10-03:
>         print("{:x}".format(counter))    # counter is Signal(intbv(0)[16:])
Assuming you want to print the actual value of counter this works:
```      print("{:x}".format(int(counter)))  ```

AND it will print 0 or 1 for a bool ...

zhijieshi, 2021-10-03:
Thanks for the suggestion. Yes. This is what I am doing. Have to put every signal into to `int()`. It would be easier and more consistent to support commonly used format specifiers.

```
                int(load), int(counter), int(p), int(x), int(y),
                int(p_en), int(x_en), int(y_en), int(done))
```

josyb, 2021-10-03:
(@jck) Adding `__format__` is relatively simple: just delegate if to the Signal's `_val` - no architectural decisions to be made; eventually we end up at `int` or `bool` and there the work is already done.

josyb, 2021-10-03:
However it may/will become an issue in conversion (of testbenches) as the conversion tries to make a `writeline()` out of the `print()`