myhdl/myhdl #205: Better error messages when converting to Verilog

josselex, 2016-11-25:
I was trying to convert to verilog and got the strange error
' AttributeError: 'Call' object has no attribute 'starargs' '

It took me some time to figure out that I had a residual "randrange(30)" in my code which of course can not be converted.

Would it be possible to make a better error Message for this? Maybe even stating which line or object can not be converted?

 I just started yesterday with myhdl so I hope this request is not nonsense.

qrqiuren, 2017-04-03:
It seems that the error is thrown by the compiler. Would you please show us which line of which file throws this error?

qrqiuren, 2017-04-03:
Are you using Python >= 3.5 but with an older version of myhdl? It seems that the latest code in the repo has already fixed this problem. See [_FirstPassVisitor.visit_Call](https://github.com/myhdl/myhdl/blob/master/myhdl/conversion/_analyze.py#L270)

josselex, 2017-05-31:
Hi,
I thinl I was using python 3.4
Unfortunately I cannot test this at the moment.

Thomas