myhdl/myhdl #209: Can't use class method multiple times

NicoPy, 2017-01-13:
Instancing class method multiple times does not convert correctly and generates a surprising hierarchy in simulation.

Using the following code :
``` python
from myhdl import block, Signal, always_seq, posedge, always_comb

class FF():
    def __init__(self):
        pass
        
    @block
    def inst(self, i_Clk, i_Data, o_Data):
        
        DataOut = Signal(bool(1))
        
        @always_seq(posedge(i_Clk), reset=None)
        def proc():
          DataOut.next = i_Data
          
        @always_comb
        def logic():
            o_Data.next = DataOut
        
        return proc, logic
         
@ block 
def double_ff(i_Clk, i_Data1, o_Data1, i_Data2, o_Data2):
    
    inst1 = FF().inst(i_Clk, i_Data1, o_Data1)
    inst2 = FF().inst(i_Clk, i_Data2, o_Data2)
    
    return inst1, inst2

if __name__ == "__main__" :
    
    from myhdl import always, delay, instance, posedge
    
    @block
    def test_ff():
        
        i_Clk = Signal(bool(0))
        
        i_Data1 = Signal(bool(0))
        o_Data1 = Signal(bool(0))
        i_Data2 = Signal(bool(0))
        o_Data2 = Signal(bool(0))
        
        dut_inst = double_ff(i_Clk, i_Data1, o_Data1, i_Data2, o_Data2)
        
        @always(delay(10))
        def clk():
            i_Clk.next = not i_Clk
            
        @instance
        def GenerateData1():
            str = "00011011101100"
            for c in str :
                i_Data1.next = c=="1"
                yield i_Clk.posedge
                
        @instance
        def GenerateData_2():
            data = "011101110000111"
            for c in data :
                i_Data2.next = c=="1"
                yield i_Clk.posedge
                
            
        return clk, dut_inst, GenerateData1, GenerateData_2 

    def simulate(timesteps):
        inst = test_ff()
        inst.config_sim(trace=True, timescale='1ns', tracebackup=False)
        inst.run_sim(duration=timesteps)
    
    simulate(300)
    
    def convert_deglitch():
        i_Clk = Signal(bool(0))
        
        i_Data1 = Signal(bool(0))
        o_Data1 = Signal(bool(0))
        i_Data2 = Signal(bool(0))
        o_Data2 = Signal(bool(0))
        
        dut_inst = double_ff(i_Clk, i_Data1, o_Data1, i_Data2, o_Data2)
         
        dut_inst.convert(hdl='VHDL', name="double_ff", path="./vhdl", initial_values=True)
     
    convert_deglitch()
```

Conversion generates the following :
```vhdl
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;
use std.textio.all;

use work.pck_myhdl_10.all;

entity double_ff is
    port (
        i_Clk: in std_logic;
        i_Data1: in std_logic;
        o_Data1: out std_logic;
        i_Data2: in std_logic;
        o_Data2: out std_logic
    );
end entity double_ff;

architecture MyHDL of double_ff is

signal inst_1_DataOut: std_logic := '1';
signal inst_1_DataOut: std_logic := '1';

begin

DOUBLE_FF_INST_1_PROC: process (i_Clk) is
begin
    if rising_edge(i_Clk) then
        inst_1_DataOut <= i_Data1;
    end if;
end process DOUBLE_FF_INST_1_PROC;

o_Data1 <= inst_1_DataOut;

DOUBLE_FF_INST_1_PROC: process (i_Clk) is
begin
    if rising_edge(i_Clk) then
        inst_1_DataOut <= i_Data2;
    end if;
end process DOUBLE_FF_INST_1_PROC;

o_Data2 <= inst_1_DataOut;

end architecture MyHDL;
```
Signal names and process names are duplicated.

When simulating the hierarchy seen in gtkwave is :
[image]

However, the simulation result is correct.

Replacing class FF() with a function solves the problem.

josyb, 2017-01-13:
Nicolas,

I have this working, but I'm not using the ```@block``` yet. You can see what I do over here: https://gist.github.com/josyb/afd84c9a06fdec77f2fd

NicoPy, 2017-01-13:
@josyb 
I've tested your design with and without @block
As a result, simulation and convertion problems appeared when using @block.

@cfelton 
Yes, the above code is a minimal example to show an issue.
The MyHDL version I use is the master one with PR #201 applied on it. However, PR #201 is very light and does not impact core functionalities.

> I reproduced the error with the master branch. What would be useful is to create a PR with a conversion test that shows the above fails.

I don't understand why I should create a PR while I don't have any fix to submit.

josyb, 2017-01-13:
> @josyb
I've tested your design with and without @block
As a result, simulation and convertion problems appeared when using @block.

@NicoPy 
So my code works without the ```@block``` applied ...

@cfelton 
The first question is whether this __class based__ approach is to be supported by MyHDL?
Our _elusive_ BDFL @jandecaluwe will want to have his say on that.

NicoPy, 2017-01-13:
@josyb 
Yes, your code works without @block

cfelton, 2017-01-13:
@josyb yes, it is supported (might have to find the thread(s) that say so :) and, best of my knowledge, it works in 0.9.

josyb, 2017-01-13:
@cfelton
our BDFL once commented that he was surprised that this code converted.
I don't think it is documented either.

jck, 2017-01-14:
> I don't understand why I should create a PR while I don't have any fix to submit.

You don't have to. It just makes the job easier for whoever chooses to fix the bug.

hgomersall, 2017-01-19:
I think this is now fixed in #189 .

NicoPy, 2017-01-20:
I confirm. This is fixed.
Thanks a lot for your support.

jandecaluwe, 2018-02-25:
As #189 is merged into master, I suppose I can close this again?

hgomersall, 2018-02-26:
Yes, I think this is now closable.

closed 2018-02-26 by jandecaluwe