myhdl/myhdl #210: RAM inference not working for VHDL (Verilog is fine)

gasrodriguez, 2017-01-15:
Hi all, I started learning MyHDL these days, and of course I went right to try the development/unstable version first! I really liked so far, but I just discovered a strange (and a little bit scary) bug with the VHDL conversion when tried to generate a simple RAM. 

For make things easier I copied the code from the MyHDL [manual](http://docs.myhdl.org/en/latest/manual/conversion_examples.html?highlight=ram#ram-inference), tweaking a few minor details to make it work with 1.0Dev (block, instances).

    @block
    def RAM(dout, din, addr, we, clk, depth=128):
        """ Ram model """
        mem = [Signal(intbv(0)[8:]) for i in range(depth)]

        @always(clk.posedge)
        def write():
            if we:
                mem[addr].next = din

        @always_comb
        def read():
            dout.next = mem[addr]

        return instances()

The VHDL process created looks good, but the 'mem' signal is not there!

    architecture MyHDL of RAM is

    begin

The example in the manual showed this as the expected output:

    architecture MyHDL of ram is

    type t_array_mem is array(0 to 128-1) of unsigned(7 downto 0);
    signal mem: t_array_mem;

    begin

The Verilog code looks just fine, in particular this:
`reg [7:0] mem [0:128-1];`

Here are the source .py file used along with the .v and .vhd generated:
[RAM.zip](https://github.com/myhdl/myhdl/files/706734/RAM.zip)

Additional info: I downloaded the .zip with source files and installed using `python setup.py install` and I'm using this version of Python in a Mac:
`Python 3.5.2 |Anaconda 4.2.0 (x86_64)` 

Hope it helps!

jck, 2017-01-15:
I can confirm for this example. However, `conversion/general/test_ram.py` seems to be working fine.

gasrodriguez, 2017-01-15:
Hi @jck, thanks for your reply! I played around with the examples in ´test_ram.py´ and I noticed something really strange: it seems that if you convert "the same instance" to Verilog first and then to VHDL, the VHDL version doesn´t have the mem signal declared, as I shown in my original post. But in the other hand if you convert first to VHDL and then to Verilog, VHDL is ok and Verilog fails!

    inst = RAM(clk=clk,
               addr=addr,
               we=we,
               din=din,
               dout=dout)

    inst.convert(hdl="VHDL", path="out")
    inst.convert(hdl="Verilog", path="./out", testbench=False)

VHDL:

    architecture MyHDL of RAM is

    type t_array_mem is array(0 to 128-1) of unsigned(7 downto 0);
    signal mem: t_array_mem;

    begin

Verilog:

    module RAM (
        dout,
        din,
        addr,
        we,
        clk
    );
    // Ram model 

    output [31:0] dout;
    wire [31:0] dout;
    input [31:0] din;
    input [7:0] addr;
    input we;
    input clk;

    assign dout = mem[addr];

    always @(posedge clk) begin: RAM_WRITE
        if (we) begin
            mem[addr] <= din;
        end
    end

    endmodule

However, if you create two different instances and generate each of them, both version works:

    inst1 = RAM(clk=clk,
               addr=addr,
               we=we,
               din=din,
               dout=dout)
    inst2 = RAM(clk=clk,
               addr=addr,
               we=we,
               din=din,
               dout=dout)
    inst1.convert(hdl="VHDL", path="out")
    inst2.convert(hdl="Verilog", path="./out", testbench=False)

jck, 2017-01-15:
Great catch! Looks like conversion modifies the state of the instance somehow. It should not.

cfelton, 2017-01-15:
If I recall correctly, this was a know issue and there might be a PR for this (or a submitted issue).  I believe it manifested differently before but the same issue, if conversion back to back the second had issues.

jck, 2017-04-09:
Related to #176 
That issue seems to have been fixed though.

closed 2018-02-25 by jandecaluwe