myhdl/myhdl #212: Ports got rearranged while conversion

abhisheietk, 2017-02-02:
I have converted a complex bus into a class in order to have simple implementation. But while conversion ports of the bus got rearranged. 

Please see the test code
```
from myhdl import *

class testBus:
    def __init__(self):
        self.a   = Signal(bool(0))
        self.b   = Signal(bool(0))
        self.c   = Signal(bool(0))
        self.d   = Signal(bool(0))
        self.e   = Signal(bool(0))
                         
            
@block
def test(bus):    
    @always(bus.a, bus.b, bus.c, bus.d, bus.e)
    def logic():
        pass
        
    return instances()
    
def convert():
    bus = testBus()
    
    test_inst = test(bus)
    test_inst.convert(hdl='Verilog')
    
if __name__ == '__main__':
    convert()  
 ```

This is the converted code
```
`timescale 1ns/10ps

module test (
    bus_a,
    bus_c,
    bus_b,
    bus_e,
    bus_d
);

input bus_a;
input bus_c;
input bus_b;
input bus_e;
input bus_d;

always @(bus_a, bus_b, bus_c, bus_d, bus_e) begin: TEST_LOGIC
    // pass
end

endmodule  
```

hashhsah, 2017-02-02:
@cfelton I think this is a flaw of [MEP-107](http://dev.myhdl.org/meps/mep-107.html).

One can see why this happens in this source code line:
https://github.com/myhdl/myhdl/blob/master/myhdl/conversion/_analyze.py#L1253

the `__dict__` of an object instance is not ordered, and will return ports in arbitrary order. Previously I mistakenly thought that python3.6 would have ordered attributes, and this problem would go away. Recently I learnt that only class attributes have order preservation in python3.6, while object attributes do not. 

A possible fix is for the `testBus` class to have a `ports()` method, which returns the interface signals in the correct order.
```python
class testBus:
    def __init__(self):
        self.a   = Signal(bool(0))
        self.b   = Signal(bool(0))
        self.c   = Signal(bool(0))
        self.d   = Signal(bool(0))
        self.e   = Signal(bool(0))

    def ports(self):
        return ['a', 'b', 'c', 'd', 'e']
```

Then, in _analyze.py, we could test if the object has a `ports()` method. If it does, we would use it to iterate through interface signals instead of relying on `vars(obj).items()`.

I wrote some dirty hack to make the above work about 2 years ago, but didn't push an PR. Now that I started using MyHDL once again, I found my patch conflicting majorly with the HEAD. I don't currently have time to sort this out at this moment, but do like to see ordered interface port in MyHDL.

cfelton, 2017-02-02:
Couple things to not here, using interfaces for top-level ports is possible but there are many limitations.  As noted the port order and the names may not be preserved.  At this point in time it is not suggested to use interfaces for the top-level ports.

josyb, 2017-02-02:
One quick workaround is to used _named mapping_ in the Verilog/VHDL top-level file?

hashhsah, 2017-02-02:
> At this point in time it is not suggested to use interfaces for the top-level ports.

I have a use-case for interface at top-level: in a parameterized code generator, there are optional top-level pins that comes and go depending on configuration. In the constructor of the Interface class, it reads the configuration file and setup signals in the interface accordingly.

For example, a memory tester (BIST) needs to talk to various types of SRAMs (single-port, dual-port, bit-enable, byte-enable, etc). Therefore the BIST top-level needs a parameterized top-level SRAM interface.

jck, 2017-02-04:
There is no simple way to find the declaration order of class/instance attributes in Python 2.

https://www.python.org/dev/peps/pep-0520/

jck, 2017-02-04:
I think your code would just work with the correct order in Python 3.6

hashhsah, 2017-02-04:
After a simple test, I think jck is right. The following code does print instance property in order.

```python
class A(object):
    def __init__(self):
        self.a = 1; self.b = 2; self.c = 3; self.d = 4;
        self.e = 5; self.f = 6; self.g = 7; self.h = 8;

a=A()
for k,v in vars(a).items(): print(k,v)
```

```
$ python3.6 abc.py 
a 1
b 2
c 3
d 4
e 5
f 6
g 7
h 8
```

Thanks for the metaclass trick for python2.

Would be great if myhdl documentation include pointers to both solutions.

abhisheietk, 2017-02-05:
Thanks allot for the great discussion and help. Its working with python 3.6. 

Regards

closed 2017-02-05 by abhisheietk

jck, 2017-02-06:
Dictionaries being ordered is a side effect of some optimizations in Python 3.6, and is not a part of the Python spec. It might not work with other Python interpreters or cPython 3.7+. Python 3 has clean ways to replace a class's dict with a `collections.OrderedDict` using the `__prepare__` method in metaclasses.