myhdl/myhdl #213: Verilog conversion problem (wrt expression bit width)

vcanja, 2017-02-26:
Hi,

"reg [15:0] a, b, answer; // 16-bit regs
the intent is to evaluate the expression
answer = (a + b) >> 1; " 
taken from http://staff.ustc.edu.cn/~songch/download/IEEE.1364-2005.pdf.

That Verilog code will evaluate (a+b) into a 16 bit result and the most significant bit will be lost before the shift. A MyHDL code that converts into equivalent Verilog code will not lose that bit and the simulation will behave differently than the Verilog code.

josyb, 2017-02-26:
The same applies for VHDL conversion
```python

'''
Created on 26 Feb 2017

@author: Josy
'''

from myhdl import Signal, intbv, ResetSignal, always_seq, toVHDL, instance, StopSimulation, toVerilog
from Utilities import hdlutils

def addshr(clk, reset, a, b, answer):

    @always_seq(clk.posedge, reset=reset)
    def taddshr():
        answer.next = (a + b) >> 1

    return taddshr

def convert():
    clk = Signal(bool(0))
    reset = ResetSignal(0, 1, True)
    a, b, answer = [Signal(intbv(0)[16:]) for _ in range(3)]

    toVHDL(addshr, clk, reset, a, b, answer)
    toVerilog(addshr, clk, reset, a, b, answer)

def tb_addshr():
    Clk = Signal(bool(0))
    Reset = ResetSignal(0, 1, True)
    a, b, answer = [Signal(intbv(0)[16:]) for _ in range(3)]

    dut = addshr(Clk, Reset, a, b, answer)

    tCK = 20

    @instance
    def genclkreset():
        yield hdlutils.genClkReset(Clk, tCK, Reset)

    @instance
    def stimulus():
        yield hdlutils.delayclks(Clk, tCK, 5)
        a.next = 0xfff0
        b.next = 0x000f
        yield hdlutils.delayclks(Clk, tCK, 1)
        a.next = 0xffff
        b.next = 0x0002
        yield hdlutils.delayclks(Clk, tCK, 10)
        raise StopSimulation

    return dut, genclkreset, stimulus

hdlutils.simulate(0, tb_addshr)
convert()

```
genererates the following VHDL:
```VHDL

library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;
use std.textio.all;

use work.pck_myhdl_10.all;

entity addshr is
	port(
		clk    : in  std_logic;
		reset  : in  std_logic;
		a      : in  unsigned(15 downto 0);
		b      : in  unsigned(15 downto 0);
		answer : out unsigned(15 downto 0)
	);
end entity addshr;

architecture MyHDL of addshr is
begin
	taddshr : process(clk, reset) is
	begin
		if (reset = '1') then
			answer <= to_unsigned(0, 16);
		elsif rising_edge(clk) then
			answer <= shift_right((a + b), 1);
		end if;
	end process taddshr;

end architecture MyHDL;
```
The MyHDL simulation:
[image]

does not corroborate with the ModelSim output:
[image]

jck, 2017-04-05:
Has this issue existed before the `block` decorator, or is it a regression?

josyb, 2017-04-06:
@jck Actually I obtained the above results without the ```@block``` which I do not have in my own local branch, so IMO it is fair to say the issue existed before.
I am afraid this may be a tricky issue to resolve, as it has an impact in the simulation particularly how the LHS is evaluated.

vcanja, 2017-04-06:
Hi,

IMO, I think it would be easier to modify the Verilog translator to
generate code that behaves like the current python simulator. Signals in
the RHS could be extended (when needed) to make sure the results will fit.

Best regards,
Viorel

josyb, 2017-04-06:
I mixed up LHS and RHS (as I do with left and right ...). 
Detecting the _construct_ to generate _adapted_ Verilog (and VHDL) code is problematic too. The actual conversion is correct, so we should address the simulation, shouldn't we?

vcanja, 2017-04-06:
I don't know which is correct (the python simulation or the Verilog code).
It depends on whether you want MyHDL to share the same behavior wrt
expression bit width with Verilog (and VHDL) or not. If the desired
behavior is the current python simulation, IMO, the problem is easier to
fix.

There are other areas where MyHDL and Verilog differ for the sake of ease
of use (mixing signed and unsigned operands comes to mind).

Best regards,
Viorel

josyb, 2017-04-06:
The VHDL code :-)
We can depend on ModelSim simulation to follow the VHDL standard.
I don't agree that MyHDL and Verilog differ, at least not in the generated core. It is imperative that the MyHDL simulation exactly matches the simulation of the generated Verilog (initially as VHDL support was added later) code, otherwise CoSimulation would not really work.
You are correct in stipulating that we have to make up our mind how to handle this issue, whether we stick to what Verilog and VHDL do or not.

forrestv, 2017-07-28:
I noticed this issue in a context simplified to the following:

```
x = Signal(intbv(9, min=0, max=10))
@always(clk.posedge)
def f():
    x.next = (x + x) % 3
```
converts to this VHDL:
```
signal x: unsigned(3 downto 0) := 4X"9";
...
if rising_edge(clk) then
    x <= ((x + x) mod 3);
end if;
```
If `x` is `9`, then MyHDL calculates `x.next` as `0`, while the generated VHDL code calculates `x.next` as `2`, since `9+9` is wrapped to `2` before the `%` operator.

**The MyHDL result is the definitely correct one.** Integers in MyHDL are never supposed to implicitly wrap, according to the documentation.

I believe that the best (only?) way to fix this in the general case is to propagate integer bounds through expressions (using interval arithmetic) and generate Verilog/VHDL expressions that never wrap internally (by resizing operands), with only a final potentially wrapping `resize(...)` at the end to make the RHS of the assignment the correct type. That final `resize(...)` will trigger a warning if it does wrap in most GHDL simulators. Synthesis tools should easily be able to get rid of the few unused MSBs that this technique would potentially create.

qrqiuren, 2017-07-28:
I have implemented the code in Verilog similar to @forrestv's case.

```
module test_bhv;

wire [3:0] x;
reg rst, clk;

bhv b(rst, clk, x);

initial begin
    $dumpfile("overflow.vcd");
    $dumpvars(0, test_bhv);

    clk = 0;
    rst = 1;
    #1; rst = 0;
    #1; rst = 1;

    #1; clk = 1;
    #1; clk = 0;
    #1; clk = 1;
    #1; clk = 0;
    #1; clk = 1;
    #1; clk = 0;
    #1; clk = 1;
    #1; clk = 0;

    $finish;
end

endmodule

module bhv(rst, clk, x);

input wire rst, clk;
output reg [3:0] x;

always @ (posedge clk or negedge rst) begin
    if (!rst) begin
        x <= 4'h9;
    end
    else begin
        x <= (x + x) % 3;
    end
end

endmodule
```

The result running in `iverilog` is the same as the result of MyHDL.

josyb, 2017-07-30:
At least VHDL is consistent in both cases  in resizing the parenthesised operation  before applying the *modulo* or *shift* operator. 
The fact that they differ for the *modulo* operation and agree on the *shift* operation doesn't help at all.
Of course the VHDL and Verillog simulation must match the MyHDL simulation.
Being a VHDL-olic I tend to vote for adapting the MyHDL simulation ... I understand that the Verilog-estants may disagree, and they may have point.
@forrestv I don't entirely agree on your view that MyHDL doesn't wrap implicitly; it will throw a HandleBounds error on:
```python
    a, b, c = [Signal(intbv(0)[4:]) for _ in range(3)]

    c.next = a + b
```
whenever the RH value exceeds the range of the LH.

I suggest we ask our BDFL @jandecaluwe for his opinion on this?

forrestv, 2017-07-30:
@josyb By "doesn't wrap implicitly," I meant exactly that: that an overflow will cause an error rather than silently wrapping. To get wrapping behavior, you have to *explicitly* use the `%` or `&` operators. :)

josyb, 2017-07-31:
@forrestv After a night's sleep I totally agree we should keep the MyHDL simulator as the reference and thus make the VHDL (and Verilog ...) conversion match the MyHDL simulation. We will have to introduce *corner cases* for the *shift* and *mod* operations and inspect/traverse the left operand. Note that this left operand can be substantially more complex than the exercise shown. But I guess we can expand your suggestion to a *brute force resize* of every constituent:
```
 x <= resize(((resize(x, 5) + resize(x, 5)) mod 3), 4);
```
Regards,
Josy