myhdl/myhdl #215: Release v0.10.0

jck, 2017-03-27:
I propose that the next version of MyHDL should be 0.10

There are no backwards incompatible features yet. Using the legacy conversion and simulation functions produce warnings. Users can incrementally migrate to using the block decorator.

We can remove support for the old way in the next major release (v1.0). That will open up the possibility for lots of cleanup.

cfelton, 2017-04-04:
I agree with the proposed release plan.

jck, 2017-04-09:
@cfelton @jandecaluwe @hgomersall @josyb 
I've spent some time thinking about the current implementation of `block` and the outstanding issues and I have a radical proposal.

We _desperately_ need a new release on PyPI. `0.9` does not run on Python 3.5 and above. There are many issues pending relating to the block decorator, and some of them are probably stemming from the complexity of supporting both APIs. We do not need to delay the release of the new API just because our implementation is currently lacking. It is really easy to expose the `block` api by wrapping it around the legacy API similar to how https://github.com/jck/uhdl does it.

```py
from functools import wraps

from myhdl import toVerilog

def block(func):
    @wraps(func)
    def deco(*args, **kwargs):
        return Block(func, *args, **kwargs)
    return deco

class Block:
    def __init__(self, func, *args, **kwargs):
        self.func = func
        self.args = args
        self.kwargs = kwargs

    def convert(self):
        toVerilog(self.func, *self.args, **self.kwargs)
```

This serves multiple purposes:

* Provide users an opportunity to move to the new API with zero chance of bugs.
* Our tests using the new API will give us a baseline to isolate existing issues from block based issues.
* There are multiple places where it is unclear whether an issue is a new one relating to the block api or whether it was an existing issue. We will be able to run block based tests with `0.10` to confirm or rule out whether new issues are independent of the implementation of block.

Soon after `0.10`, we can work on cleanly implementing the block API without legacy concerns. Issues will be more easily caught since the tests already use the block API. We'll be able to release a better `1.0`.

If Jan agrees with me I'll send a PR which does what I've described and brings MyHDL to a state where it can be released ASAP.

hgomersall, 2017-04-09:
I've been using the new block stuff in anger for well over a year on actual code we're using and pushing hard. I appreciate there may still be bugs, but I'm really quite happy with its current implementation. Are there outstanding bugs with the current block implementation?

hgomersall, 2017-04-09:
As discussed in #164, `functools.wraps` is inadequate for the job. The core reason being that it handles classes differently to functions, so there won't be zero chance of bugs with the API, there will be all the bugs that we've fixed that are related to convertible class methods.

cfelton, 2017-04-09:
@jck @hgomersall I have lost track of the `@block` issues?  Can someone list them other than the class.method issue discussed in #164 (which seems resolved?).

hgomersall, 2017-04-09:
I think there is only #189  which fixes an issue with naming class methods to allow multiple class instances.

NicoPy, 2017-05-04:
I have created PR#201 to add simulation options with new API.
Unfortunately, this PR is marked as failed because there where network errors during Travis tests.
I believe this PR is useful.

cfelton, 2017-05-05:
@NicoPy you should be able to restart the test in travis (there is a button to rerun the build/tests).  Or you can make a small change and update the PR.

NicoPy, 2017-05-05:
@cfelton : I've not found any way to rerun the tests.

cfelton, 2017-05-05:
Usually there is a "Restart build" button that will rerun the tests.  I think some of the older PRs got disconnected from travis because the repo name changed.  If you push a minor change to your `simulation_configuration` branch the PR should update. 

[image]

NicoPy, 2017-05-14:
@cfelton Done and Tests passed :)