myhdl/myhdl #218: python3.6.1 and myhdl0.9.0 to convert MyHDL to VHDL

JIANIXNYUN, 2017-04-04:
below is code,a FSM in myhdl
`from

 myhdl import *

ACTIVE_LOW = bool(0)
FRAME_SIZE = 8
t_State = enum('SEARCH', 'CONFIRM', 'SYNC', encoding='one_hot')

def FramerCtrl(SOF, state, syncFlag, clk, reset_n):

    ''' Framing control FSM.

    SOF -- start-of-frame output bit
    state -- FramerState output
    syncFlag -- sync pattern found indication input
    clk -- clock input
    reset_n -- active low reset

    '''

    index = Signal(intbv(0)[8:]) # position in frame

    @always(clk.posedge, reset_n.negedge)
    def FSM():
        if reset_n == ACTIVE_LOW:
            SOF.next = 0
            index.next = 0
            state.next = t_State.SEARCH
        else:
            index.next = (index + 1) % FRAME_SIZE
            SOF.next = 0
            if state == t_State.SEARCH:
                index.next = 1
                if syncFlag:
                    state.next = t_State.CONFIRM
            elif state == t_State.CONFIRM:
                if index == 0:
                    if syncFlag:
                        state.next = t_State.SYNC
                    else:
                        state.next = t_State.SEARCH
            elif state == t_State.SYNC:
                if index == 0:
                    if not syncFlag:
                        state.next = t_State.SEARCH
                SOF.next = (index == FRAME_SIZE-1)
            else:
                raise ValueError('ndefined state')

    return FSM`
and I convert it to Verilog/VHDL,as below
`SOF = Signal(bool(0))
syncFlag = Signal(bool(0))
clk = Signal(bool(0))
reset_n = Signal(bool(1))
state = Signal(t_State.SEARCH)
toVerilog(FramerCtrl, SOF, state, syncFlag, clk, reset_n)`
the resullt as below
`Python 3.6.1 (v3.6.1:69c0db5, Mar 21 2017, 18:41:36) [MSC v.1900 64 bit (AMD64)] on win32
Type "copyright", "credits" or "license()" for more information.
>>> 
============ RESTART: C:\Users\22011\Documents\4565.py ============
Traceback (most recent call last):
  File "C:\Users\22011\Documents\4565.py", line 56, in <module>
    toVerilog(FramerCtrl, SOF, state, syncFlag, clk, reset_n)
  File "C:\Users\22011\AppData\Local\Programs\Python\Python36\lib\site-packages\myhdl\conversion\_toVerilog.py", line 151, in __call__
    genlist = _analyzeGens(arglist, h.absnames)
  File "C:\Users\22011\AppData\Local\Programs\Python\Python36\lib\site-packages\myhdl\conversion\_analyze.py", line 164, in _analyzeGens
    v.visit(tree)
  File "C:\Users\22011\AppData\Local\Programs\Python\Python36\lib\ast.py", line 253, in visit
    return visitor(node)
  File "C:\Users\22011\AppData\Local\Programs\Python\Python36\lib\ast.py", line 261, in generic_visit
    self.visit(item)
  File "C:\Users\22011\AppData\Local\Programs\Python\Python36\lib\ast.py", line 253, in visit
    return visitor(node)
  File "C:\Users\22011\AppData\Local\Programs\Python\Python36\lib\site-packages\myhdl\conversion\_analyze.py", line 290, in visit_FunctionDef
    self.visitList(node.body)
  File "C:\Users\22011\AppData\Local\Programs\Python\Python36\lib\site-packages\myhdl\conversion\_misc.py", line 162, in visitList
    self.visit(n)
  File "C:\Users\22011\AppData\Local\Programs\Python\Python36\lib\ast.py", line 253, in visit
    return visitor(node)
  File "C:\Users\22011\AppData\Local\Programs\Python\Python36\lib\site-packages\myhdl\conversion\_analyze.py", line 312, in visit_If
    self.generic_visit(node)
  File "C:\Users\22011\AppData\Local\Programs\Python\Python36\lib\ast.py", line 261, in generic_visit
    self.visit(item)
  File "C:\Users\22011\AppData\Local\Programs\Python\Python36\lib\ast.py", line 253, in visit
    return visitor(node)
  File "C:\Users\22011\AppData\Local\Programs\Python\Python36\lib\site-packages\myhdl\conversion\_analyze.py", line 312, in visit_If
    self.generic_visit(node)
  File "C:\Users\22011\AppData\Local\Programs\Python\Python36\lib\ast.py", line 261, in generic_visit
    self.visit(item)
  File "C:\Users\22011\AppData\Local\Programs\Python\Python36\lib\ast.py", line 253, in visit
    return visitor(node)
  File "C:\Users\22011\AppData\Local\Programs\Python\Python36\lib\site-packages\myhdl\conversion\_analyze.py", line 312, in visit_If
    self.generic_visit(node)
  File "C:\Users\22011\AppData\Local\Programs\Python\Python36\lib\ast.py", line 261, in generic_visit
    self.visit(item)
  File "C:\Users\22011\AppData\Local\Programs\Python\Python36\lib\ast.py", line 253, in visit
    return visitor(node)
  File "C:\Users\22011\AppData\Local\Programs\Python\Python36\lib\site-packages\myhdl\conversion\_analyze.py", line 312, in visit_If
    self.generic_visit(node)
  File "C:\Users\22011\AppData\Local\Programs\Python\Python36\lib\ast.py", line 261, in generic_visit
    self.visit(item)
  File "C:\Users\22011\AppData\Local\Programs\Python\Python36\lib\ast.py", line 253, in visit
    return visitor(node)
  File "C:\Users\22011\AppData\Local\Programs\Python\Python36\lib\ast.py", line 263, in generic_visit
    self.visit(value)
  File "C:\Users\22011\AppData\Local\Programs\Python\Python36\lib\ast.py", line 253, in visit
    return visitor(node)
  File "C:\Users\22011\AppData\Local\Programs\Python\Python36\lib\site-packages\myhdl\conversion\_analyze.py", line 264, in visit_Call
    if node.starargs:
AttributeError: 'Call' object has no attribute 'starargs'`
how do i solve it??

JIANIXNYUN, 2017-04-04:
this is also same issue,when i run the code as below graph
[image]
the result as below graph
[image]

josyb, 2017-04-04:
Apparently 0.9 fails here for Python 3.5 and above. This has been addressed in the master MyHDL branch.

jck, 2017-04-04:
You can install the master branch with `pip install git+https://github.com/myhdl/myhdl`

closed 2017-04-04 by jck

JIANIXNYUN, 2017-04-12:
@jck 
well,I did it as you say,but result as below
[image]

jck, 2017-04-12:
I'm guessing it means that you don't have git installed. There should be
tutorials online on how to install git on Windows.

JIANIXNYUN, 2017-04-12:
thank you  @jck ,I use myhdl1.0 to convert to VHDL/Verilog,even though red warnings,but generate VHDL/Verilog successfully as below graph
[image]
[image]

jck, 2017-04-12:
Those are just warnings. Follow the link in the warning to see how to fix
them(use block decorator).