myhdl/myhdl #219: I recommand MyHDL should be called PyHDL

JIANIXNYUN, 2017-04-05:
PyHDL sounds very straitforward,it is better to call 'PyHDL' than 'MyHDL',because 'PyHDL' indicates it is a HDL underlying Python

JIANIXNYUN, 2017-04-05:
I found that someone has registered 'PyHDL' trademark,and its website [http://pyhdl.readthedocs.io/en/latest/](url)

jck, 2017-04-09:
Thanks for the suggestion. It doesn't make any sense to change the name at this point though.

closed 2017-04-09 by jck

JIANIXNYUN, 2017-04-12:
PyHDL sounds very Python.I think PyHDL IS better than MyHDL.just like VerilogHDL/VHDL @jck

josyb, 2017-04-12:
Anyway, the name is taken: http://pyhdl.readthedocs.io/en/latest/

JIANIXNYUN, 2017-04-12:
@josyb
I knew PyHDL a little,I will contact PyHDL initiators and MyHDL initiators to develop an easier way to decribe hardware,I found that PyHDL is more like VHDL,just like VHDL in Python

JIANIXNYUN, 2017-04-12:
and what I most want to see is that exist EDA tools(like Quartus Prime) supports MyHDL,and exist development board(such as DE2 board) can download  MyHDL directly into it

joshuisken, 2017-04-13:
I'm trying to do something in this direction... if time permits.

JianXin-Yun, 2017-04-28:
PyHDL initiators@SdNssr 
I have learnt MyHDL,and I found PyHDL,seen its website I found PyHDL is more like VHDL in Python,and some differences between MyHDL, and PyHDL,I hope both can take each other's benefits,and produce one HDL,what do you think about that?

JianXin-Yun, 2017-04-28:
both MyHDL and PyHDL have one goal,is let Python to EDA(like FPGA,CPLD,ASIC),which is let Python to be a HDL better than VerilogHDL/VHDL,I think the biggest problem comes from Intel/Altera, nowadays the most popular EDA tool in college is Quartus,which is developed by Intel(Altera),I think if Quartus can support 
MyHDL/PyHDL(just like VerilogHDL/VHDL),then it will become more popular to decribe hardware in Python,even take place of VerilogHDL/VHDL!!!